reserve85/HoymilesZeroExport #93: ERROR Exception at SetLimit

Tarnatos, 2023-08-16:
Setup:
HMT-2250T
Ahoy-DTU 0.7.26 auf ESP32 mit CMT2300A

Dev Branch auf commit:
https://github.com/reserve85/HoymilesZeroExport/commit/df75ab74f04dec498621e37494c079c3c33a8254

Config:
https://gist.github.com/Tarnatos/4829ac4208d7dcf3fbe8805246e72392

Moin,

nachdem im aktuellen dev branch ja nun auf eine Bestätigung des "Set Limit" gewartet wird, habe ich das mal in meinem Setup aufgespielt.

Ich habe aber immer wieder Fehler im Log. 

Siehe:
2023-08-16 08:31:55,904 INFO     powermeter Shelly 3EM Pro: -205  Watt
2023-08-16 08:31:56,936 INFO     powermeter Shelly 3EM Pro: -205  Watt
2023-08-16 08:31:58,079 INFO     intermediate meter Shelly 3EM: -333  Watt
2023-08-16 08:31:58,079 INFO     overproducing: reduce limit based on actual power
2023-08-16 08:31:58,080 INFO     setting new limit to 225 Watt
2023-08-16 08:31:58,080 INFO     Ahoy: Inverter "GARAGE-HMT-2250": setting new limit from 2250 Watt to 225 Watt
2023-08-16 08:31:58,779 ERROR    Exception at SetLimit
2023-08-16 08:31:58,780 ERROR    'power_limit_ack'
2023-08-16 08:32:18,851 INFO     Ahoy: Inverter "GARAGE-HMT-2250" reachable: True
2023-08-16 08:32:18,886 INFO     powermeter Shelly 3EM Pro: -13  Watt
2023-08-16 08:32:19,917 INFO     powermeter Shelly 3EM Pro: -13  Watt

Wird für die Funktion noch eine neue Ahoy Version benötigt?

Gruß
Tarnatos

reserve85, 2023-08-16:
Ja, du benötigst Ahoy > V0.7.31

Tarnatos, 2023-08-16:
Ah check. Dann warte ich auf das nächste Release.

reserve85, 2023-08-16:
du kannst aber auch die DEV Version von Ahoy aufspielen. https://github.com/lumapu/ahoy/actions/runs/5862171406

closed 2023-08-16 by Tarnatos