reserve85/HoymilesZeroExport #95: [Feature requests] Wechselrichter Priorität

scharrrfi, 2023-08-30:
Hallo zusammen,

ich betreibe das Script für eine Nulleinspeisung nachts.
Ich habe zwei HM-300 und einen HM-1500 an unterschiedlichen Batterien im Einsatz. Die beiden HM-300 decken die Grundlast und der HM-1500 soll - so meine Idee - nur einspringen, wenn die HM-300 es nicht schaffen.

Daher mein Feature requests:
Wäre es vielleicht auch für andere Szenarien sinnvoll, eine Art Wechselrichter Priorität zu integrieren? 
So könnte ich beide HM-300 mit Prio 1 versehen und den HM-1500 mit Prio 2. Wäre natürlich auch denkbar mit 1,2 und 3 usw. 
Funktionsweise: So sollen bis zum definierten Max Leistung der Wechselrichter erst die mit einer höheren Priorität zum Einsatz kommen. Bei Überschreitung kommen die mit niedriger Priorität hinzu.

Was haltet ihr davon?

Ollipop030, 2023-08-30:
Vielleicht gar nicht mit Prioritäten arbeiten, sondern einfach nach der Reihenfolge, wie die WR in Ahoy eingetragen sind:

WR1 (HM300) und WR2(HM1500) dümpeln beim eingetragenen Minimum rum, also sagen wir mal 30Watt (HM300) und 150Watt(HM1500). Jetzt werden aber 300 Watt gebraucht, also Leistung um 120Watt erhöhen, aber nur beim HM300. Wenn bei nächsten Durchlauf die Leistung immer noch nicht reicht, erst dann die Leistung beim HM1500 erhöhen.

scharrrfi, 2023-08-30:
@Ollipop030 Bei deinem Lösungsvorschlag wäre dann aber das Problem, dass man z. B. nicht zwei Wechselrichter auf gleiche Priorität setzen könnte, um sie gleich zu belasten.

reserve85, 2023-09-01:
ich schau mir das irgendwann mal an, bin bisher nicht dazu gekommen und das wird die nächsten Tage auch nix. Die Idee finde ich gut, ich würde den Vorschlag von @scharrrfi bevorzugen.
Das funktioniert aber auch nur für Akkubetrieb - bei Solar wird das denke ich nix, da bei geringer Sonneneinstrahlung z.B. eine Limitänderung zwischen 40% und 100% nix am Ausgangssignal ändert.

scharrrfi, 2023-09-16:
Wenn ich was testen soll, sag Bescheid! :)

reserve85, 2023-09-17:
ja gerne, ich kann das leider überhaupt nicht testen.

scharrrfi, 2023-09-19:
Wenn du die beiden letzten Fixes in Dev eincheckst, würde ich heute Abend mal testen :)

reserve85, 2023-09-19:
Habs integriert, erwarte aber nich zu viel - da ist bestimmt noch irgendwo ein Wurm drin...

scharrrfi, 2023-09-19:
Hast recht :)

Er setzt gar keine Limits für die einzelnen Inverter.

```
Sep 19 20:52:14 Batteriesteuerung python3[14490]: 2023-09-19 20:52:14 INFO     Ahoy: Inverter "HM-300_2": Turn on
Sep 19 20:52:24 Batteriesteuerung python3[14490]: 2023-09-19 20:52:24 INFO     Ahoy: Inverter "HM-1500": Turn on
Sep 19 20:52:34 Batteriesteuerung python3[14490]: 2023-09-19 20:52:34 INFO     Lowest panel voltage inverter "HM-300_1": 26.5 Volt
Sep 19 20:52:34 Batteriesteuerung python3[14490]: 2023-09-19 20:52:34 INFO     Lowest panel voltage inverter "HM-300_2": 26.6 Volt
Sep 19 20:52:34 Batteriesteuerung python3[14490]: 2023-09-19 20:52:34 INFO     Lowest panel voltage inverter "HM-1500": 39.6 Volt
Sep 19 20:52:34 Batteriesteuerung python3[14490]: 2023-09-19 20:52:34 INFO     Ahoy: Inverter "HM-300_1" power producing: 142  Watt
Sep 19 20:52:34 Batteriesteuerung python3[14490]: 2023-09-19 20:52:34 INFO     Ahoy: Inverter "HM-300_2" power producing: 58  Watt
Sep 19 20:52:34 Batteriesteuerung python3[14490]: 2023-09-19 20:52:34 INFO     Ahoy: Inverter "HM-1500" power producing: 336  Watt
Sep 19 20:52:34 Batteriesteuerung python3[14490]: 2023-09-19 20:52:34 INFO     setting new limit to 0 Watt
Sep 19 20:52:34 Batteriesteuerung python3[14490]: 2023-09-19 20:52:34 INFO     powermeter HomeAssistant: -165  Watt
Sep 19 20:52:39 Batteriesteuerung python3[14490]: 2023-09-19 20:52:39 INFO     ---Start Zero Export---
Sep 19 20:52:40 Batteriesteuerung python3[14490]: 2023-09-19 20:52:40 INFO     Ahoy: Inverter "HM-300_1" reachable: True
Sep 19 20:52:40 Batteriesteuerung python3[14490]: 2023-09-19 20:52:40 INFO     Ahoy: Inverter "HM-300_2" reachable: True
Sep 19 20:52:40 Batteriesteuerung python3[14490]: 2023-09-19 20:52:40 INFO     Ahoy: Inverter "HM-1500" reachable: True
Sep 19 20:52:40 Batteriesteuerung python3[14490]: 2023-09-19 20:52:40 INFO     Lowest panel voltage inverter "HM-300_1": 26.5 Volt
Sep 19 20:52:40 Batteriesteuerung python3[14490]: 2023-09-19 20:52:40 INFO     Lowest panel voltage inverter "HM-300_2": 26.6 Volt
Sep 19 20:52:40 Batteriesteuerung python3[14490]: 2023-09-19 20:52:40 INFO     Lowest panel voltage inverter "HM-1500": 39.6 Volt
Sep 19 20:52:40 Batteriesteuerung python3[14490]: 2023-09-19 20:52:40 INFO     powermeter HomeAssistant: -165  Watt
Sep 19 20:52:46 Batteriesteuerung python3[14490]: 2023-09-19 20:52:46 INFO     powermeter HomeAssistant: -163  Watt
Sep 19 20:52:52 Batteriesteuerung python3[14490]: 2023-09-19 20:52:52 INFO     powermeter HomeAssistant: -163  Watt
Sep 19 20:52:58 Batteriesteuerung python3[14490]: 2023-09-19 20:52:58 INFO     overproducing: reduce limit based on previous limit setpoint by approximation
Sep 19 20:52:58 Batteriesteuerung python3[14490]: 2023-09-19 20:52:58 INFO     Inverterlimit already at 0 Watt
Sep 19 20:52:58 Batteriesteuerung python3[14490]: 2023-09-19 20:52:58 INFO     Ahoy: Inverter "HM-300_1" reachable: True
Sep 19 20:52:58 Batteriesteuerung python3[14490]: 2023-09-19 20:52:58 INFO     Ahoy: Inverter "HM-300_2" reachable: True
Sep 19 20:52:58 Batteriesteuerung python3[14490]: 2023-09-19 20:52:58 INFO     Ahoy: Inverter "HM-1500" reachable: True
Sep 19 20:52:58 Batteriesteuerung python3[14490]: 2023-09-19 20:52:58 INFO     Lowest panel voltage inverter "HM-300_1": 26.5 Volt
Sep 19 20:52:58 Batteriesteuerung python3[14490]: 2023-09-19 20:52:58 INFO     Lowest panel voltage inverter "HM-300_2": 26.6 Volt
Sep 19 20:52:59 Batteriesteuerung python3[14490]: 2023-09-19 20:52:59 INFO     Lowest panel voltage inverter "HM-1500": 39.6 Volt
```

reserve85, 2023-09-19:
OK, muss ich morgen anschauen wenn wieder Sonne da ist. Wie gesagt, ist absolut ungetestet...

reserve85, 2023-09-19:
Hab auf die schnelle den Fehler hoffentlich gefunden und gefixt

scharrrfi, 2023-09-19:
Sieht besser aus. Also Limits werden gesetzt und die angegebenen Prio (zumindest in meinem Fall) scheint auch zu funktionieren.

Ich werde es weiter beobachten und berichten. Hab aktuell das Gefühl (was auch täuschen kann), dass er nervöser (und ggf. auch in kürzeren Intervallen) die Limits hin und her setzt.

scharrrfi, 2023-09-20:
Kann es sein, dass die Drosselung mit der Prio über HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V und dann vermutlich auch HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V nicht funktioniert?

Zumindest heute Nacht wurde bei einem Inverter das HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V unterschritten, er ist aber munter weiter im über dem HOY_BATTERY_REDUCE_WATT gelaufen. Ich beobachte das weiter.

scharrrfi, 2023-09-20:
Vermutlich war es das. 

Hab nun beim Stöbern im Log auch gesehen, dass er das Limit für die betroffene Batterie bei der nächsten Limitänderung tatsächlich reduziert wurde. Da die Spannung nach der Drosselung dann aber direkt wieder angestiegen ist, wurde die Drosselung bei der nächsten Limitänderung direkt wieder aufgehoben. Hier muss ich vielleicht auch noch einmal an meiner Config arbeiten.

ragman11, 2023-09-20:
Hallo, funktioniert die Priorität auch bei nicht Battery Powered Inverters. Ich habe eine Mischung, 2 x HM400 ohne battery, und einen HM800 mit Zendure Battery. Ich möchte den HM800 mit Prio 5, da es primär zuerst die hm400 regeln soll.
lg
Mike

reserve85, 2023-09-20:
> Hallo, funktioniert die Priorität auch bei nicht Battery Powered Inverters. Ich habe eine Mischung, 2 x HM400 ohne battery, und einen HM800 mit Zendure Battery. Ich möchte den HM800 mit Prio 5, da es primär zuerst die hm400 regeln soll. lg Mike

nein, aktuell nur Rein Batterie - kein Mischbetrieb. Das ist richtig kompliziert im Mischbetrieb, kann ich mir irgendwann mal anschauen wenn es läuft.

ragman11, 2023-09-20:
Im Moment regle ich nur den HM800, wo das Zendure dranhängt. Dachte mir ein Priomodus bei unterschiedlichen Inverter wäre recht praktisch. Da im Moment die Prozent der WR gleich verteilt werden auch wenn diese verschiedene Wattpeak haben.
2 Scripte wären noch eine Idee, könnte ich mal probieren.
Danke
lg
Michael

scharrrfi, 2023-09-22:
Also ich kann nun erstmal grünes Licht geben. Scheint alles zu funktionieren. :) Denke, du kannst das ganze nach MAIN schieben. 

Leider hat sich meine ursprüngliche Idee als Schnapsidee herausgestellt. Der HM-1500 mag anscheinend ein 0% Limit nicht so gerne und brauch dann gut und gerne mal 30 Minuten, bis er wieder startet und produziert... So funktioniert die Priorisierung mit ihm als Reserve für Leistungspeaks leider nicht. Gleiches gilt auch nach einem Turn-On-Befehl, wenn er über Ahoy abgeschaltet war.

reserve85, 2023-09-22:
Danke, du kannst aber die HOY_MIN_WATT_IN_PERCENT in der Config einstellen. Weniger als 5% geht glaub ich nicht.
Hast du die letzte Version aus der DEV benutzt (1.60)?

reserve85, 2023-09-23:
@scharrrfi probier bitte mal die V1.61 wegen dem MinWatt Problem...

scharrrfi, 2023-09-23:
Läuft! Danke! 
Ich muss dir mal ein Bierchen ausgeben, wenn ich das nächste mal in Aschebersch bin.

reserve85, 2023-09-23:
Ja, sagst mal bescheid 🍻

closed 2023-10-29 by reserve85