reserve85/HoymilesZeroExport #96: Exception bei CheckBattery / GetHoymilesPanelMinVoltage

scharrrfi, 2023-08-30:
Hallo zusammen,

wollte gerade einmal die Batteriesteuerung im Script testen und bekomme Exceptions:
```
Aug 30 15:17:35 Batteriesteuerung python3[1945]: 2023-08-30 15:17:35 ERROR    Exception at GetHoymilesPanelMinVoltage, Inverter 0 not reachable
Aug 30 15:17:35 Batteriesteuerung python3[1945]: 2023-08-30 15:17:35 ERROR    Exception at CheckBattery, Inverter 0 not reachable
Aug 30 15:17:35 Batteriesteuerung python3[1945]: 2023-08-30 15:17:35 ERROR    Exception at GetHoymilesPanelMinVoltage, Inverter 1 not reachable
Aug 30 15:17:35 Batteriesteuerung python3[1945]: 2023-08-30 15:17:35 ERROR    Exception at CheckBattery, Inverter 1 not reachable
Aug 30 15:17:35 Batteriesteuerung python3[1945]: 2023-08-30 15:17:35 ERROR    Exception at GetHoymilesPanelMinVoltage, Inverter 2 not reachable
Aug 30 15:17:35 Batteriesteuerung python3[1945]: 2023-08-30 15:17:35 ERROR    Exception at CheckBattery, Inverter 2 not reachable
```

Denke, da passt irgendwas noch. Die Wechselrichter sind erreichbar.

reserve85, 2023-08-30:
da hat sich ein Fehler eingeschlichen, sollte jetzt wieder laufen.

scharrrfi, 2023-09-16:
Moin,

bin nun endlich zum Testen gekommen. Danke fürs fixen! Gibt nun keine Exception mehr. 

Allerdings bin ich mir aktuell nicht sicher, ob der Batterie-Mode bei mehreren Wechselrichtern korrekt funktioniert.

Beim ersten Start gibt er mir im Log noch die korrekten Spannungswerte aus:
```
Sep 16 07:54:23 Batteriesteuerung python3[10379]: 2023-09-16 07:54:23 INFO     Lowest panel voltage: 26.7 Volt
Sep 16 07:54:23 Batteriesteuerung python3[10379]: 2023-09-16 07:54:23 INFO     Lowest panel voltage: 26.7 Volt
Sep 16 07:54:23 Batteriesteuerung python3[10379]: 2023-09-16 07:54:23 INFO     Lowest panel voltage: 39.2 Volt
```

Danach jedoch nur noch vom letzten Wechselrichter.
```
Sep 16 07:55:31 Batteriesteuerung python3[10379]: 2023-09-16 07:55:31 INFO     Ahoy: Inverter "HM-300_1" reachable: True
Sep 16 07:55:31 Batteriesteuerung python3[10379]: 2023-09-16 07:55:31 INFO     Ahoy: Inverter "HM-300_2" reachable: True
Sep 16 07:55:31 Batteriesteuerung python3[10379]: 2023-09-16 07:55:31 INFO     Ahoy: Inverter "HM-1500" reachable: True
Sep 16 07:55:31 Batteriesteuerung python3[10379]: 2023-09-16 07:55:31 INFO     Lowest panel voltage: 39.2 Volt
Sep 16 07:55:31 Batteriesteuerung python3[10379]: 2023-09-16 07:55:31 INFO     Lowest panel voltage: 39.2 Volt
Sep 16 07:55:31 Batteriesteuerung python3[10379]: 2023-09-16 07:55:31 INFO     Lowest panel voltage: 39.2 Volt
```

reserve85, 2023-09-16:
Kannst du mir mal das komplette Log geben (wenigstens die ersten 2 Minuten) und die config? Kannst du hier als txt Anhängen. Auf die schnelle kann ich nichts finden, ist die gleiche Funktion die aufgerufen wird.

scharrrfi, 2023-09-16:
Sieht auf den ersten Blick gut aus. Genaueres kann ich dir leider erst morgen sagen, da der HM-1500 heute Nacht nicht im Betrieb ist, da ich die Batterien aktuell nochmal einzeln laden muss (unbalanced...).

closed 2023-09-17 by scharrrfi