reserve85/HoymilesZeroExport #100: Battery_Mode: Set Limit nach "Turn on"

scharrrfi, 2023-09-18:
Hallo zusammen,

wenn im Battery_Mode ein Wechselrichter nach erreichen des `HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V` wieder eingeschaltet wird, dann hat er bis zu einer Änderung des Limits weiterhin sein altes, letztes Limit.

Ich denke, es wäre sinnvoll den Inverter hier nach dem Einschalten neu zu initialisieren und auf sein eingestelltes Minimallimit zu ändern. 

Konkreter Usecase: Bei mir ist normalerweise tagesüber der Strombedarf durch die PV gedeckt. Heißt, das Limit im Script ist bei 0W. Wenn ein Inverter nun nach dem Aufladen wieder eingeschaltet wird, speist er mit seinem "alten" Limit (also das vor dem Ausschalten) wieder ins Hausnetz ein. Das ist aber nicht gewollt.

reserve85, 2023-09-18:
sollte funktionieren

scharrrfi, 2023-09-19:
Super danke! Sobald sich mal wieder eine Wechselrichter abgeschaltet hat, werde ich berichten, ob es geklappt hat!

closed 2023-09-19 by scharrrfi