reserve85/HoymilesZeroExport #99: Battery_Mode: Inverter werden plötzlich abgeschaltet

scharrrfi, 2023-09-18:
Moin,

anscheinend ist mit dem Battery_Mode doch noch nicht alles in Ordnung. Ich bin noch auf dem Main-Kanal.

Letzte Nacht haben wurden durch das Script erst meine kleinen Inverter beide zeitgleich ohne ersichtlichen Grund abgeschaltet und später dann auch der große. Alle Abschaltungen gehen mit einer vorhergehenden Exception einher. Die eingestellten Spannungen in der Config für die Abschaltung sind deutlich geringer. 

Hier die Logauszüge zu der entsprechenden Uhrzeit:
```
Sep 17 22:42:21 Batteriesteuerung python3[11890]: 2023-09-17 22:42:21 INFO     powermeter HomeAssistant: -5  Watt
Sep 17 22:42:27 Batteriesteuerung python3[11890]: 2023-09-17 22:42:27 INFO     powermeter HomeAssistant: -5  Watt
Sep 17 22:42:33 Batteriesteuerung python3[11890]: 2023-09-17 22:42:33 INFO     powermeter HomeAssistant: -13  Watt
Sep 17 22:42:39 Batteriesteuerung python3[11890]: 2023-09-17 22:42:39 INFO     Inverterlimit already at 306 Watt
Sep 17 22:42:40 Batteriesteuerung python3[11890]: 2023-09-17 22:42:40 INFO     Ahoy: Inverter "HM-300_1" reachable: True
Sep 17 22:42:40 Batteriesteuerung python3[11890]: 2023-09-17 22:42:40 INFO     Ahoy: Inverter "HM-300_2" reachable: True
Sep 17 22:42:50 Batteriesteuerung python3[11890]: 2023-09-17 22:42:50 ERROR    Exception at GetHoymilesAvailable, Inverter 2 (HM-1500) not reachable
Sep 17 22:42:50 Batteriesteuerung python3[11890]: 2023-09-17 22:42:50 ERROR    HTTPConnectionPool(host='192.168.101.170', port=80): Max retries exceeded with url: /api/ind>
Sep 17 22:42:53 Batteriesteuerung python3[11890]: 2023-09-17 22:42:53 INFO     Lowest panel voltage: 26.7 Volt
Sep 17 22:42:53 Batteriesteuerung python3[11890]: 2023-09-17 22:42:53 INFO     Ahoy: Inverter "HM-300_1": Turn off
Sep 17 22:43:03 Batteriesteuerung python3[11890]: 2023-09-17 22:43:03 INFO     Lowest panel voltage: 26.6 Volt
Sep 17 22:43:03 Batteriesteuerung python3[11890]: 2023-09-17 22:43:03 INFO     Ahoy: Inverter "HM-300_2": Turn off
Sep 17 22:43:34 Batteriesteuerung python3[11890]: 2023-09-17 22:43:34 INFO     Ahoy: Inverter "HM-300_1" reachable: False
Sep 17 22:43:34 Batteriesteuerung python3[11890]: 2023-09-17 22:43:34 INFO     Ahoy: Inverter "HM-300_2" reachable: True
Sep 17 22:43:34 Batteriesteuerung python3[11890]: 2023-09-17 22:43:34 INFO     Ahoy: Inverter "HM-1500" reachable: True
Sep 17 22:43:34 Batteriesteuerung python3[11890]: 2023-09-17 22:43:34 INFO     Ahoy: Inverter "HM-300_2" / serial number "112182852852" / temperature 32.8 degC
Sep 17 22:43:34 Batteriesteuerung python3[11890]: 2023-09-17 22:43:34 INFO     Ahoy: Inverter "HM-1500" / serial number "116190748522" / temperature 42.5 degC
Sep 17 22:43:34 Batteriesteuerung python3[11890]: 2023-09-17 22:43:34 INFO     Lowest panel voltage: 26.7 Volt
Sep 17 22:43:34 Batteriesteuerung python3[11890]: 2023-09-17 22:43:34 INFO     Lowest panel voltage: 39.7 Volt
```

```
Sep 18 01:28:40 Batteriesteuerung python3[11890]: 2023-09-18 01:28:40 INFO     powermeter HomeAssistant: -6  Watt
Sep 18 01:28:46 Batteriesteuerung python3[11890]: 2023-09-18 01:28:46 INFO     powermeter HomeAssistant: -6  Watt
Sep 18 01:28:52 Batteriesteuerung python3[11890]: 2023-09-18 01:28:52 INFO     powermeter HomeAssistant: -5  Watt
Sep 18 01:28:58 Batteriesteuerung python3[11890]: 2023-09-18 01:28:58 INFO     Inverterlimit already at 399 Watt
Sep 18 01:28:58 Batteriesteuerung python3[11890]: 2023-09-18 01:28:58 INFO     Ahoy: Inverter "HM-300_1" reachable: True
Sep 18 01:28:58 Batteriesteuerung python3[11890]: 2023-09-18 01:28:58 INFO     Ahoy: Inverter "HM-300_2" reachable: True
Sep 18 01:28:58 Batteriesteuerung python3[11890]: 2023-09-18 01:28:58 INFO     Ahoy: Inverter "HM-1500" reachable: True
Sep 18 01:29:08 Batteriesteuerung python3[11890]: 2023-09-18 01:29:08 ERROR    Exception at GetHoymilesPanelMinVoltage, Inverter 0 not reachable
Sep 18 01:29:08 Batteriesteuerung python3[11890]: 2023-09-18 01:29:08 ERROR    Exception at CheckBattery, Inverter 0 not reachable
Sep 18 01:29:11 Batteriesteuerung python3[11890]: 2023-09-18 01:29:11 INFO     Lowest panel voltage: 26.7 Volt
Sep 18 01:29:11 Batteriesteuerung python3[11890]: 2023-09-18 01:29:11 INFO     Ahoy: Inverter "HM-300_2": Turn off
Sep 18 01:29:22 Batteriesteuerung python3[11890]: 2023-09-18 01:29:22 INFO     Lowest panel voltage: 39.5 Volt
Sep 18 01:29:22 Batteriesteuerung python3[11890]: 2023-09-18 01:29:22 INFO     Ahoy: Inverter "HM-1500": Turn off
Sep 18 01:29:52 Batteriesteuerung python3[11890]: 2023-09-18 01:29:52 INFO     Ahoy: Inverter "HM-300_1" reachable: True
Sep 18 01:29:52 Batteriesteuerung python3[11890]: 2023-09-18 01:29:52 INFO     Ahoy: Inverter "HM-300_2" reachable: True
Sep 18 01:29:52 Batteriesteuerung python3[11890]: 2023-09-18 01:29:52 INFO     Ahoy: Inverter "HM-1500" reachable: True
Sep 18 01:29:52 Batteriesteuerung python3[11890]: 2023-09-18 01:29:52 INFO     Lowest panel voltage: 26.8 Volt
Sep 18 01:29:52 Batteriesteuerung python3[11890]: 2023-09-18 01:29:52 INFO     Lowest panel voltage: 26.7 Volt
Sep 18 01:29:52 Batteriesteuerung python3[11890]: 2023-09-18 01:29:52 INFO     Lowest panel voltage: 39.7 Volt
```

Kannst du dir das mal ansehen?

reserve85, 2023-09-18:
Generell ist es so, dass bei einer Exception versucht wird der betroffene Inverter zur Sicherheit abzuschalten. Ein paar (oder einer?) User wollten diesen "Not-Aus"-Versuch im Batteriemodus. Wenn er wieder erreichbar ist wird er ja wieder eingeschaltet (wenn über der Anschalt-Spannung)

Warum aber auch die anderen Inverter abschalten ist mir noch nicht ganz klar, muss ich mir anschauen... 
Edit: Hierzu habe ich noch ein Bug entdeckt, ich checke es heute Abend ein. Danke fürs melden!

scharrrfi, 2023-09-18:
Danke fürs Fixen!

Zu der generellen Thematik: Wäre es nicht sinnvoll, den betroffenen Wechselrichter bei Erreichbarkeit und "Normalspannung" wieder einzuschalten (also das Not-Aus aufzuheben) anstatt bei der Anschalt-Spannung? Ich verstehe die Anschaltspannung eigentlich so, dass sie als Ladeabschluss-Spannung gedacht ist, um einen komplett entladenen Akku erst nach dem vollständigen Aufladen wieder einzuschalten und nicht während des Ladens.

Weiterhin waren ja beide Exceptions quasi einmalige Timeouts (aus unterschiedlichen Gründen). Ist hier dann, bei anschließender Erreichbarkeit wirklich auch schon ein Not-Aus notwendig? Sowas kann schließlich in der Funkkommunikation immer mal vorkommen.

reserve85, 2023-09-18:
bitte nochmal testen, da war ein Fehler drin.

scharrrfi, 2023-09-19:
Also heute Nacht gab es keine Exception und auch keine Abschaltung. Ich werde es weiter beobachten.

closed 2023-09-19 by scharrrfi