reserve85/HoymilesZeroExport #101: Nur ein WR wird geregelt

mtdnet, 2023-09-20:
Hallo zusammen, 
habe das Problem das nur einer der beiden HM1500 vom Script geregelt wird.
Hier der Output vom Script

Sep 20 12:28:58 ux-is python3[72670]: 2023-09-20 12:28:58 INFO     Ahoy: Inverter "HM-1500": setting new limit from 747 Watt to 747 Watt
Sep 20 12:28:58 ux-is python3[72670]: 2023-09-20 12:28:58 ERROR    Exception at SetLimit
Sep 20 12:28:58 ux-is python3[72670]: 2023-09-20 12:28:58 ERROR    'power_limit_ack'
Sep 20 12:29:18 ux-is python3[72670]: 2023-09-20 12:29:18 INFO     Ahoy: Inverter "HM-1500" reachable: True
Sep 20 12:29:18 ux-is python3[72670]: 2023-09-20 12:29:18 INFO     Ahoy: Inverter "HM-1500B" reachable: True
Sep 20 12:29:18 ux-is python3[72670]: 2023-09-20 12:29:18 INFO     Ahoy: Inverter "HM-1500" temperature: 43.1 degC
Sep 20 12:29:18 ux-is python3[72670]: 2023-09-20 12:29:18 INFO     Ahoy: Inverter "HM-1500B" temperature: 34.6 degC

AhoyDTU V 0.7.6

Könnt ihr mir weiterhelfen?
Danke!

reserve85, 2023-09-20:
ja, du musst deine AhoyDTU updaten, mindestens auf V0.7.29.

mtdnet, 2023-09-20:
Ach Gott -- das habe ich übersehen
VIELEN VIELEN Dank -- Funktioniert nun!

closed 2023-09-20 by mtdnet