reserve85/HoymilesZeroExport #102: Limit wird nicht gesetzt obwohl acknowledged

Thomas-D-de, 2023-09-22:
Hallo,
vielen Dank für das Script.
Ich hatte den Eindruck, dass es schon mal funktioniert hatte, jetzt wird das Limit nicht mehr gesetzt. Zwischendurch habe ich mit der Original-DTU das VDE-Profil eingestellt.
Im Log steht:
2023-09-22 10:44:39 INFO     Log write to file: False
2023-09-22 10:44:39 INFO     Python Version: 3.9.2 (default, Mar 12 2021, 04:06:34) 
[GCC 10.2.1 20210110]
2023-09-22 10:44:39 INFO     Author: Tobias Kraft / Script Version: 1.56
2023-09-22 10:44:39 INFO     read config file: /home/pi/HoymilesZeroExport_Config.ini
2023-09-22 10:44:39 INFO     Config file V 1.50
2023-09-22 10:44:39 INFO     ---Init---
2023-09-22 10:44:39 INFO     OpenDTU: Inverter "yet unknown" reachable: True
2023-09-22 10:44:39 INFO     OpenDTU: Inverter "WR2-277" / serial number "x" / temperature 22.1 degC
2023-09-22 10:44:39 INFO     OpenDTU: Inverter "yet unknown" reachable: True
2023-09-22 10:44:39 INFO     OpenDTU: Inverter "WR2-277" / serial number "x" / temperature 22.1 degC
2023-09-22 10:44:40 INFO     OpenDTU: Inverter "WR1-700" / serial number "x" / temperature 20.0 degC
2023-09-22 10:44:40 INFO     OpenDTU: Inverter "WR2-277": Turn on
2023-09-22 10:44:50 INFO     OpenDTU: Inverter "WR1-700": Turn on
2023-09-22 10:45:00 INFO     OpenDTU: Inverter "WR2-277" power producing: 383  Watt
2023-09-22 10:45:00 INFO     OpenDTU: Inverter "WR1-700" power producing: 111  Watt
2023-09-22 10:45:00 INFO     setting new limit to 80 Watt
2023-09-22 10:45:00 INFO     OpenDTU: Inverter "WR2-277": setting new limit from 0 Watt to 40 Watt
2023-09-22 10:45:01 INFO     OpenDTU: Inverter "WR2-277": Limit acknowledged
2023-09-22 10:45:01 INFO     OpenDTU: Inverter "WR1-700": setting new limit from 0 Watt to 40 Watt
2023-09-22 10:45:01 INFO     OpenDTU: Inverter "WR1-700": Limit acknowledged
2023-09-22 10:45:01 INFO     powermeter VZLogger: -143  Watt
2023-09-22 10:45:06 INFO     ---Start Zero Export---
2023-09-22 10:45:07 INFO     OpenDTU: Inverter "WR2-277" reachable: True
2023-09-22 10:45:07 INFO     OpenDTU: Inverter "WR1-700" reachable: True
2023-09-22 10:45:07 INFO     powermeter VZLogger: -900  Watt
2023-09-22 10:45:08 INFO     powermeter VZLogger: -900  Watt
2023-09-22 10:45:09 INFO     powermeter VZLogger: -900  Watt
2023-09-22 10:45:10 INFO     powermeter VZLogger: -900  Watt
2023-09-22 10:45:11 INFO     powermeter VZLogger: -900  Watt

In der OpenDTU steht aber:
[image]

Es sind zwei HMS-2000, einmal mit 4 Panels (420 Watt) und der andere mit 2 Panels (420 Watt)
Das Limit wird acknowledged, aber in der OpenDTU nicht gesetzt. Wenn ich in der OpenDTU manuell das Limit setze, wird auch die Produktion reduziert.

OpenDTU Version v23.8.22
Script vom 22.9.23

Hat jemand eine Idee?

reserve85, 2023-09-22:
Kann ich bei mir nicht bestätigen, hier läuft es. Habe auch testweise openDTU auf die V23.9.18 geupdated und das Script aus dem DEV-Pfad verwendet. 
Ich kann allerdings nur einen WR testen, da ich nur einen habe.

```2023-09-22 12:26:14 INFO     Log write to file: False
2023-09-22 12:26:14 INFO     Python Version: 3.11.4 (tags/v3.11.4:d2340ef, Jun  7 2023, 05:45:37) [MSC v.1934 64 bit (AMD64)]
2023-09-22 12:26:14 INFO     Author: Tobias Kraft / Script Version: 1.60
2023-09-22 12:26:14 INFO     read config file: C:\Users\reserve\Downloads\HoymilesZeroExport-dev_new\HoymilesZeroExport_Config.ini
2023-09-22 12:26:14 INFO     Config file V 1.60
2023-09-22 12:26:14 INFO     ---Init---
2023-09-22 12:26:14 INFO     OpenDTU: Inverter "yet unknown" reachable: True
2023-09-22 12:26:14 INFO     OpenDTU: Inverter "HM1500" / serial number "116181853182" / temperature 36.2 degC
2023-09-22 12:26:14 INFO     OpenDTU: Inverter "HM1500": Turn on
2023-09-22 12:26:24 INFO     setting new limit to 75 Watt
2023-09-22 12:26:24 INFO     OpenDTU: Inverter "HM1500": setting new limit from 0 Watt to 75 Watt
2023-09-22 12:26:29 INFO     OpenDTU: Inverter "HM1500": Limit acknowledged```

<img width="406" alt="image" src="https://github.com/reserve85/HoymilesZeroExport/assets/111107925/f41c3b1d-37fa-41d2-931c-b32d73bfdc15">

Thomas-D-de, 2023-09-22:
Fehler gefunden. DTU-Passwort im Config-File war falsch. Merkwürdig das das neue Limit trotzdem "acknowledged" wird.

closed 2023-09-22 by Thomas-D-de