reserve85/HoymilesZeroExport #103: Nach upgrade kein start mehr möglich

tqma1, 2023-10-10:
Lange habe ich hier nichts mehr an meinem Server gemacht und heute das Upgrade von Version 1.31 auf die aktuelle 1.60 gemacht.
Leider startet es nun nicht mehr wie gewünscht und aus der Fehlermeldung werde ich nicht schlau ...

Okt 10 18:11:25 Debian-Server systemd[1]: Started HoymilesZeroExport Service.
Okt 10 18:11:25 Debian-Server python3[5156]: Traceback (most recent call last):
Okt 10 18:11:25 Debian-Server python3[5156]:   File "/home/user/HoymilesZeroExport/HoymilesZeroExport.py", line 32, in <module>
Okt 10 18:11:25 Debian-Server python3[5156]:     from packaging import version
Okt 10 18:11:25 Debian-Server python3[5156]: ModuleNotFoundError: No module named 'packaging'
Okt 10 18:11:25 Debian-Server systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Okt 10 18:11:25 Debian-Server systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Okt 10 18:11:26 Debian-Server systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 1.
Okt 10 18:11:26 Debian-Server systemd[1]: Stopped HoymilesZeroExport Service.
Okt 10 18:11:26 Debian-Server systemd[1]: Started HoymilesZeroExport Service.
Okt 10 18:11:26 Debian-Server python3[5160]: Traceback (most recent call last):
Okt 10 18:11:26 Debian-Server python3[5160]:   File "/home/user/HoymilesZeroExport/HoymilesZeroExport.py", line 32, in <module>
Okt 10 18:11:26 Debian-Server python3[5160]:     from packaging import version
Okt 10 18:11:26 Debian-Server python3[5160]: ModuleNotFoundError: No module named 'packaging'
Okt 10 18:11:26 Debian-Server systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Okt 10 18:11:26 Debian-Server systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Okt 10 18:11:26 Debian-Server systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 2.
Okt 10 18:11:26 Debian-Server systemd[1]: Stopped HoymilesZeroExport Service.
Okt 10 18:11:26 Debian-Server systemd[1]: Started HoymilesZeroExport Service.
Okt 10 18:11:26 Debian-Server python3[5161]: Traceback (most recent call last):
Okt 10 18:11:26 Debian-Server python3[5161]:   File "/home/user/HoymilesZeroExport/HoymilesZeroExport.py", line 32, in <module>
Okt 10 18:11:26 Debian-Server python3[5161]:     from packaging import version
Okt 10 18:11:26 Debian-Server python3[5161]: ModuleNotFoundError: No module named 'packaging'
Okt 10 18:11:26 Debian-Server systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Okt 10 18:11:26 Debian-Server systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Okt 10 18:11:27 Debian-Server systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 3.
Okt 10 18:11:27 Debian-Server systemd[1]: Stopped HoymilesZeroExport Service.
Okt 10 18:11:27 Debian-Server systemd[1]: Started HoymilesZeroExport Service.
Okt 10 18:11:27 Debian-Server python3[5162]: Traceback (most recent call last):
Okt 10 18:11:27 Debian-Server python3[5162]:   File "/home/user/HoymilesZeroExport/HoymilesZeroExport.py", line 32, in <module>
Okt 10 18:11:27 Debian-Server python3[5162]:     from packaging import version
Okt 10 18:11:27 Debian-Server python3[5162]: ModuleNotFoundError: No module named 'packaging'
Okt 10 18:11:27 Debian-Server systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Okt 10 18:11:27 Debian-Server systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Okt 10 18:11:27 Debian-Server systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 4.
Okt 10 18:11:27 Debian-Server systemd[1]: Stopped HoymilesZeroExport Service.
Okt 10 18:11:27 Debian-Server systemd[1]: Started HoymilesZeroExport Service.
Okt 10 18:11:27 Debian-Server python3[5163]: Traceback (most recent call last):
Okt 10 18:11:27 Debian-Server python3[5163]:   File "/home/user/HoymilesZeroExport/HoymilesZeroExport.py", line 32, in <module>
Okt 10 18:11:27 Debian-Server python3[5163]:     from packaging import version
Okt 10 18:11:27 Debian-Server python3[5163]: ModuleNotFoundError: No module named 'packaging'
Okt 10 18:11:27 Debian-Server systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Okt 10 18:11:27 Debian-Server systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Okt 10 18:11:28 Debian-Server systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 5.
Okt 10 18:11:28 Debian-Server systemd[1]: Stopped HoymilesZeroExport Service.
Okt 10 18:11:28 Debian-Server systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly.
Okt 10 18:11:28 Debian-Server systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Okt 10 18:11:28 Debian-Server systemd[1]: Failed to start HoymilesZeroExport Service.

tqma1, 2023-10-10:
Korrigiere:
aktuelle Version ist 1.61 - nur im Config-File steht Version 1.60

sobald ich in der HoymilesZeroExport.py folgendes auskommentiere läuft es ganz normal:
# from packaging import version

reserve85, 2023-10-10:
Servus,

du musst nur das package installieren. Siehe https://github.com/reserve85/HoymilesZeroExport/blob/main/CHANGELOG.md#script-8

tqma1, 2023-10-10:
Ist eigentlich installiert ....

root@Debian-Server:/home/user/HoymilesZeroExport# sudo pip3 install packaging
Requirement already satisfied: packaging in /usr/local/lib/python3.11/site-packages (23.2)

reserve85, 2023-10-10:
welche python version erscheint im Log wenn du "from packaging import version" auskommentierst?

tqma1, 2023-10-10:
Ja, ursprünglich Python 3.9 installiert und seit längerem auf Python 3.11.3.
Welches Log meinst Du nun ?

reserve85, 2023-10-10:
Das von mir, beim startup schreibt er die verwendete Python Version ins log. Es kann nämlich sein, dass du das Package für das neue python installiert hast aber nicht für das alte oder andersrum.

tqma1, 2023-10-10:
Klingt logisch - sollte ich das Log im HoymilesZeroExport Verzeichnis finden oder wo liegt das ?

reserve85, 2023-10-11:
das kannst du dir hiermit ansehen: 
```sudo journalctl -u HoymilesZeroExport.service -n 20000 -e -f```

tqma1, 2023-10-11:
Ah, jetzt weiss ich was Du Log meinst - wieder mal den Wald vor lauter Bäumen nicht gesehen 

Das wars, ich musste für Python3.9 das packaging installieren - DANKE

reserve85, 2023-10-11:
Bedeutet dann aber, dass du V3.11 nicht als standard benutzt. Das musst du dann irgendwie händisch umrouten.