reserve85/HoymilesZeroExport #104: ERROR "Exception at GetPowermeterWatts" gefolgt von ERROR "local variable 'Watts' referenced before assignment"

LHansen367, 2023-10-11:
Setup:
HM-1500
Ahoy-DTU 0.7.26 auf ESP32
ioBroker und Phyton auf einem Raspi 400
Der Stromzähler wird via ioBroker mit einem Hichi USB-Lesekopf ausgelesen (das funktioniert sehr gut)

Problem: 
Die Nulleinspeisung startet nicht, weil obige Fehler auftauchen.
Das Logfile zeigt folgendes:
-----Schnipp-----
Okt 08 13:10:33 iobroker systemd[1]: Started HoymilesZeroExport Service.
Okt 08 13:10:33 iobroker python3[1322616]: 2023-10-08 13:10:33 INFO     Log write to file: True
Okt 08 13:10:33 iobroker python3[1322616]: 2023-10-08 13:10:33 INFO     Python Version: 3.9.2 (default, Feb 28 2021, 17:03:44)
Okt 08 13:10:33 iobroker python3[1322616]: [GCC 10.2.1 20210110]
Okt 08 13:10:33 iobroker python3[1322616]: 2023-10-08 13:10:33 INFO     Author: Tobias Kraft / Script Version: 1.61
Okt 08 13:10:33 iobroker python3[1322616]: 2023-10-08 13:10:33 INFO     read config file: /home/pi/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Okt 08 13:10:33 iobroker python3[1322616]: 2023-10-08 13:10:33 INFO     Config file V 1.60
Okt 08 13:10:33 iobroker python3[1322616]: 2023-10-08 13:10:33 INFO     ---Init---
Okt 08 13:10:33 iobroker python3[1322616]: 2023-10-08 13:10:33 INFO     Ahoy: Current Version: 0.7.36
Okt 08 13:10:34 iobroker python3[1322616]: 2023-10-08 13:10:34 INFO     Ahoy: Inverter "yet unknown" reachable: True
Okt 08 13:10:34 iobroker python3[1322616]: 2023-10-08 13:10:34 INFO     Ahoy: Inverter "HM-1500" / serial number "116183067980" / temperature 44.2 degC
Okt 08 13:10:34 iobroker python3[1322616]: 2023-10-08 13:10:34 INFO     Ahoy: Inverter "HM-1500": Turn on
Okt 08 13:10:44 iobroker python3[1322616]: 2023-10-08 13:10:44 INFO     setting new limit to 225 Watt
Okt 08 13:10:44 iobroker python3[1322616]: 2023-10-08 13:10:44 INFO     Ahoy: Inverter "HM-1500": setting new limit from 0 Watt to 225 Watt
Okt 08 13:10:46 iobroker python3[1322616]: 2023-10-08 13:10:46 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
Okt 08 13:10:46 iobroker python3[1322616]: 2023-10-08 13:10:46 INFO     Ahoy: Inverter "HM-1500" power producing: 741  Watt
Okt 08 13:10:46 iobroker python3[1322616]: 2023-10-08 13:10:46 ERROR    Exception at GetPowermeterWatts
Okt 08 13:10:46 iobroker python3[1322616]: 2023-10-08 13:10:46 ERROR    local variable 'Watts' referenced before assignment
Okt 08 13:11:06 iobroker python3[1322616]: 2023-10-08 13:11:06 INFO     ---Start Zero Export---
Okt 08 13:11:06 iobroker python3[1322616]: 2023-10-08 13:11:06 INFO     Ahoy: Inverter "HM-1500" reachable: True
Okt 08 13:11:06 iobroker python3[1322616]: 2023-10-08 13:11:06 INFO     Ahoy: Inverter "HM-1500" temperature: 44.3 degC
Okt 08 13:11:06 iobroker python3[1322616]: 2023-10-08 13:11:06 ERROR    Exception at GetPowermeterWatts
Okt 08 13:11:06 iobroker python3[1322616]: 2023-10-08 13:11:06 ERROR    local variable 'Watts' referenced before assignment
-----Schnapp-----

Die letzten vier Zeilen wiederholen sich ab jetzt.

Ich vermute, dass im ioBroker noch etwas fehlt...
Ich habe unter den Objekten einen 'State' angelegt: simple-api.admin.alias.0.Zaehler.Zaehler_CurrentWatt, der mit einem Skript die Werte vom Smartmeter des aktuellen Verbrauchs (Momentanwert Gesamtwirkleistung) überträgt.
Ebenfalls habe ich unter 'alias' jeweils states angelegt: 
alias.0.Zaehler.Zaehler_CurrentWatt
und
alias.0.simple-api.admin.0.Zaehler.Zaehler_CurrentWatt,
die dann mittels einem Alias direkt dem Wert vom Smartmeter bekommen.

Ich habe die Issues in diesem und anderen Projekten in Github durchsucht, finde aber keine Lösung.
Mit dem ioBroker arbeite ich schon einige Zeit, würde aber nicht behaupten, dass ich hier ein weit fortgeschrittener Anwender bin... Eher ein fortgeschrittener Anfänger...  ;-)
Wenn noch weitere Angaben benötigt werden, lasst es mich wissen.
Danke vorab für die Hilfe!

reserve85, 2023-10-11:
Hey, ja vermutlich kommt von IOBroker nix oder was falsches zurück.
du kannst das "relativ" einfach herausfinden:
```http://{IOBROKER_IP}:{IOBROKER_PORT}/getBulk/{IOBROKER_CURRENT_POWER_ALIAS}```
In der URL die drei Werte IOBROKER_IP, IOBROKER_PORT und IOBROKER_CURRENT_POWER_ALIAS aus der config ersetzen (die geschweifte Klammer entfernen) und dann kannst du schauen ob was zurückkommt.

LHansen367, 2023-10-11:
Ok, die Zeile 
"http://[192.168.178.186:8087/getBulk/Zaehler_CurrentWatt" 
im Browser ergibt folgende Anzeige:
[{"id":"alias.0.Zaehler.Zaehler_CurrentWatt","val":355.87,"ts":1697039498799,"ack":true}]
Die Zahl "355.87" entspricht meinem aktuellen Bezug.
Das sieht für mich erstmal gut aus, oder?

reserve85, 2023-10-11:
Sieht soweit gut aus, kannst du mir die config hier mal anhängen? Kannst du als *.txt umbenennen dann geht das als Anhang. Ist IOBROKER_POWER_CALCULATE auf false?

LHansen367, 2023-10-11:
Anbei die config. Das mit der Umbenennung versuche ich mal...
[HoymilesZeroExport_Config.txt](https://github.com/reserve85/HoymilesZeroExport/files/12872639/HoymilesZeroExport_Config.txt)

reserve85, 2023-10-11:
Kommt was zurück wenn du 

http://[192.168.178.186:8087/getBulk/
alias.0.Zaehler.Zaehler_CurrentWatt

eingibst?

LHansen367, 2023-10-11:
Im ioBroker habe ich jetzt den Value "Momentanwert Gesamtwirkleistung" vom Smartmeter mit dem Namen "Zaehler_CurrentWatt" umbenannt (die anderen 'Versuche' unter alias... und simple-api... habe ich ebenfalls umbenannt, sodass kein doppelter Name mehr vorhanden ist) und erhalte nun folgende Rückmeldung im Browser:
[{"id":"smartmeter.0.1-0:16_7_0__255.value","val":472.47,"ts":1697043887782,"ack":true}]

Gebe ich "http://192.168.178.186:8087/getBulk/alias.0.Zaehler.Zaehler_CurrentWatt" ein, zeigt er mir 
"[{"id":"alias.0.Zaehler.Zaehler_CurrentWatt","val":619.86,"ts":1697045537773,"ack":true}]"
an.

LHansen367, 2023-10-11:
Sodele, nach Neustart des Skrips schaut es nun so aus:

-----Schnipp-----
Okt 11 19:39:52 iobroker systemd[1]: Started HoymilesZeroExport Service.
Okt 11 19:39:53 iobroker python3[37926]: 2023-10-11 19:39:53 INFO     Log write to file: True
Okt 11 19:39:53 iobroker python3[37926]: 2023-10-11 19:39:53 INFO     Python Version: 3.9.2 (default, Feb 28 2021, 17:03:44)
Okt 11 19:39:53 iobroker python3[37926]: [GCC 10.2.1 20210110]
Okt 11 19:39:53 iobroker python3[37926]: 2023-10-11 19:39:53 INFO     Author: Tobias Kraft / Script Version: 1.61
Okt 11 19:39:53 iobroker python3[37926]: 2023-10-11 19:39:53 INFO     read config file: /home/pi/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Okt 11 19:39:53 iobroker python3[37926]: 2023-10-11 19:39:53 INFO     Config file V 1.60
Okt 11 19:39:53 iobroker python3[37926]: 2023-10-11 19:39:53 INFO     ---Init---
Okt 11 19:39:53 iobroker python3[37926]: 2023-10-11 19:39:53 INFO     Ahoy: Current Version: 0.7.36
Okt 11 19:39:53 iobroker python3[37926]: 2023-10-11 19:39:53 INFO     Ahoy: Inverter "yet unknown" reachable: False
Okt 11 19:39:53 iobroker python3[37926]: 2023-10-11 19:39:53 INFO     powermeter IOBROKER: 691  Watt
Okt 11 19:39:58 iobroker python3[37926]: 2023-10-11 19:39:58 INFO     ---Start Zero Export---
Okt 11 19:39:58 iobroker python3[37926]: 2023-10-11 19:39:58 INFO     Ahoy: Inverter "yet unknown" reachable: False
Okt 11 19:40:18 iobroker python3[37926]: 2023-10-11 19:40:18 INFO     Ahoy: Inverter "yet unknown" reachable: False
-----Schnapp-----

Leider ist jetzt schon die Sonne futsch, aber die Info meldet einen Wert vom Stromzähler zurück...
Wir müssen und leider bis morgen früh gedulden....
Vermutlich hätte ich lediglich das Objekt, welches den aktuellen Verbrauch bzw. Einspeisung ausliest und anzeigt mit dem Namen "Zaehler_CurrentWatt" benennen müssen...
Ich gebe morgen ein update, ob meine Theorie stimmt.  ;-)
Erstmal vielen Dank für die Fehlersuche und einen schönen Abend!

reserve85, 2023-10-11:
Perfekt, das passt jetzt so! 
Ich werde ggf. beim nächsten Check-In die Fehlermeldung etwas überarbeiten damit es klarer wird.

LHansen367, 2023-10-12:
Heute ist es sehr bewölkt, aber der Wechselrichter produziert wenigstens ein paar Watt.
Das Logfile zeigt jetzt keinen Fehler mehr an. Somit scheint der Knoten gelöst zu sein.
Hier der Log:
-----Schnipp-----
Okt 12 10:15:18 iobroker systemd[1]: Started HoymilesZeroExport Service.
Okt 12 10:15:18 iobroker python3[63351]: 2023-10-12 10:15:18 INFO     Log write to file: True
Okt 12 10:15:18 iobroker python3[63351]: 2023-10-12 10:15:18 INFO     Python Version: 3.9.2 (default, Feb 28 2021, 17:03:44)
Okt 12 10:15:18 iobroker python3[63351]: [GCC 10.2.1 20210110]
Okt 12 10:15:18 iobroker python3[63351]: 2023-10-12 10:15:18 INFO     Author: Tobias Kraft / Script Version: 1.61
Okt 12 10:15:18 iobroker python3[63351]: 2023-10-12 10:15:18 INFO     read config file: /home/pi/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Okt 12 10:15:18 iobroker python3[63351]: 2023-10-12 10:15:18 INFO     Config file V 1.60
Okt 12 10:15:18 iobroker python3[63351]: 2023-10-12 10:15:18 INFO     ---Init---
Okt 12 10:15:18 iobroker python3[63351]: 2023-10-12 10:15:18 INFO     Ahoy: Current Version: 0.7.36
Okt 12 10:15:18 iobroker python3[63351]: 2023-10-12 10:15:18 INFO     Ahoy: Inverter "yet unknown" reachable: True
Okt 12 10:15:18 iobroker python3[63351]: 2023-10-12 10:15:18 INFO     Ahoy: Inverter "HM-1500" / serial number "116183067980" / temperature 20.6 degC
Okt 12 10:15:18 iobroker python3[63351]: 2023-10-12 10:15:18 INFO     Ahoy: Inverter "HM-1500": Turn on
Okt 12 10:15:29 iobroker python3[63351]: 2023-10-12 10:15:29 INFO     setting new limit to 225 Watt
Okt 12 10:15:29 iobroker python3[63351]: 2023-10-12 10:15:29 INFO     Ahoy: Inverter "HM-1500": setting new limit from 0 Watt to 225 Watt
Okt 12 10:15:29 iobroker python3[63351]: 2023-10-12 10:15:29 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
Okt 12 10:15:29 iobroker python3[63351]: 2023-10-12 10:15:29 INFO     Ahoy: Inverter "HM-1500" power producing: 37  Watt
Okt 12 10:15:29 iobroker python3[63351]: 2023-10-12 10:15:29 INFO     powermeter IOBROKER: 236  Watt
Okt 12 10:15:34 iobroker python3[63351]: 2023-10-12 10:15:34 INFO     ---Start Zero Export---
Okt 12 10:15:34 iobroker python3[63351]: 2023-10-12 10:15:34 INFO     Ahoy: Inverter "HM-1500" reachable: True
Okt 12 10:15:34 iobroker python3[63351]: 2023-10-12 10:15:34 INFO     Ahoy: Inverter "HM-1500" temperature: 20.6 degC
Okt 12 10:15:35 iobroker python3[63351]: 2023-10-12 10:15:35 INFO     powermeter IOBROKER: 236  Watt
Okt 12 10:15:35 iobroker python3[63351]: 2023-10-12 10:15:35 INFO     setting new limit to 1500 Watt
Okt 12 10:15:35 iobroker python3[63351]: 2023-10-12 10:15:35 INFO     Ahoy: Inverter "HM-1500": setting new limit from 225 Watt to 1500 Watt
Okt 12 10:15:35 iobroker python3[63351]: 2023-10-12 10:15:35 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
Okt 12 10:15:50 iobroker python3[63351]: 2023-10-12 10:15:50 INFO     Ahoy: Inverter "HM-1500" reachable: True
Okt 12 10:15:50 iobroker python3[63351]: 2023-10-12 10:15:50 INFO     Ahoy: Inverter "HM-1500" temperature: 20.6 degC
Okt 12 10:15:50 iobroker python3[63351]: 2023-10-12 10:15:50 INFO     powermeter IOBROKER: 217  Watt
Okt 12 10:15:50 iobroker python3[63351]: 2023-10-12 10:15:50 INFO     Inverterlimit already at 1500 Watt
Okt 12 10:16:05 iobroker python3[63351]: 2023-10-12 10:16:05 INFO     Ahoy: Inverter "HM-1500" reachable: True
Okt 12 10:16:05 iobroker python3[63351]: 2023-10-12 10:16:05 INFO     Ahoy: Inverter "HM-1500" temperature: 20.6 degC
Okt 12 10:16:05 iobroker python3[63351]: 2023-10-12 10:16:05 INFO     powermeter IOBROKER: 216  Watt
Okt 12 10:16:05 iobroker python3[63351]: 2023-10-12 10:16:05 INFO     Inverterlimit already at 1500 Watt
-----Schnapp-----

Zusammenfassung:
Anscheinend habe ich die Anleitung falsch verstanden und ein neues Objekt (alias.0.Zaehler.Zaehler_CurrentWatt) im ioBroker erstellt.
Es muss aber lediglich das Objekt, welches den aktuellen Wert vom Stromzähler anzeigt (bei mir "1-0:16_7_0__255" mit dem Namen "Zaehler_CurrentWatt" versehen werden...
Nochmals vielen Dank für die schnelle Hilfe!