reserve85/HoymilesZeroExport #105: keine Kommulikation

Boemelmann, 2023-10-13:
Moin,

ich bekomme das Skript leider nicht zum Laufen. Ich verwende einen Raspberry zero, SHELLY 3EM und eine Ahoy DTU. Alle Lifedaten der DTU kann ich über den Browser sehen. Muss die Inverternummer nirgendwo eingegeben werden?

Hier ein Auszug aus der .ini

  GNU nano 7.2                                                                        HoymilesZeroExport_Config.ini
# --- define your DTU (only one) ---
USE_AHOY = true
USE_OPENDTU = false

[SELECT_POWERMETER]
# --- define your Powermeter (only one) ---
USE_TASMOTA = false
USE_SHELLY_EM = false
USE_SHELLY_3EM = true
USE_SHELLY_3EM_PRO = false
USE_SHRDZM = false
USE_EMLOG = false
USE_IOBROKER = false
USE_HOMEASSISTANT = false
USE_VZLOGGER = false

[AHOY_DTU]
# --- defines for AHOY-DTU ---
# in settings/inverter set interval to 6 seconds!
AHOY_IP = 192.168.178.113

[OPEN_DTU]
# --- defines for OPEN-DTU ---
OPENDTU_IP = xxx.xxx.xxx.xxx
OPENDTU_USER =
OPENDTU_PASS =

[TASMOTA]
# --- defines for Tasmota Smartmeter Modul---
TASMOTA_IP = 192.168.10.90
# the following three constants describes how to navigate through the Tasmota-JSON.
# if you do not know the format of your Tasmota reader, open a browser and put in the following string replacing xxx with the IP address of your Tasmota device: http://xxx.xxx.xxx.xxx/cm?cmnd=stat>
# e.g. JSON_Result = {"StatusSNS":{"Time":"2023-02-28T12:49:49","SML":{"total_kwh":15011.575,"curr_w":-71}}}
TASMOTA_JSON_STATUS = StatusSNS
# Prefix for Web UI and MQTT JSON payload
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = SML
# Power-MQTT label (the current power in Watt, positive (import more energy than export) or negative (export more energy than import)).
TASMOTA_JSON_POWER_MQTT_LABEL = curr_w
# if your powermeter does NOT output the current power: you need to calculate it -> Power(W) = OBIS(1.7.0) - OBIS(2.7.0)
TASMOTA_JSON_POWER_CALCULATE = FALSE
# Power-MQTT Input label (positive active instantaneous power, e.g. OBIS Code 1.7.0)
TASMOTA_JSON_POWER_INPUT_MQTT_LABEL =
# Power-MQTT output label (negative active instantaneous power, e.g. OBIS Code 2.7.0)
TASMOTA_JSON_POWER_OUTPUT_MQTT_LABEL =

[SELECT_POWERMETER]
# --- define your Powermeter (only one) ---
USE_TASMOTA = false
USE_SHELLY_EM = false
USE_SHELLY_3EM = true
USE_SHELLY_3EM_PRO = false
USE_SHRDZM = false
USE_EMLOG = false
USE_IOBROKER = false
USE_HOMEASSISTANT = false
USE_VZLOGGER = false

[AHOY_DTU]
# --- defines for AHOY-DTU ---
# in settings/inverter set interval to 6 seconds!
AHOY_IP = 192.168.178.113

[OPEN_DTU]
[... 292 lines cut ...]
Oct 13 09:25:53 raspberrypi systemd[1]: Started HoymilesZeroExport.service - Hoy                                                                                                                     milesZeroExport Service.
Oct 13 09:25:54 raspberrypi python3[1801]: Traceback (most recent call last):
Oct 13 09:25:54 raspberrypi python3[1801]:   File "/home/admin/HoymilesZeroExpor                                                                                                                     t/HoymilesZeroExport.py", line 32, in <module>
Oct 13 09:25:54 raspberrypi python3[1801]:     from packaging import version
Oct 13 09:25:54 raspberrypi python3[1801]: ModuleNotFoundError: No module named                                                                                                                      'packaging'
Oct 13 09:25:54 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process                                                                                                                      exited, code=exited, status=1/FAILURE
Oct 13 09:25:54 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with                                                                                                                      result 'exit-code'.
Oct 13 09:25:54 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled re                                                                                                                     start job, restart counter is at 4.
Oct 13 09:25:55 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - Hoy                                                                                                                     milesZeroExport Service.
Oct 13 09:25:55 raspberrypi systemd[1]: Started HoymilesZeroExport.service - Hoy                                                                                                                     milesZeroExport Service.
Oct 13 09:25:55 raspberrypi python3[1802]: Traceback (most recent call last):
Oct 13 09:25:55 raspberrypi python3[1802]:   File "/home/admin/HoymilesZeroExpor                                                                                                                     t/HoymilesZeroExport.py", line 32, in <module>
Oct 13 09:25:55 raspberrypi python3[1802]:     from packaging import version
Oct 13 09:25:55 raspberrypi python3[1802]: ModuleNotFoundError: No module named                                                                                                                      'packaging'
Oct 13 09:25:56 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process                                                                                                                      exited, code=exited, status=1/FAILURE
Oct 13 09:25:56 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with                                                                                                                      result 'exit-code'.
Oct 13 09:25:56 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled re                                                                                                                     start job, restart counter is at 5.
Oct 13 09:25:56 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - Hoy                                                                                                                     milesZeroExport Service.
Oct 13 09:25:56 raspberrypi systemd[1]: HoymilesZeroExport.service: Start reques                                                                                                                     t repeated too quickly.
Oct 13 09:25:56 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with                                                                                                                      result 'exit-code'.
Oct 13 09:25:56 raspberrypi systemd[1]: Failed to start HoymilesZeroExport.servi                                                                                                                     ce - HoymilesZeroExport Service.

Über eine Hilfe, würde ich mich sehr freuen.

VG Sebastian

irrwitzer42, 2023-10-13:
Hi @Boemelmann ,

die relevanten Zeilen sind die hier:

Oct 13 09:23:59 raspberrypi python3[1774]: File "/home/admin/HoymilesZeroExpor t/HoymilesZeroExport.py", line 32, in
Oct 13 09:23:59 raspberrypi python3[1774]: from packaging import version
Oct 13 09:23:59 raspberrypi python3[1774]: ModuleNotFoundError: No module named 'packaging'

Installier mal das "packaging" modul nach:

`sudo pip3 install packaging`

Boemelmann, 2023-10-13:
> Hallo Boemelmann ,
> 
> die Zeilen sind die hier:
> 
> ```
> Oct 13 09:23:59 raspberrypi python3[1774]: File "/home/admin/HoymilesZeroExpor t/HoymilesZeroExport.py", line 32, in
> Oct 13 09:23:59 raspberrypi python3[1774]: from packaging import version
> Oct 13 09:23:59 raspberrypi python3[1774]: ModuleNotFoundError: No module named 'packaging'
> ```
> 
> Installier mal das "packaging" modul nach:
> 
> `sudo pip3 install packaging`

danke für die schnelle Antwort.

dann bekomme ich folgende Fehlermeldung.

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

VG

reserve85, 2023-10-13:
versuche mal
```sudo pip3 install packaging --break-system-packages```
aus: https://stackoverflow.com/questions/75608323/how-do-i-solve-error-externally-managed-environment-every-time-i-use-pip-3

irrwitzer42, 2023-10-13:
@Boemelmann ,
oder `sudo apt install python3-packaging`

Boemelmann, 2023-10-13:
> versuche mal `sudo pip3 install packaging --break-system-packages`aus: https://stackoverflow.com/questions/75608323/how-do-i-solve-error-externally-managed-environment-every-time-i-use-pip-3

Moin,

super es hat geklappt. 💯  

Okt 13 11:37:30 raspberrypi python3[2476]: 2023-10-13 11:37:30 INFO     Ahoy: Cu                                                                                                                     rrent Version: 0.7.36
Okt 13 11:37:30 raspberrypi python3[2476]: 2023-10-13 11:37:30 INFO     Ahoy: In                                                                                                                     verter "yet unknown" reachable: True
Okt 13 11:37:31 raspberrypi python3[2476]: 2023-10-13 11:37:31 INFO     Ahoy: In                                                                                                                     verter "Balkonsolar" / serial number "116183121579" / temperature 21.6 degC
Okt 13 11:37:31 raspberrypi python3[2476]: 2023-10-13 11:37:31 INFO     Ahoy: In                                                                                                                     verter "Balkonsolar": Turn on
Okt 13 11:37:41 raspberrypi python3[2476]: 2023-10-13 11:37:41 INFO     setting                                                                                                                      new limit to 75 Watt
Okt 13 11:37:41 raspberrypi python3[2476]: 2023-10-13 11:37:41 INFO     Ahoy: In                                                                                                                     verter "Balkonsolar": setting new limit from 0 Watt to 75 Watt
Okt 13 11:37:41 raspberrypi python3[2476]: 2023-10-13 11:37:41 INFO     Ahoy: In                                                                                                                     verter "Balkonsolar": Limit acknowledged
Okt 13 11:37:42 raspberrypi python3[2476]: 2023-10-13 11:37:42 WARNING  Certific                                                                                                                     ate did not match expected hostname: 192.169.178.93. Certificate: {'subject': ((                                                                                                                     ('commonName', 'fbfanimation.com'),),), 'issuer': ((('countryName', 'GB'),), (('                                                                                                                     stateOrProvinceName', 'Greater Manchester'),), (('localityName', 'Salford'),), (                                                                                                                     ('organizationName', 'Sectigo Limited'),), (('commonName', 'Sectigo RSA Domain V                                                                                                                     alidation Secure Server CA'),)), 'version': 3, 'serialNumber': 'CE37276CC1AB86E1                                                                                                                     5D58013D6082DA60', 'notBefore': 'Oct 17 00:00:00 2022 GMT', 'notAfter': 'Nov 17                                                                                                                      23:59:59 2023 GMT', 'subjectAltName': (('DNS', 'fbfanimation.com'), ('DNS', 'www                                                                                                                     .fbfanimation.com')), 'OCSP': ('http://ocsp.sectigo.com',), 'caIssuers': ('http:                                                                                                                     //crt.sectigo.com/SectigoRSADomainValidationSecureServerCA.crt',)}
Okt 13 11:37:42 raspberrypi python3[2476]: 2023-10-13 11:37:42 ERROR    Exceptio                                                                                                                     n at GetHoymilesActualPower
Okt 13 11:37:42 raspberrypi python3[2476]: 2023-10-13 11:37:42 ERROR    HTTPSCon                                                                                                                     nectionPool(host='192.169.178.93', port=443): Max retries exceeded with url: /st                                                                                                                     atus (Caused by SSLError(CertificateError("hostname '192.169.178.93' doesn't mat                                                                                                                     ch either of 'fbfanimation.com', 'www.fbfanimation.com'")))
Okt 13 11:38:02 raspberrypi python3[2476]: 2023-10-13 11:38:02 INFO     ---Start Zero Export---                                                                                                      Okt 13 11:38:03 raspberrypi python3[2476]: 2023-10-13 11:38:03 INFO     Ahoy: Inverter "Balkonsolar" reachable: True                                                                                 Okt 13 11:38:03 raspberrypi python3[2476]: 2023-10-13 11:38:03 INFO     powermeter Shelly 3EM: 234  Watt
Okt 13 11:38:03 raspberrypi python3[2476]: 2023-10-13 11:38:03 INFO     setting new limit to 1500 Watt
Okt 13 11:38:03 raspberrypi python3[2476]: 2023-10-13 11:38:03 INFO     Ahoy: Inverter "Balkonsolar": setting new limit from 75 Watt to 1500 Watt
Okt 13 11:38:04 raspberrypi python3[2476]: 2023-10-13 11:38:04 INFO     Ahoy: Inverter "Balkonsolar": Limit acknowledged
[... cut ...]
Okt 13 11:40:06 raspberrypi python3[2476]: 2023-10-13 11:40:06 INFO     Inverterlimit already at 1500 Watt
Okt 13 11:40:21 raspberrypi python3[2476]: 2023-10-13 11:40:21 INFO     Ahoy: Inverter "Balkonsolar" reachable: True
Okt 13 11:40:21 raspberrypi python3[2476]: 2023-10-13 11:40:21 INFO     powermeter Shelly 3EM: 153  Watt
Okt 13 11:40:21 raspberrypi python3[2476]: 2023-10-13 11:40:21 INFO     Inverterlimit already at 1500 Watt

Eine Frage habe ich noch. Startet die Software morgends und nach dem reboot automatisch?

Danke für eure Hilfe.

VG

reserve85, 2023-10-13:
nene, das passt noch nicht, da kommt noch ein Zertifikatsfehler bei der Abfrage von Ahoy? Da bin ich allerdings total überfragt.
```
Okt 13 11:37:42 raspberrypi python3[2476]: 2023-10-13 11:37:42 WARNING Certificate did not match expected hostname: 192.169.178.93. Certificate: {'subject': (( ('commonName', 'fbfanimation.com'),),), 'issuer': ((('countryName', 'GB'),), ((' stateOrProvinceName', 'Greater Manchester'),), (('localityName', 'Salford'),), ( ('organizationName', 'Sectigo Limited'),), (('commonName', 'Sectigo RSA Domain V alidation Secure Server CA'),)), 'version': 3, 'serialNumber': 'CE37276CC1AB86E1 5D58013D6082DA60', 'notBefore': 'Oct 17 00:00:00 2022 GMT', 'notAfter': 'Nov 17 23:59:59 2023 GMT', 'subjectAltName': (('DNS', 'fbfanimation.com'), ('DNS', 'www .fbfanimation.com')), 'OCSP': ('http://ocsp.sectigo.com/',), 'caIssuers': ('http: //crt.sectigo.com/SectigoRSADomainValidationSecureServerCA.crt',)}
Okt 13 11:37:42 raspberrypi python3[2476]: 2023-10-13 11:37:42 ERROR Exception at GetHoymilesActualPower
Okt 13 11:37:42 raspberrypi python3[2476]: 2023-10-13 11:37:42 ERROR HTTPSConnectionPool(host='192.169.178.93', port=443): Max retries exceeded with url: /st atus (Caused by SSLError(CertificateError("hostname '192.169.178.93' doesn't mat ch either of 'fbfanimation.com', 'www.fbfanimation.com'")))

```

https://stackoverflow.com/questions/28768530/certificateerror-hostname-doesnt-match

Boemelmann, 2023-10-13:
> nene, die ganz, die nicht, da zu ein beamtenfehler bei der Abfrage von Ahoy? Da bin ich total überfragt.
> 
> ```
> Okt 13 11:37:42 raspberrypi python3[2476]: 2023-10-13 11:37:42 WARNING Certificate did not match expected hostname: 192.169.178.93. Certificate: {'subject': (( ('commonName', 'fbfanimation.com'),),), 'issuer': ((('countryName', 'GB'),), ((' stateOrProvinceName', 'Greater Manchester'),), (('localityName', 'Salford'),), ( ('organizationName', 'Sectigo Limited'),), (('commonName', 'Sectigo RSA Domain V alidation Secure Server CA'),)), 'version': 3, 'serialNumber': 'CE37276CC1AB86E1 5D58013D6082DA60', 'notBefore': 'Oct 17 00:00:00 2022 GMT', 'notAfter': 'Nov 17 23:59:59 2023 GMT', 'subjectAltName': (('DNS', 'fbfanimation.com'), ('DNS', 'www .fbfanimation.com')), 'OCSP': ('http://ocsp.sectigo.com/',), 'caIssuers': ('http: //crt.sectigo.com/SectigoRSADomainValidationSecureServerCA.crt',)}
> Okt 13 11:37:42 raspberrypi python3[2476]: 2023-10-13 11:37:42 ERROR Exception at GetHoymilesActualPower
> Okt 13 11:37:42 raspberrypi python3[2476]: 2023-10-13 11:37:42 ERROR HTTPSConnectionPool(host='192.169.178.93', port=443): Max retries exceeded with url: /st atus (Caused by SSLError(CertificateError("hostname '192.169.178.93' doesn't mat ch either of 'fbfanimation.com', 'www.fbfanimation.com'")))
> ```
> 
> https://stackoverflow.com/questions/28768530/certificateerror-hostname-doesnt-match

jetzt aber, war mein Fehler. Ich hatte in der .ini unter [SELECT_INTERMEDIATE_METER] auch die Anmeldedaten vom SHELLY_3EM hinterlegt.
Gibt es noch die Möglichgeit, dass das Skript automatisch nach dem Neutstart startet?

Vielen Dank noch einmal!

reserve85, 2023-10-13:
Ja, du musst das Scipt als Service eintragen:

```
cat << EOF | tee /etc/systemd/system/HoymilesZeroExport.service
[Unit]
Description=HoymilesZeroExport Service
After=multi-user.target
[Service]
Type=simple
Restart=always
ExecStart=/usr/bin/python3 ${SCRIPT_DIR}/HoymilesZeroExport.py
[Install]
WantedBy=multi-user.target
EOF

systemctl daemon-reload
systemctl enable HoymilesZeroExport.service
systemctl start HoymilesZeroExport.service
systemctl status HoymilesZeroExport.service
```

${SCRIPT_DIR} musst du mit deinem Dir ersetzen.

closed 2023-10-29 by reserve85