reserve85/HoymilesZeroExport #108: 2 Inverter in Reihe 1500Hm und 600Hm

Mucki76, 2023-10-26:
Hallo zusammen
erstmal bedanke ich mich für die Tolle Arbeit.
ich habe das jetzige Skript am Laufen mit ahoyDTU Shelly 3em pro  allerdings kann ich den zweiten Inverter nicht zum Laufen über das Skript.
vielleicht hat einer von euch die gleiche Konstellation mit 2 Inverten und könnte mir das Skript mit veränderten Daten Zeigen.
gruß
Darius

Mucki76, 2023-10-26:
Der 1500Hm läuft uber das Skript und der 600Hm nicht

reserve85, 2023-10-26:
Dann hänge mal die config hier an und die Fehlermeldung.

Mucki76, 2023-10-26:
[VERSION]
VERSION = 1.60

[SELECT_DTU]
# --- define your DTU (only one) ---
USE_AHOY = true
USE_OPENDTU = false

[SELECT_POWERMETER]
# --- define your Powermeter (only one) ---
USE_TASMOTA = false
USE_SHELLY_EM = false
USE_SHELLY_3EM = false
USE_SHELLY_3EM_PRO = true
USE_SHRDZM = false
USE_EMLOG = false
USE_IOBROKER = false
USE_HOMEASSISTANT = false
USE_VZLOGGER = false

[AHOY_DTU]
# --- defines for AHOY-DTU ---
# in settings/inverter set interval to 6 seconds!
AHOY_IP = 192.168.178.91

[OPEN_DTU]
# --- defines for OPEN-DTU ---
OPENDTU_IP = xxx.xxx.xxx.xxx
OPENDTU_USER = 
OPENDTU_PASS = 

[TASMOTA]
# --- defines for Tasmota Smartmeter Modul---
TASMOTA_IP = 192.168.10.90
# the following three constants describes how to navigate through the Tasmota-JSON. 
# if you do not know the format of your Tasmota reader, open a browser and put in the following string replacing xxx with the IP address of your Tasmota device: http://xxx.xxx.xxx.xxx/cm?cmnd=status%2010
# e.g. JSON_Result = {"StatusSNS":{"Time":"2023-02-28T12:49:49","SML":{"total_kwh":15011.575,"curr_w":-71}}}
TASMOTA_JSON_STATUS = StatusSNS
# Prefix for Web UI and MQTT JSON payload
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = SML
# Power-MQTT label (the current power in Watt, positive (import more energy than export) or negative (export more energy than import)).
TASMOTA_JSON_POWER_MQTT_LABEL = curr_w
# if your powermeter does NOT output the current power: you need to calculate it -> Power(W) = OBIS(1.7.0) - OBIS(2.7.0)
TASMOTA_JSON_POWER_CALCULATE = FALSE
# Power-MQTT Input label (positive active instantaneous power, e.g. OBIS Code 1.7.0)
TASMOTA_JSON_POWER_INPUT_MQTT_LABEL =
# Power-MQTT output label (negative active instantaneous power, e.g. OBIS Code 2.7.0)
TASMOTA_JSON_POWER_OUTPUT_MQTT_LABEL =

[SHELLY]
# --- defines for Shelly ---
SHELLY_IP = 192.168.178.93
# if you restricted the web login enter your username and password, for Shelly "Generation 2" devices the username is always "admin" (e.g. Shelly pro 3EM)
SHELLY_USER =
SHELLY_PASS =

[SHRDZM]
# --- defines for SHRDZM Smartmeter Modul ---
SHRDZM_IP = xxx.xxx.xxx.xxx
SHRDZM_USER =
SHRDZM_PASS =

[EMLOG]
# --- defines for EMLOG (electronic meter log) System ---
EMLOG_IP = xxx.xxx.xxx.xxx
EMLOG_METERINDEX = 

[IOBROKER]
# --- defines for IOBROKER (make sure you installed https://github.com/ioBroker/ioBroker.simple-api) ---
IOBROKER_IP = xxx.xxx.xxx.xxx
[... 187 lines cut ...]
# voltage to limit the inverter to a maximum of HOY_BATTERY_NORMAL_WATT
HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 48.5
# maximum limit in watts when battery is high (above HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V)
HOY_BATTERY_NORMAL_WATT = 1500
# maximum limit in watts when battery is low (below HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V)
HOY_BATTERY_REDUCE_WATT = 300
# reenable inverter at this voltage
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 51
# define which panels you want to ignore for the panel voltage. E.g. if you have input 1 + 2 battery powered and input 3 + 4 for solar panels you can disable panel 3 and 4.
# if you want to ignore Panel 3 and Panel 4 type: "HOY_BATTERY_IGNORE_PANELS = 3,4". leave it empty for no ignore.
HOY_BATTERY_IGNORE_PANELS = 
# define prioriry of the inverters (possible values: 1 (high) ... 5 (low); default = 1). same priorities are also possible. Not supported in mixed modes (a mix of solar and battery powered inverters)
# example 2 (default):
# inverter 2 = 300W, priority = 1 // inverter 2 = 150W, priority = 1: 
# set limit of 400W -> inverter 1 is set to 300W and inverter 2 is set to 367W
# set limit of 300W -> inverter 1 is set to 200W and inverter 2 is set to 100W
# example 2:
# inverter 2 = 300W, priority = 1 // inverter 2 = 500W, priority = 2: 
# set limit of 400W -> inverter 1 is set to 300W and inverter 2 is set to 100W
# set limit of 300W -> inverter 1 is set to 200W and inverter 2 is powered off
HOY_BATTERY_PRIORITY = 1

# [INVERTER_x]
# ... copy from [INVERTER_1] and modify
# ...and so on...

Mucki76, 2023-10-26:
[COMMON]
INVERTER_COUNT = 1
SLOW_APPROX_LIMIT_IN_PERCENT = 25
SLOW_APPROX_FACTOR_IN_PERCENT = 20
LOOP_INTERVAL_IN_SECONDS = 10
SET_LIMIT_DELAY_IN_SECONDS = 3
SET_LIMIT_DELAY_IN_SECONDS_MULTIPLE_INVERTER = 2
POLL_INTERVAL_IN_SECONDS = 1
JUMP_TO_MAX_LIMIT_ON_GRID_USAGE = true
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 15
ENABLE_LOG_TO_FILE = false
LOG_BACKUP_COUNT = 30
SET_LIMIT_RETRY = -1

[CONTROL]
POWERMETER_TARGET_POINT = -75
POWERMETER_TOLERANCE = 5
POWERMETER_MAX_POINT = 0

Mucki76, 2023-10-26:
# List of INVERTERS, based on COMMON/COUNT
[INVERTER_1]
# power rating of your inverter
HOY_MAX_WATT = 1500
# minimum limit in percent, e.g. 5%
HOY_MIN_WATT_IN_PERCENT = 5
# factor to multiply before set Limit. Some Inverters have some offsets, with that factor you can compensate it. Default = 1
HOY_COMPENSATE_WATT_FACTOR = 1
# battery powered?
HOY_BATTERY_MODE = false
# voltage to turn off the inverter
HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V = 47
# voltage to limit the inverter to a maximum of HOY_BATTERY_REDUCE_WATT
HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V = 48
# voltage to limit the inverter to a maximum of HOY_BATTERY_NORMAL_WATT
HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 48.5
# maximum limit in watts when battery is high (above HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V)
HOY_BATTERY_NORMAL_WATT = 1500
# maximum limit in watts when battery is low (below HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V)
HOY_BATTERY_REDUCE_WATT = 300
# reenable inverter at this voltage
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 51
# define which panels you want to ignore for the panel voltage. E.g. if you have input 1 + 2 battery powered and input 3 + 4 for solar panels you can disable panel 3 and 4.
# if you want to ignore Panel 3 and Panel 4 type: "HOY_BATTERY_IGNORE_PANELS = 3,4". leave it empty for no ignore.
HOY_BATTERY_IGNORE_PANELS = 
# define prioriry of the inverters (possible values: 1 (high) ... 5 (low); default = 1). same priorities are also possible. Not supported in mixed modes (a mix of solar and battery powered inverters)
# example 1 (default):
# inverter 1 = 1000W, priority = 1 // inverter 2 = 500W, priority = 1: 
# set limit of 1100W -> inverter 1 is set to 733W and inverter 2 is set to 367W
# set limit of 300W -> inverter 1 is set to 200W and inverter 2 is set to 100W
# example 2:
# inverter 1 = 1000W, priority = 1 // inverter 2 = 500W, priority = 2: 
# set limit of 1100W -> inverter 1 is set to 1000W and inverter 2 is set to 100W
# set limit of 300W -> inverter 1 is set to 300W and inverter 2 is powered off
HOY_BATTERY_PRIORITY = 1

# List of INVERTERS, based on COMMON/COUNT
# [INVERTER_2]
# power rating of your inverter
HOY_MAX_WATT = 600
# minimum limit in percent, e.g. 5%
HOY_MIN_WATT_IN_PERCENT = 5
# factor to multiply before set Limit. Some Inverters have some offsets, with that factor you can compensate it. Default = 1
HOY_COMPENSATE_WATT_FACTOR = 1
# battery powered?
HOY_BATTERY_MODE = false
# voltage to turn off the inverter
HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V = 47
# voltage to limit the inverter to a maximum of HOY_BATTERY_REDUCE_WATT
HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V = 48
# voltage to limit the inverter to a maximum of HOY_BATTERY_NORMAL_WATT
HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 48.5
# maximum limit in watts when battery is high (above HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V)
HOY_BATTERY_NORMAL_WATT = 1500
# maximum limit in watts when battery is low (below HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V)
HOY_BATTERY_REDUCE_WATT = 300
# reenable inverter at this voltage
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 51
# define which panels you want to ignore for the panel voltage. E.g. if you have input 1 + 2 battery powered and input 3 + 4 for solar panels you can disable panel 3 and 4.
# if you want to ignore Panel 3 and Panel 4 type: "HOY_BATTERY_IGNORE_PANELS = 3,4". leave it empty for no ignore.
HOY_BATTERY_IGNORE_PANELS = 
# define prioriry of the inverters (possible values: 1 (high) ... 5 (low); default = 1). same priorities are also possible. Not supported in mixed modes (a mix of solar and battery powered inverters)
# example 1 (default):
# inverter 1 = 1000W, priority = 1 // inverter 2 = 500W, priority = 1: 
# set limit of 1100W -> inverter 1 is set to 733W and inverter 2 is set to 367W
# set limit of 300W -> inverter 1 is set to 200W and inverter 2 is set to 100W
# example 2:
# inverter 1 = 1000W, priority = 1 // inverter 2 = 500W, priority = 2: 
# set limit of 1100W -> inverter 1 is set to 1000W and inverter 2 is set to 100W
# set limit of 300W -> inverter 1 is set to 300W and inverter 2 is powered off
HOY_BATTERY_PRIORITY = 1

# [INVERTER_x]
# ... copy from [INVERTER_1] and modify
# ...and so on...

Mucki76, 2023-10-26:
# List of INVERTERS, based on COMMON/COUNT
[INVERTER_1]
# power rating of your inverter
HOY_MAX_WATT = 1500
# minimum limit in percent, e.g. 5%
HOY_MIN_WATT_IN_PERCENT = 5
# factor to multiply before set Limit. Some Inverters have some offsets, with that factor you can compensate it. Default = 1
HOY_COMPENSATE_WATT_FACTOR = 1
# battery powered?
HOY_BATTERY_MODE = false
# voltage to turn off the inverter
HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V = 47
# voltage to limit the inverter to a maximum of HOY_BATTERY_REDUCE_WATT
HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V = 48
# voltage to limit the inverter to a maximum of HOY_BATTERY_NORMAL_WATT
HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 48.5
# maximum limit in watts when battery is high (above HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V)
HOY_BATTERY_NORMAL_WATT = 1500
# maximum limit in watts when battery is low (below HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V)
HOY_BATTERY_REDUCE_WATT = 300
# reenable inverter at this voltage
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 51
# define which panels you want to ignore for the panel voltage. E.g. if you have input 1 + 2 battery powered and input 3 + 4 for solar panels you can disable panel 3 and 4.
# if you want to ignore Panel 3 and Panel 4 type: "HOY_BATTERY_IGNORE_PANELS = 3,4". leave it empty for no ignore.
HOY_BATTERY_IGNORE_PANELS = 
# define prioriry of the inverters (possible values: 1 (high) ... 5 (low); default = 1). same priorities are also possible. Not supported in mixed modes (a mix of solar and battery powered inverters)
# example 1 (default):
# inverter 1 = 1000W, priority = 1 // inverter 2 = 500W, priority = 1: 
# set limit of 1100W -> inverter 1 is set to 733W and inverter 2 is set to 367W
# set limit of 300W -> inverter 1 is set to 200W and inverter 2 is set to 100W
# example 2:
# inverter 1 = 1000W, priority = 1 // inverter 2 = 500W, priority = 2: 
# set limit of 1100W -> inverter 1 is set to 1000W and inverter 2 is set to 100W
# set limit of 300W -> inverter 1 is set to 300W and inverter 2 is powered off
HOY_BATTERY_PRIORITY = 1

# List of INVERTERS, based on COMMON/COUNT
# [INVERTER_2]
# power rating of your inverter
HOY_MAX_WATT = 600
# minimum limit in percent, e.g. 5%
HOY_MIN_WATT_IN_PERCENT = 5
# factor to multiply before set Limit. Some Inverters have some offsets, with that factor you can compensate it. Default = 1
HOY_COMPENSATE_WATT_FACTOR = 1
# battery powered?
HOY_BATTERY_MODE = false
# voltage to turn off the inverter
HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V = 47
# voltage to limit the inverter to a maximum of HOY_BATTERY_REDUCE_WATT
HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V = 48
# voltage to limit the inverter to a maximum of HOY_BATTERY_NORMAL_WATT
HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 48.5
# maximum limit in watts when battery is high (above HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V)
HOY_BATTERY_NORMAL_WATT = 1500
# maximum limit in watts when battery is low (below HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V)
HOY_BATTERY_REDUCE_WATT = 300
# reenable inverter at this voltage
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 51
# define which panels you want to ignore for the panel voltage. E.g. if you have input 1 + 2 battery powered and input 3 + 4 for solar panels you can disable panel 3 and 4.
# if you want to ignore Panel 3 and Panel 4 type: "HOY_BATTERY_IGNORE_PANELS = 3,4". leave it empty for no ignore.
HOY_BATTERY_IGNORE_PANELS = 
# define prioriry of the inverters (possible values: 1 (high) ... 5 (low); default = 1). same priorities are also possible. Not supported in mixed modes (a mix of solar and battery powered inverters)
# example 1 (default):
# inverter 1 = 1000W, priority = 1 // inverter 2 = 500W, priority = 1: 
# set limit of 1100W -> inverter 1 is set to 733W and inverter 2 is set to 367W
# set limit of 300W -> inverter 1 is set to 200W and inverter 2 is set to 100W
# example 2:
# inverter 1 = 1000W, priority = 1 // inverter 2 = 500W, priority = 2: 
# set limit of 1100W -> inverter 1 is set to 1000W and inverter 2 is set to 100W
# set limit of 300W -> inverter 1 is set to 300W and inverter 2 is powered off
HOY_BATTERY_PRIORITY = 1

# [INVERTER_x]
# ... copy from [INVERTER_1] and Modi
# ...and so on...

reserve85, 2023-10-26:
Du musst
`INVERTER_COUNT = 2`
einstellen.

Mucki76, 2023-10-26:
Ich danke dir für deine schnelle Hilfe ,auf den ersten Blick funktioniert das jetzt .Mal schauen wenn die Sonne scheint. Und Hut ab für die Arbeit von dir.
Gruß  Darius

closed 2023-12-01 by reserve85