reserve85/HoymilesZeroExport #111: Fehlermeldung

fatzo, 2023-12-06:
Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Dez 05 18:16:51 raspberrypi python3[1754]: Traceback (most recent call last):
Dez 05 18:16:51 raspberrypi python3[1754]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 32, in <module>
Dez 05 18:16:51 raspberrypi python3[1754]:     from packaging import version
Dez 05 18:16:51 raspberrypi python3[1754]: ModuleNotFoundError: No module named 'packaging'
Moin
habe den raspberry neu aufgesetzt
nach der anpassung habe ich nun diese fehlermeldung, vorher lief der sript ohne Probleme

fatzo

reserve85, 2023-12-06:
Moin,
```pip3 install -r requirements.txt```
sollte alle notwendigen Pakete installieren. 
Alternativ:
```pip3 install packaging```

fatzo, 2023-12-06:
neue Fehlermeldung :-(
Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Dez 06 09:08:16 raspberrypi python3[1846]: 2023-12-06 09:08:16 INFO     Error on reading ENABLE_LOG_TO_FILE, set it to DISABLED
Dez 06 09:08:16 raspberrypi python3[1846]: 2023-12-06 09:08:16 ERROR    While reading from '/home/pi/HoymilesZeroExport/HoymilesZeroExport_Config.ini' [line 267]: option 'hoy_max_watt' in section 'INVERTER_1' already exists
Dez 06 09:08:16 raspberrypi python3[1846]: 2023-12-06 09:08:16 INFO     Log write to file: False
Dez 06 09:08:16 raspberrypi python3[1846]: 2023-12-06 09:08:16 INFO     Python Version: 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0]
Dez 06 09:08:16 raspberrypi python3[1846]: 2023-12-06 09:08:16 INFO     Author: Tobias Kraft / Script Version: 1.62
Dez 06 09:08:16 raspberrypi python3[1846]: 2023-12-06 09:08:16 INFO     read config file: /home/pi/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Dez 06 09:08:16 raspberrypi python3[1846]: 2023-12-06 09:08:16 INFO     read additional config file: /home/pi/HoymilesZeroExport/HoymilesZeroExport_Config_Override.ini
Dez 06 09:08:16 raspberrypi python3[1846]: Traceback (most recent call last):
Dez 06 09:08:16 raspberrypi python3[1846]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 937, in <module>
Dez 06 09:08:16 raspberrypi python3[1846]:     VERSION = config.get('VERSION', 'VERSION')
Dez 06 09:08:16 raspberrypi python3[1846]:               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Dez 06 09:08:16 raspberrypi python3[1846]:   File "/usr/lib/python3.11/configparser.py", line 815, in get
Dez 06 09:08:16 raspberrypi python3[1846]:     return self._interpolation.before_get(self, section, option, value,
Dez 06 09:08:16 raspberrypi python3[1846]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Dez 06 09:08:16 raspberrypi python3[1846]:   File "/usr/lib/python3.11/configparser.py", line 396, in before_get
Dez 06 09:08:16 raspberrypi python3[1846]:     self._interpolate_some(parser, option, L, value, section, defaults, 1)
Dez 06 09:08:16 raspberrypi python3[1846]:   File "/usr/lib/python3.11/configparser.py", line 413, in _interpolate_some
Dez 06 09:08:16 raspberrypi python3[1846]:     p = rest.find("%")
Dez 06 09:08:16 raspberrypi python3[1846]:         ^^^^^^^^^
Dez 06 09:08:16 raspberrypi python3[1846]: AttributeError: 'list' object has no attribute 'find'
Dez 06 09:08:16 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Dez 06 09:08:16 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.

reserve85, 2023-12-06:
da stimmt vermutlich was in der config nicht. Hänge am besten die config hier mal als Datei an.

fatzo, 2023-12-06:
Habs zum laufen bekommen. war ein zahlendreher noch in der config

Vielen dank für die schnelle Antwort.

fatzo

closed 2023-12-06 by fatzo