reserve85/HoymilesZeroExport #112: Keine Verbindung zur DTU möglich

HippiePaul, 2023-12-10:
Hallo zusammen,

ich habe das Problem, dass das Script keine Verbindung zur DTU aufbauen kann, da es scheinbar ein Netzwerkproblem gibt. Ich habe schon einiges versucht, aber nichts davon brachte die Lösung, evtl. kann mir hier geholfen werden.

**Konstellation:**
 - IO Broker auf Raspberry Pi 4
 - HoymilesZero Export auf dem gleichen Raspberry Pi 4 wie IO Broker
 - Hoymiles HM600
 - Ahoy DTU 
     - mit WR verbunden, per Browser ansprechbar, Werte via MQTT im IO Broker vorhanden, IP 192.168.178.80
 - Hitchi IR Wifi Lesekopf für digitalen Zähler 
     - TASMOTA, per Browser ansprechbar, Werte via Sonoff Adapter im IO Broker vorhanden, IP 192.168.178.82

**Konfiguration (...Config_Override.ini):**

[SELECT_DTU]
USE_AHOY = true

[SELECT_POWERMETER]
USE_TASMOTA = true

[AHOY_DTU]
AHOY_IP = 192.168.168.80

[TASMOTA]
TASMOTA_IP = 192.168.168.82
TASMOTA_JSON_POWER_MQTT_LABEL = Watt_Summe

[COMMON]
POLL_INTERVAL_IN_SECONDS = 10

[INVERTER_1]
HOY_MAX_WATT = 600

**Log HoymilesZeroExport:**
(Am Ende habe ich die DTU angepingt um zu zeigen, dass die Verbindung grundsätzlich verfügbar wäre)

Dez 10 11:49:53 iobroker systemd[1]: Started HoymilesZeroExport Service.
Dez 10 11:49:53 iobroker python3[1866]: 2023-12-10 11:49:53 INFO     Log write to file: False
Dez 10 11:49:53 iobroker python3[1866]: 2023-12-10 11:49:53 INFO     Python Version: 3.9.2 (default, Feb 28 2021, 17:03:44)
Dez 10 11:49:53 iobroker python3[1866]: [GCC 10.2.1 20210110]
Dez 10 11:49:53 iobroker python3[1866]: 2023-12-10 11:49:53 INFO     Author: Tobias Kraft / Script Version: 1.62
Dez 10 11:49:53 iobroker python3[1866]: 2023-12-10 11:49:53 INFO     read config file: /home/hippie_paul/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Dez 10 11:49:53 iobroker python3[1866]: 2023-12-10 11:49:53 INFO     read additional config file: /home/hippie_paul/HoymilesZeroExport/HoymilesZeroExport_Config_Override.ini
Dez 10 11:49:53 iobroker python3[1866]: 2023-12-10 11:49:53 INFO     Config file V 1.62
Dez 10 11:49:53 iobroker python3[1866]: 2023-12-10 11:49:53 INFO     ---Init---
Dez 10 11:50:03 iobroker python3[1866]: 2023-12-10 11:50:03 ERROR    HTTPConnectionPool(host='192.168.168.80', port=80): Max retries exceeded with url: /api/system (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7fa550a250>, 'Connection to 192.168.168.80 timed out. (connect timeout=10)'))
Dez 10 11:50:23 iobroker python3[1866]: 2023-12-10 11:50:23 INFO     ---Start Zero Export---
Dez 10 11:50:33 iobroker python3[1866]: 2023-12-10 11:50:33 ERROR    Exception at GetHoymilesAvailable, Inverter 0 (yet unknown) not reachable
Dez 10 11:50:33 iobroker python3[1866]: 2023-12-10 11:50:33 ERROR    HTTPConnectionPool(host='192.168.168.80', port=80): Max retries exceeded with url: /api/index (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7fa550ad30>, 'Connection to 192.168.168.80 timed out. (connect timeout=10)'))
Dez 10 11:51:03 iobroker python3[1866]: 2023-12-10 11:51:03 ERROR    Exception at GetHoymilesAvailable, Inverter 0 (yet unknown) not reachable
Dez 10 11:51:03 iobroker python3[1866]: 2023-12-10 11:51:03 ERROR    HTTPConnectionPool(host='192.168.168.80', port=80): Max retries exceeded with url: /api/index (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7fa54ab700>, 'Connection to 192.168.168.80 timed out. (connect timeout=10)'))
^C
hippie_paul@iobroker:~/HoymilesZeroExport $ ping 192.168.178.80
PING 192.168.178.80 (192.168.178.80) 56(84) bytes of data.
64 bytes from 192.168.178.80: icmp_seq=1 ttl=255 time=4.73 ms
64 bytes from 192.168.178.80: icmp_seq=2 ttl=255 time=4.04 ms
64 bytes from 192.168.178.80: icmp_seq=3 ttl=255 time=8.50 ms
64 bytes from 192.168.178.80: icmp_seq=4 ttl=255 time=4.50 ms
64 bytes from 192.168.178.80: icmp_seq=5 ttl=255 time=23.2 ms
^C
--- 192.168.178.80 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4006ms
rtt min/avg/max/mdev = 4.035/8.996/23.215/7.286 ms
hippie_paul@iobroker:~/HoymilesZeroExport $

[image]

Bisher erfolglos versuchte Maßnahmen:

 - Restart Service
 - Restart Pi
 - Deinstallation mit anschließender Reinstallation des Service
 - Neustart AHOY
 - Update AHOY auf neueste Version
 - Check der installierten Packages siehe Log:

**Packages Log:**

hippie_paul@iobroker:~/HoymilesZeroExport $ pip3 install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: certifi==2023.11.17 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 1)) (2023.11.17)
Requirement already satisfied: charset-normalizer==3.3.2 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 2)) (3.3.2)
Requirement already satisfied: idna==3.4 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 3)) (3.4)
Requirement already satisfied: packaging==23.2 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 4)) (23.2)
Requirement already satisfied: requests==2.31.0 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 5)) (2.31.0)
Requirement already satisfied: urllib3==2.1.0 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 6)) (2.1.0)

Kann mir evlt. jemand bei meinem Problem helfen ?

Grüße

reserve85, 2023-12-10:
ja, ich glaube du hast die falsche IP Adresse eingetippt:

```
[AHOY_DTU]
AHOY_IP = 192.168.178.80
```

HippiePaul, 2023-12-10:
Vielen Dank, da war ich wohl betriebsblind :-)
Vielen Dank auch für Deine schnelle Antwort und den Support

reserve85, 2023-12-11:
Kein Ding, gib mal Bescheid ob alles geht.

HippiePaul, 2023-12-12:
Sorry für die späte Rückmeldung, ich war gestern bissl viel beschäftigt :-)
Nach der Änderung der IP geht alles wunderbar, allerdings ist gerade keine Sonne da, so dass nix geregelt werden muss und das Power Limit immer bei den vollen 600 W bleibt.

Vielen Dank für Deine Arbeit und Deinen Support, das ist echt ein cooles Script :-)

Wenn mal die Sonne scheint schau ich ob es auch mit der Regelung klappt....

hier noch die letzten Zeilen des aktuellen Logs:
(man erkennt deutlich die Heizstäbe des Geschirrspülers)

2023-12-12 12:22:51,827 INFO     powermeter Tasmota: 457  Watt
2023-12-12 12:22:51,827 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:23:06,930 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:23:06,979 INFO     powermeter Tasmota: 490  Watt
2023-12-12 12:23:06,979 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:23:22,087 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:23:22,138 INFO     powermeter Tasmota: 496  Watt
2023-12-12 12:23:22,138 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:23:37,202 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:23:37,295 INFO     powermeter Tasmota: 430  Watt
2023-12-12 12:23:37,295 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:23:52,398 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:23:52,494 INFO     powermeter Tasmota: 442  Watt
2023-12-12 12:23:52,494 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:24:07,554 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:24:07,648 INFO     powermeter Tasmota: 426  Watt
2023-12-12 12:24:07,648 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:24:22,716 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:24:22,791 INFO     powermeter Tasmota: 437  Watt
2023-12-12 12:24:22,792 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:24:37,864 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:24:37,910 INFO     powermeter Tasmota: 439  Watt
2023-12-12 12:24:37,910 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:24:53,033 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:24:53,086 INFO     powermeter Tasmota: 431  Watt
2023-12-12 12:24:53,086 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:25:08,176 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:25:08,246 INFO     powermeter Tasmota: 437  Watt
2023-12-12 12:25:08,247 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:25:23,302 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:25:23,398 INFO     powermeter Tasmota: 420  Watt
2023-12-12 12:25:23,399 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:25:38,484 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:25:38,560 INFO     powermeter Tasmota: 443  Watt
2023-12-12 12:25:38,560 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:25:53,607 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:25:53,659 INFO     powermeter Tasmota: 445  Watt
2023-12-12 12:25:53,660 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:26:08,796 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:26:08,845 INFO     powermeter Tasmota: 460  Watt
2023-12-12 12:26:08,846 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:26:23,952 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:26:23,999 INFO     powermeter Tasmota: 557  Watt
2023-12-12 12:26:24,000 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:26:39,110 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:26:39,198 INFO     powermeter Tasmota: 473  Watt
2023-12-12 12:26:39,198 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:26:54,260 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:26:54,348 INFO     powermeter Tasmota: 493  Watt
2023-12-12 12:26:54,348 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:27:09,417 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:27:09,499 INFO     powermeter Tasmota: 440  Watt
2023-12-12 12:27:09,499 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:27:24,550 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:27:24,601 INFO     powermeter Tasmota: 501  Watt
2023-12-12 12:27:24,601 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:27:39,659 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:27:39,710 INFO     powermeter Tasmota: 479  Watt
2023-12-12 12:27:39,710 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:27:54,754 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
[... 68 lines cut ...]
2023-12-12 12:33:43,357 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:33:43,409 INFO     powermeter Tasmota: 1964  Watt
2023-12-12 12:33:43,410 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:33:58,512 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:33:58,582 INFO     powermeter Tasmota: 1970  Watt
2023-12-12 12:33:58,583 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:34:13,633 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:34:13,684 INFO     powermeter Tasmota: 1959  Watt
2023-12-12 12:34:13,685 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:34:28,823 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
[... cut ...]
2023-12-12 12:34:43,998 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:34:59,142 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:34:59,208 INFO     powermeter Tasmota: 1994  Watt
2023-12-12 12:34:59,209 INFO     Inverterlimit already at 600 Watt
2023-12-12 12:35:14,295 INFO     Ahoy: Inverter "Hoymiles_HM600" reachable: True
2023-12-12 12:35:14,359 INFO     powermeter Tasmota: 2005  Watt
2023-12-12 12:35:14,360 INFO     Inverterlimit already at 600 Watt

Grüße 

Hippie_Paul

closed 2024-01-02 by reserve85