reserve85/HoymilesZeroExport #114: OpenDTU: Inverter "yet unknown" reachable: False

dadibsy, 2023-12-20:
Hallo,

ich habe das Script erfolgreich installiert. der Wert vom Powermeter wird mir angezeigt, Die Daten der OpenDTU leider nicht. 

Config:

[SELECT_DTU]
USE_OPENDTU = true

[SELECT_POWERMETER]
USE_TASMOTA = true

[OPEN_DTU]
OPENDTU_IP = 192.168.1.80
OPENDTU_USER = admin
OPENDTU_PASS = ********

[TASMOTA]
TASMOTA_IP = 192.168.1.88
TASMOTA_JSON_STATUS = StatusSNS
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = Energie
TASMOTA_JSON_POWER_MQTT_LABEL = Power

[COMMON]
ENABLE_LOG_TO_FILE = true
LOG_BACKUP_COUNT = 30
LOG_TEMPERATURE = true
INVERTER_COUNT = 1

[INVERTER_1]
HOY_MAX_WATT = 1800

Wenn ich http://192.168.1.80/api/livedata/status/inverters/ im Browser aufrufe bekomme ich ein JSON von der DTU zurück. Auch ein "wget http://192.168.1.80/api/livedata/status/inverters/" auf der Bash liefert ein File mit gleichem Inhalt, Daher gehe ich davon aus das die OpenDTU erreichbar ist. Klar der WR produziert gerade nicht aber die Seriennummer müsste doch auslesbar sein. Im Pyton Script sehe ich das 

def GetHoymilesAvailableOpenDTU(pInverterId):
    url = f'http://{OPENDTU_IP}/api/livedata/status/inverters'
    ParsedData = requests.get(url, auth=HTTPBasicAuth(OPENDTU_USER, OPENDTU_PASS), timeout=10).json()
    Reachable = bool(ParsedData["inverters"][pInverterId]["reachable"])
    logger.info('OpenDTU: Inverter "%s" reachable: %s',NAME[pInverterId],Reachable)
    return Reachable
    
Auch nichts anderes macht als das was ich händisch gemacht habe. 

Das JSON was ich erhalte:

{
    "inverters": [
        {
            "serial": "138281xxxxxx",
            "name": "HMT-1800",
            "order": 0,
            "data_age": 22580,
            "poll_enabled": true,
            "reachable": false,
            "producing": false,
            "limit_relative": 100,
            "limit_absolute": 1800,
            "AC": {
                "0": {
                    "Power": {
                        "v": 0,
                        "u": "W",
                        "d": 1
                    },
                    "Voltage": {
                        "v": 0,
                        "u": "V",
                        "d": 1
                    },
                    "Current": {
[... 261 lines cut ...]
        "YieldDay": {
            "v": 0,
            "u": "Wh",
            "d": 0
        },
        "YieldTotal": {
            "v": 0,
            "u": "kWh",
            "d": 3
        }
    },
    "hints": {
        "time_sync": false,
        "radio_problem": false,
        "default_password": false
    }
}

Kann es an einer inkompatiblen Version von OpenDTU liegen? Meine Version  v23.6.1 ist schon etwas älter und ich mache eigentlich keine Updates solange alles so läuft wie ich es mir vorstelle.  

Wo könnte ich ansetzen?

Viele Grüße

Axel

reserve85, 2023-12-20:
Servus, ist ganz Normal. Erst wenn der inverter reachable wird aktualisiert sich auch die Seriennummer im Log. Sollte also alles passen. 

Gruß

dadibsy, 2023-12-20:
Vielen Dank für die schnelle Antwort. Ich warte dann mal was morgen im Log erscheint wenn der WR produziert und werde berichten.

Gruß Axel

dadibsy, 2023-12-21:
Der OpenDTU/WR wurde heute wieder nicht errkannt. Ich habe dann die OpenDTU neu gestartet und die Daten wurden daraufhin vom Script abgerufen. Zur Regelung kann ich aktuell nichts sagen, das Wetter war heute zu schlecht.

Ich habe aber noch eine Frage zu den Logs. Ich habe mir den Python Code angeschaut und wenn ich das richtig sehe wird nirgendwo die aktuelle Produktion des/der WR ausgegeben, oder habe ich etwas übersehen?

Gruß Axel

reserve85, 2023-12-21:
Hi,
nee das wird nicht ausgegeben, ich frage immer nur das ab was ich brauche.
Gruß Tobias

dadibsy, 2023-12-21:
Zu Debugzwecken wäre es meiner Meinung nach gut, wenn man Verbrauch, Limit und Produktion parallel im Log hätte. Ist aber kein Problem die Daten liegen ja auch anderweitig im System vor und ich verzichte dann auf das Log File und Sammel mir die Daten im ioBroker über ein eigenes HelferDebugScript ein.

closed 2023-12-21 by dadibsy