reserve85/HoymilesZeroExport #115: Service startet nicht

PolyVenyl, 2023-12-31:
Leider startet der Service nicht.
Da ich mich nicht mit Python auskenne, hier die Fakten.

Log:

Dez 31 16:17:18 DebianBE systemd[1]: Started HoymilesZeroExport Service.
Dez 31 16:17:18 DebianBE python3[3162334]: 2023-12-31 16:17:18 INFO     Log write to file: False
Dez 31 16:17:18 DebianBE python3[3162334]: 2023-12-31 16:17:18 INFO     Python Version: 3.9.2 (default, Feb 28 2021, 17:03:44)
Dez 31 16:17:18 DebianBE python3[3162334]: [GCC 10.2.1 20210110]
Dez 31 16:17:18 DebianBE python3[3162334]: 2023-12-31 16:17:18 INFO     Author: Tobias Kraft / Script Version: 1.62
Dez 31 16:17:18 DebianBE python3[3162334]: 2023-12-31 16:17:18 INFO     read config file: /ahoy/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Dez 31 16:17:18 DebianBE python3[3162334]: 2023-12-31 16:17:18 INFO     read additional config file: /ahoy/HoymilesZeroExport/HoymilesZeroExport_Config_Override.ini
Dez 31 16:17:18 DebianBE python3[3162334]: 2023-12-31 16:17:18 INFO     Config file V 1.62
Dez 31 16:17:18 DebianBE python3[3162334]: Traceback (most recent call last):
Dez 31 16:17:18 DebianBE python3[3162334]:   File "/usr/lib/python3.9/configparser.py", line 789, in get
Dez 31 16:17:18 DebianBE python3[3162334]:     value = d[option]
Dez 31 16:17:18 DebianBE python3[3162334]:   File "/usr/lib/python3.9/collections/__init__.py", line 941, in __getitem__
Dez 31 16:17:18 DebianBE python3[3162334]:     return self.__missing__(key)            # support subclasses that define __missing__
Dez 31 16:17:18 DebianBE python3[3162334]:   File "/usr/lib/python3.9/collections/__init__.py", line 933, in __missing__
Dez 31 16:17:18 DebianBE python3[3162334]:     raise KeyError(key)
Dez 31 16:17:18 DebianBE python3[3162334]: KeyError: 'hoy_min_watt_in_percent'
Dez 31 16:17:18 DebianBE python3[3162334]: During handling of the above exception, another exception occurred:
Dez 31 16:17:18 DebianBE python3[3162334]: Traceback (most recent call last):
Dez 31 16:17:18 DebianBE python3[3162334]:   File "/ahoy/HoymilesZeroExport/HoymilesZeroExport.py", line 1062, in <module>
Dez 31 16:17:18 DebianBE python3[3162334]:     HOY_MIN_WATT.append(int(HOY_MAX_WATT[i] * config.getint('INVERTER_' + str(i + 1), 'HOY_MIN_WATT_IN_PERCENT') / 100))
Dez 31 16:17:18 DebianBE python3[3162334]:   File "/usr/lib/python3.9/configparser.py", line 818, in getint
Dez 31 16:17:18 DebianBE python3[3162334]:     return self._get_conv(section, option, int, raw=raw, vars=vars,
Dez 31 16:17:18 DebianBE python3[3162334]:   File "/usr/lib/python3.9/configparser.py", line 808, in _get_conv
Dez 31 16:17:18 DebianBE python3[3162334]:     return self._get(section, conv, option, raw=raw, vars=vars,
Dez 31 16:17:18 DebianBE python3[3162334]:   File "/usr/lib/python3.9/configparser.py", line 803, in _get
Dez 31 16:17:18 DebianBE python3[3162334]:     return conv(self.get(section, option, **kwargs))
Dez 31 16:17:18 DebianBE python3[3162334]:   File "/usr/lib/python3.9/configparser.py", line 792, in get
Dez 31 16:17:18 DebianBE python3[3162334]:     raise NoOptionError(option, section)
Dez 31 16:17:18 DebianBE python3[3162334]: configparser.NoOptionError: No option 'hoy_min_watt_in_percent' in section: 'INVERTER_2'
Dez 31 16:17:18 DebianBE systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Dez 31 16:17:18 DebianBE systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Dez 31 16:17:18 DebianBE systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 5.
Dez 31 16:17:18 DebianBE systemd[1]: Stopped HoymilesZeroExport Service.
Dez 31 16:17:18 DebianBE systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly.
Dez 31 16:17:18 DebianBE systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Dez 31 16:17:18 DebianBE systemd[1]: Failed to start HoymilesZeroExport Service.

HoymilesZeroExport_Config_Override.ini:

[SELECT_DTU]
USE_AHOY = true

[SELECT_POWERMETER]
USE_TASMOTA = true

[AHOY_DTU]
AHOY_IP = 192.168.1.121

[TASMOTA]
TASMOTA_IP = 192.168.1.144
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = strom
TASMOTA_JSON_POWER_MQTT_LABEL = curr_pwr

[COMMON]
INVERTER_COUNT = 3

[INVERTER_1]
HOY_MAX_WATT = 800

[INVERTER_2]
HOY_MAX_WATT = 600

[INVERTER_3]
HOY_MAX_WATT = 1500

Sonstiges:

Python 3.9.2 (default, Feb 28 2021, 17:03:44)
[GCC 10.2.1 20210110] on linux

Requirement already satisfied: certifi==2023.11.17 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 1)) (2023.11.17)
Requirement already satisfied: charset-normalizer==3.3.2 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 2)) (3.3.2)
Requirement already satisfied: idna==3.4 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 3)) (3.4)
Requirement already satisfied: packaging==23.2 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 4)) (23.2)
Requirement already satisfied: requests==2.31.0 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 5)) (2.31.0)
Requirement already satisfied: urllib3==2.1.0 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 6)) (2.1.0)

Maßnahmen:
python + HoymilesZeroExport jeweils 2x remove und install, keine änderung.

Danke

philippsandhaus, 2023-12-31:
Im Configfile scheinen einige Einträge zu fehlen konkret beschwert sich das Skript hier gerade über einen fehlenden Eintrag `HOY_MIN_WATT_IN_PERCENT = ?`, wahrscheinlich in der Aktion `[INVERTER_2]`, wenn Du das hinzufügst werden aber weitere Fehlermeldungen über zusätzliche fehlende Einträge kommen. Du benutzt die Variante `HoymilesZeroExport_Config_Override.ini`, d.h. die Default-Werte werden `HoymilesZeroExport_Config.ini` übernommen und können überschrieben werden. Hier gibt es aber nur Default-Were für `[INVERTER_1]`. Sobald du mehr als einen Wechselrichter hast, musst hier alle Parameter definieren.

AN deiner Stelle würde ich die gesamte Sektion für Inverter 1 aus `HoymilesZeroExport_Config.ini`  kopieren und alle Werte anpassen. Das gleiche für die Sektionen `[INVERTER_2]` und `[INVERTER_3]`

PolyVenyl, 2023-12-31:
Ok, jetzt läuft es soweit.
Bin davon ausgegangen, das Defaultwerte angenommen werden, wenn in der
HoymilesZeroExport_Config_Override.ini keine weiteren Werte für weitere Inverter angegeben werden.

Danke für die schnelle Antwort @philippsandhaus 
Mfg PolyVenyl

closed 2024-01-02 by reserve85