reserve85/HoymilesZeroExport #118: Stimmt die Installationsanleitung noch?

jimknopf63, 2024-01-06:
Ich versuche gerade mein Zeroexport auf einem Proxmox LXC Ubuntu 23.10 Container zu installieren. Python3 - 3.11.xx ist vorhanden. Wenn ich die Befehle wie in der Anleitung eingebe wird auch alles installiert. Aber der Service will nicht starten, läuft 5x durch und das war es. Ich habe aus einer alten Version (1.39) die Config in Override kopiert. Die Daten werden auch in die neue Config übernommen. Sehe ich ja wenn ich mit nano aufrufe. Nach dem Speichern  und ./restart.sh läuft der Service nicht. Versucht 5x zu starten.

Ich habe die Version 1.39 auf einem anderen LXC laufen und die funktioniert, leider nur nicht mehr mit der neusten Ahoy Version.  es werden jetzt zu kleine Limits gesetzt. Statt z.B. 87%  nun 8,7%

reserve85, 2024-01-06:
Hi,

dir fehlt das Package "requests" - das kannst du nachinstallieren mit 
`pip3 install requests`

Ich empfehle dir allerdings auf die aktuelle Version zu updaten, da die Version 1.39 nicht mehr mit den neuen Ahoy Versionen läuft.
Du kannst die Install-Anweisung aus der Readme befolgen - du musst danach nur deine spezifischen Einstellungen in die HoymilesZeroExport_Config_Override.ini reinkopieren.

Gruß Tobias

jimknopf63, 2024-01-07:
> Hi,
>
> dir fehlt das Package "requests" - das kannst du nachinstallieren mit `pip3 install requests`
>
> Ich empfehle dir allerdings auf die aktuelle Version zu updaten, da die Version 1.39 nicht mehr [...]

Erst mal vielen Danke für deine Hilfe, es scheint danach zu funktionieren. Ein Problem besteht allerdings noch. Sobald ich den 2ten WR einbinde läuft das Script wieder nicht. Habe das Script vom 1. kopiert und beim 2. Inverser eingefügt. Angepasst und die Anzahl der Inverser bei common auf 2 gesetzt. Gespeichert, neu gestartet und dann läuft es wieder nicht. Danach Daten vom 2 WR gelöscht und das Script läuft wieder. Vllt siehst Du ja was ich verkehrt gemacht habe.

[image]

reserve85, 2024-01-07:
Hängst du bitte die override ini mal an oder schickst sie mir?

jimknopf63, 2024-01-07:
> Hängst du bitte die override ini mal an oder schickst sie mir? update: ich habe die HoymilesZeroExport_Config.ini neu angepasst, damit sollte es jetzt gehen. Müsstest du einfach ersetzen.

Guten Tag, ganz herzlichen Dank für deine Hilfe. Mit der ausgetauschten Config sieht das ganze schon mal besser aus. So ganz scheint es noch nicht zu funktionieren. Die Timeouts hatte ich in der alten Version nicht. Und einfach Klasse wie das Projekt hier umgesetzt wird. Dickes Lob die Alte Version lief seit dem erscheinen ohne Abstürze zuverlässig. Wollte halt jetzt mit der neuen Ahoy Version nach dem Update nicht mehr.

[image]

[HoymilesZeroExport_Config_Override.ini.zip](https://github.com/reserve85/HoymilesZeroExport/files/13853151/HoymilesZeroExport_Config_Override.ini.zip)

reserve85, 2024-01-07:
ok, 
zu allererst: dein Powermeter funktioniert nicht. Da stimmt deine Config nicht, anscheinend kommt kein Wert zurück.

Das Connection Timeout Problem liegt an deiner WLAN Verbindung oder Ahoy hat mal nicht reagiert, ist aber erstmal nicht tragisch.

jimknopf63, 2024-01-07:
Grüß Dich,

nachdem ich die Fehlermeldung gelesen habe habe ich mir den Smartmeter Adapter im IoBroker mal angeschaut. Der aktualisiert aber schön alle 5 Sekunden seinen Wert (in den Objekten nachvollziehbar). Ist ein Hichi IR Lesekopf per USB direkt am IoBroker eingestöpselt. Aber danke, werde das mal weiter beobachten. Und beim Ahoy stehen beide Balken (rechts und links) stabil bis am oberen Rand. Ist auch nur eine Terrassentür zwischen Ahoy und WR, knapp 8 Meter entfernt. Allerdings produziert die Anlage heute kaum wegen Schneefall und es geht immer mal ein WR offline.

Gruß Stephan

reserve85, 2024-01-07:
ja mag sein dass IOBroker funktioniert, aber das Script kann den Wert von IOBroker nicht auslesen. Dann funktioniert auch die Nulleinspeisung nicht. Du kannst das hier mal in deinen Brower eingeben um zu sehen ob es funktioniert:
`http://{IOBROKER_IP}:{IOBROKER_PORT}/getBulk/{IOBROKER_CURRENT_POWER_ALIAS}'`
natürlich vorher die {..} durch den Inhalt deiner Config ersetzen.

Zur Verbindung: mache mal von deinem Server aus einen Ping:
`ping 192.168.2.59 -n 100`
Ggf. lässt sich hier was ableiten.

jimknopf63, 2024-01-07:
Dank deines Hinweises habe ich mir noch einmal die Override genau angesehen. Beim Power Alias habe ich beim kopieren einen Fehler gemacht…. die 2 Zeichen vor smartmeter.0. entfernt.
Somit kommen Daten vom Smartmeter jetzt an. Und sieh das Problem als gelöst an. Der Depp saß vor dem PC. Ich hatte noch die alte Instanz parallel auf der gleichen IP laufen, abgeschaltet und nun kommen keine Fehler mehr. Ich kann mich gar nicht genug bedanken für deine Zeit und Hilfe.

Schönen Tag noch

Stephan

reserve85, 2024-01-07:
Kein Problem, gern geschehen!

jimknopf63, 2024-01-07:
Danke, Problem gelöst!

closed 2024-01-07 by jimknopf63