reserve85/HoymilesZeroExport #119: Hoymiles DTU Wlite S & Tibber

effe232, 2024-01-08:
Hey, dein Script sieht fantastisch aus, ich wollte das für mich umsetzen. Ich habe eine Frage dazu:

Ich würde gerne den Tibber Pulse als Zähler in Home Assistant integrieren und in deinem Script als Quelle angeben. So wie ich das gesehen habe, sollte das auch klappen, wenn ich entsprechend HA als Quelle angebe. Mich würde allerdings noch interessieren, ob du eine Möglichkeit siehst, mit der Stock Hoymiles DTU WLITE S zu arbeiten. Oder denkst du, ich soll mir lieber eine OPEN DTU zusammenbauen?

Viele Grüße and keep up the Great work!!! :)

reserve85, 2024-01-08:
Hi,

es wird nur openDTU und Ahoy unterstützt. Ansonsten läuft Tibber mit HomeAssistant so wie du es beschrieben hast. 

Beste Grüße,
Tobias

scharrrfi, 2024-01-10:
Hatte ich - da ich kürzlich zu Tibber gewechselt bin - auch kurz über Home Assistant mit der offline API im Einsatz und kann nur von abraten die Werte mit dem Skript zu nutzen. 
Die aktuellen Verbrauchswerte über tibber kommen zwar im 2-4 Sekundentakt, aber anscheinend immer mit einer gewissen Verzögerung. So regelt sich das Skript zu Tode und trifft eigentlich nie den Nullpunkt.

closed 2024-02-06 by reserve85