reserve85/HoymilesZeroExport #120: Feature Request: Einstellbares Delay bei HOY_BATTERY_THRESHOLD

scharrrfi, 2024-01-10:
Gude,

ich hätte da glaube ich (mal wieder) einen sinnvollen Feature Request. Und zwar, dass man eine Art Verzögerung bei den Schwellen des Batterymode einstellen kann. Quasi, erst wenn die definierte Schwelle für x Sekunden / Minuten unter- oder überschritten ist, Drossel den Wechselrichter oder schalte ihn ab oder an.

Hintergrund: Bei den aktuellen Außentemperaturen steht meine Batteriebank relativ kühl. Dies führt dazu, dass ich bei höherer Last auf die Batterien einen kurzzeitigen Spannungsabfall habe, der zum Abschalten führt, obwohl die Batterien noch ordentlich Kapazität haben. Bei wärmeren Temperaturen tritt das Problem nicht auf, weswegen ich ungerne die Schwelle verschieben würde. Daher würde ich einfach 2-3 Minuten Verzögerung einstellen, so dass die Batterien erst nach einer Unterschreitung der Schwellen über diesen Zeitraum gedrosselt oder abgeschaltet werden. 

Viele Grüße
Ramon

reserve85, 2024-01-10:
Probiere mal bitte den dev-zweig. Da kannst du in der config die Anzahl der Messungen eintragen (also anzahl der "loops"). Ich hoffe das löst dein Problem.

scharrrfi, 2024-01-10:
Du bist schneller als die Polizei erlaubt! :) Danke!

Ich hab es gerade mal getestet, kann aber nicht sicher sagen, ob es funktioniert. 
Im Journal aktualisieren anscheinend die "Lowest panel voltage"-Einträge nicht (mehr?) immer. Heißt, der Wert dort bleibt stehen, obwohl in der AhoyDTU bereits ein niedriger Wert angezeigt wird und auch über MQTT veröffentlicht wird. Irgendwann wird dann gedrosselt bzw. abgeschaltet. Kann also sein, dass es funktioniert und nur im Printout nicht aktualisiert wird.

reserve85, 2024-01-10:
Ja gerade gesehen. Ich habe daraufhin die Log-Ausgabe auf den Durchschnittswert angepasst. Sollte jetzt verständlicher sein.

Gruß Tobi

reserve85, 2024-01-14:
@scharrrfi konntest du die neue Version testen?

scharrrfi, 2024-01-15:
Also sie läuft und drosselt und ich hatte zuletzt kein ungewolltes Abschalten. 
Ob sie tatsächlich nach dem eingestellten Szenario (in meinem Fall 10 Messungen) drosselt, konnte ich leider noch nicht nachvollziehen, weil hier wettertechnisch die Batterien aktuell nicht wirklich viel aufladen. 

Also kaputt macht sie auf jeden Fall nichts, daher kannst du sie glaube ich in den Main-Zweig mergen. Ich melde mich, sobald ich das Verhalten genau nachvollziehen konnte.

closed 2024-02-06 by reserve85