reserve85/HoymilesZeroExport #121: Fehler beim start

Delta1987, 2024-01-12:
[image]

Ich habe immer wieder das gleiche problem.

Vielleicht wiesst ihr da weiter

Hab euch mal meine Config mit geschickt

Gruß Dennis

# HoymilesZeroExport - https://github.com/reserve85/HoymilesZeroExport
# Copyright (C) 2023, Tobias Kraft

# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.

# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.

# You should have received a copy of the GNU General Public License
# along with this program.  If not, see <http://www.gnu.org/licenses/>.

# ---------------------------------------------------------------------
# --- DO NOT DELETE ANY ENTRIES HERE - else your script won´t start ---
# ---------------------------------------------------------------------

[VERSION]
VERSION = 1.64

[SELECT_DTU]
# --- define your DTU (only one) ---
USE_AHOY = false
USE_OPENDTU = false

[SELECT_POWERMETER]
# --- define your Powermeter (only one) ---
USE_TASMOTA = false
USE_SHELLY_EM = false
USE_SHELLY_3EM = true
USE_SHELLY_3EM_PRO = false
USE_SHRDZM = false
USE_EMLOG = false
USE_IOBROKER = false
USE_HOMEASSISTANT = false
USE_VZLOGGER = false

[AHOY_DTU]
# --- defines for AHOY-DTU ---
# in settings/inverter set interval to 6 seconds!
AHOY_IP = xxx.xxx.xxx.xxx

[OPEN_DTU]
# --- defines for OPEN-DTU ---
OPENDTU_IP = xxx.xxx.xxx.xxx
OPENDTU_USER = 
OPENDTU_PASS = 

[TASMOTA]
# --- defines for Tasmota Smartmeter Modul---
TASMOTA_IP = xxx.xxx.xxx.xxx
# the following three constants describes how to navigate through the Tasmota-JSON. 
# if you do not know the format of your Tasmota reader, open a browser and put in the following string replacing xxx with the IP address of your Tasmota device: http://xxx.xxx.xxx.xxx/cm?cmnd=status%2010
# e.g. JSON_Result = {"StatusSNS":{"Time":"2023-02-28T12:49:49","SML":{"total_kwh":15011.575,"curr_w":-71}}}
TASMOTA_JSON_STATUS = StatusSNS
# Prefix for Web UI and MQTT JSON payload
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = SML
[... 201 lines cut ...]
# set limit of 300W -> inverter 1 is set to 300W and inverter 2 is powered off
HOY_BATTERY_PRIORITY = 1

# grid power
#    ...
#     |
#     | -------> if you defined ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT > 0, then the limit will jump to the defined percent of HOY_MAX_WATT when reaching this point. else: increasing limit <-------
#     |
#   [0W]      [POWERMETER_MAX_POINT]
#     |
#     | -------> increasing limit <-------
#     |
#  [-50W]     [POWERMETER_TARGET_POINT + POWERMETER_TOLERANCE]
#     |
#     | -------> no limit change between -100W ... -50W <-------
#     |
#  [-75W]     [POWERMETER_TARGET_POINT]
#     |
#     | -------> no limit change between -100W ... -50W <-------
#     |
#  [-100W]    [POWERMETER_TARGET_POINT - POWERMETER_TOLERANCE]
#     |
#     | -------> decreasing limit <-------
#     |
#    ...

reserve85, 2024-01-12:
Hi,

schau mal hier: https://github.com/reserve85/HoymilesZeroExport/issues/118

Delta1987, 2024-01-12:
Hab ich,. Habe das auch noch installiert und dennoch kommt diese Meldung

reserve85, 2024-01-12:
Mein Fehler:

`pip3 install packaging`

Delta1987, 2024-01-13:
dennis@pi:~ $ pip3 install packaging
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

das bekomme ich rasu wenn ich das eingebe

reserve85, 2024-01-13:
`pip3 install packaging --break-system-packages`

Delta1987, 2024-01-13:
[image]
 also wenn ich das so eingebe, 4mal mein rasp passwort eingebe dann starte er es aber er überschreibt nicht die DTU

reserve85, 2024-01-13:
dann bist du ja schon mal einen schritt weiter. Das mit den ganzen Passworteingaben und installieren der Packages liegt an deinem System, das kenne ich so nicht.

mit 
`sudo journalctl -u HoymilesZeroExport.service -n 20000 -e -f`
kannst du dir das Log anschauen, vermutlich gibt es noch ein Problem in der config irgendwo.

Delta1987, 2024-01-13:
[Log Datei.txt](https://github.com/reserve85/HoymilesZeroExport/files/13928649/Log.Datei.txt)

hab die jetzt mal rauskopiert, da steht was von zeile 32, hab nachgeschaut das ist die info wegen Shelly3

Delta1987, 2024-01-13:
Ich kann auch gerne das ganze teil neu aufsetzten habe damit kein problem, hab das zwar schon mindestens 10mal gemacht und immer hänge ich an der gleichen stelle

reserve85, 2024-01-13:
Sorry aber das liegt an deiner Python Installation, dir fehlt definitiv das Package „packaging“. Das musst du installieren.

irrwitzer42, 2024-01-14:
Hey @Delta1987 ,
Du schreibst, Du hast es schon 10mal neu aufgesetzt - meint das den Checkout vom ZeroExport Skript, oder meinst Du damit, dass Du den ganzen Rechner, auf dem das Skript läuft?

Du könntest es Dir deutlich einfacher machen und die Docker-Lösung einsetzen, dann musst Du Dich nicht mit Python und den Abhängigkeiten rumärgern, weil die alle im Container enthalten sind.

```
sudo apt-install docker.io docker-compose # sofern Du ein debian basiertes System verwendest
cat > docker-compose.yml <<EOF
version: '3.3'
services:
  hoymileszeroexport:
    image: ghcr.io/reserve85/hoymileszeroexport:main
    volumes:
      - ./HoymilesZeroExport_Config_Override.ini:/app/config.ini
    command: -c /app/config.ini
EOF
# in der Annahme, Du hast schon eine HoymilesZeroExport_Config_Override.ini angelegt:
docker-compose up -d && docker-compose logs -f
```

in der Theorie sollte das das Programm dann laufen.  Die Ausgabe des Logfiles kannst Du dann mit STRG+C Stoppen, der Container und das Programm laufen weiter.

Delta1987, 2024-01-16:
also egal was ich mache ich bekomme es nicht zum laufen.

Ich habe eine Rasp 3 zur verfügung, Open DTU und Shelly 3EM

Den Rasp installiere ich mit dem Rasp Installer, richte dort schon gleich den SSH und das wlan ein. 

Versucht habe ich jetzt mehrfach den Zeroexport und mehrfach die Nulleinspeisung von selbstbau PV, beide laufen nicht.

Die Anleitung ist ja hier super aber sie funktioniert halt bei mir leider nicht.

@reserve85  okay und wie installiere ich nur das nach?

Delta1987, 2024-01-16:
Also ich setzte den Rasp neu auf.
[image]
 das ist die software die ich immer wähle, stimmt die soweit?

irrwitzer42, 2024-01-16:
Nimm lieber die hier:

Moderne 64bit und Du brauchst auf dem Ding definitiv kein Desktop-Environment.

Delta1987, 2024-01-16:
@irrwitzer42, okay die hab ich jetzt drauf

Delta1987, 2024-01-16:
Hab es zum laufen gebracht, jetzt setzt er den WR immer auf 25W, aber ich bin schonmal weiter^^

reserve85, 2024-01-16:
ja, bekommt keine Daten vom Shelly. Zeig mal deine komplette config.

Delta1987, 2024-01-17:
es läuft, ich hatte im shelly die anmeldung beschrängt, wieso auch immer.....

jetzt musst ich nur noch die feineinstellung der config machen.

danke vielmals für eure hilfe.

closed 2024-01-17 by reserve85