reserve85/HoymilesZeroExport #122: Fehler bei der Installation auf Raspberry Pi 2 zero w - Docker

crisi-solar, 2024-01-13:
Hallo, habe heute komplett alles neu auf eine 64 GB-MicroSD-Karte installiert, erhalte aber beim installieren im Dockerteil diese Fehlermeldungen:

[image]

reserve85, 2024-01-14:
Da kann ich leider wenig helfen, da ich kein Docker in Verwendung habe. 
Ggf. könnte hier @irrwitzer42 oder @tomquist unterstützen?

irrwitzer42, 2024-01-14:
Hi @crisi-solar ,
zu allererst: ich habe das HoymilesZeroExport auf nem Pi 4 in Docker laufen, nicht auf einem Pi Zero - das _kann_ einen Unterschied machen.

Der untere Teil Deines Screenshots ist einfach zu lösen: Das "Skript" in der Readme unter Docker/docker-compose musst Du kopieren und innerhalb des Ordners, in dem Du das Python-skript und die Configs liegen hast, in eine neue Datei namens "docker-compose.yml" einfügen (z.B. mit nano, pico, vi, emacs...)
Anschliessend kannst Du disen docker "stack" dann mit
`docker-compose up -d`  bzw. `docker compose up -d` starten (startet dann bei Reboot auch automatisch wieder mit).

Falls docker-compose bei Dir noch nicht installiert ist:
`apt install docker.io docker-compose`

Im manuellen docker run  im oberen Teil Deines Screenshots ist glaube ich noch ein Fehler drin. Probiers mal mit:

`docker run -d --name hoymileszeroexport \
    -v ${PWD}/HoymilesZeroExport_Config_Override.ini:/app/HoymilesZeroExport_Config_Override.ini \
    ghcr.io/reserve85/hoymileszeroexport:main -c /app/HoymilesZeroExport_Config_Override.ini`

Man beachte hier den anderen Pfad für den Parameter -c beim Aufruf. Das könnte das "no such file or directory" erklären, aber wieso da "unknown" steht, weiss ich nicht.

Good luck! Und gib bitte kurz Bescheid, obs dann funktioniert.

reserve85, 2024-01-15:
das `-c /app/HoymilesZeroExport_Config_Override.ini` müsste dann noch in der readme angepasst werden wenn es funktioniert.?!

irrwitzer42, 2024-01-15:
Ja genau, falls Crisi-Solar positives Feedback gibt, kriegts Du noch nen PR mit ein paar readme Anpassungen, vielleicht etwas mehr Howto.

karrenbauer, 2024-01-15:
gibt es eine einfache Anleitung, wie man das Ganze in eine Synology DS918 mit Docker machen kann?

irrwitzer42, 2024-01-15:
@karrenbauer ich würde annehmen, dass das mit den Infos (docker parameter / docker-compose file) aus der readme genauso funktionieren würde, wie mit jedem anderen Docker Container auf der Synology. Mangels eigenem NAS kann ich das jedoch nicht ausprobieren.

tomquist, 2024-01-15:
@crisi-solar Ich habe einen fix hier: https://github.com/reserve85/HoymilesZeroExport/pull/123

Das Problem was, dass das Alpine image die Python binary nicht finden kann weil es das build result vom inkompatiblem Debian nutzt.

crisi-solar, 2024-01-16:
Hallo!
Sorry dass ich mich erst jetzt melde, habe inzwischen eine Lösung mit der Nulleinspeisung direkt auf der openDTU onBattery laufen, funktioniert super und es wird keine zweite Hardware benötigt.
Trotzdem vielen Dank und sorry dass ich euch mit meinem Problem hier belästigt habe.

PR #123 (2024-01-16): docker: fix alpine-based image
fixes #122
closed 2024-01-16 by reserve85