reserve85/HoymilesZeroExport #124: Tatsächliches Limit vom Inverter abfragen

Ollipop030, 2024-01-20:
Moin,

könnte man nicht einen Counter einbauen, der, wenn "Inverter Limit already xxx Watt" 10x hintereinander gemeldet wird, das tatsächliche Limit vom Inverter abfragt?
Zum Hintergrund:
Ich habe den Retry Counter auf 1. Was jetzt aber häufiger morgens bei der ersten Kommunikation passiert, ist dass das Script 1x das Limit auf 100% setzt. Die Inverter sind aber noch nicht ganz wach, und dann kommt ein "Limit timeout". Die 100% sind also nicht angekommen, die Inverter haben ein tatsächliches Limit vom Vortag. Das Script aber meldet immer "Inverter Limit already xxx Watt" und wird das Limit nie erhöhen. Daher die Idee, ab und zu mal die Inverter abzufragen, welches Limit tatsächlich gesetzt ist.

reserve85, 2024-01-22:
Ich fände es besser den retry counter zu entfernen und weiterprobieren bis ACK erfolgreich von der DTU (inverter) bestätigt wird...

Ollipop030, 2024-01-22:
Kann es sein, der der Retry Counter bei einem Timeout nicht greift? Hab den auf 5 gesetzt, und es wird kein neuer Anlauf versucht. Oder ist der Counter noch vom Vortag auf 5? 
Aber die Idee ist gut, den Counter komplett rauszunehmen. Die Ahoy Versionen ab 0.8.50 laufen auch wirklich sauber.

Ollipop030, 2024-01-22:
```
2024-01-20 08:28:40,702 INFO     Ahoy: Inverter "HM-1500" reachable: False
2024-01-20 08:28:40,729 INFO     Ahoy: Inverter "HM-300" reachable: False
2024-01-20 08:28:53,811 INFO     Ahoy: Inverter "HM-1500" reachable: True
2024-01-20 08:28:53,862 INFO     Ahoy: Inverter "HM-1500" / serial number "116183924915" / temperature 4.9 degC
2024-01-20 08:28:53,887 INFO     Ahoy: Inverter "HM-300" reachable: True
2024-01-20 08:28:53,935 INFO     Ahoy: Inverter "HM-1500" / serial number "116183924915" / temperature 4.9 degC
2024-01-20 08:28:53,982 INFO     Ahoy: Inverter "HM-300" / serial number "112174953798" / temperature 5.4 degC
2024-01-20 08:28:54,006 INFO     powermeter EMLOG: 176  Watt
2024-01-20 08:28:54,007 INFO     setting new limit to 1800 Watt
2024-01-20 08:28:54,008 INFO     Ahoy: Inverter "HM-1500": setting new limit from 1500 Watt to 1500 Watt
2024-01-20 08:29:01,488 INFO     Ahoy: Inverter "HM-1500": Limit timeout!
2024-01-20 08:29:01,490 INFO     Ahoy: Inverter "HM-300": setting new limit from 300 Watt to 300 Watt
2024-01-20 08:29:08,864 INFO     Ahoy: Inverter "HM-300": Limit timeout!
2024-01-20 08:29:20,947 INFO     Ahoy: Inverter "HM-1500" reachable: True
2024-01-20 08:29:20,974 INFO     Ahoy: Inverter "HM-300" reachable: True
2024-01-20 08:29:20,998 INFO     powermeter EMLOG: 175  Watt
2024-01-20 08:29:20,999 INFO     Retry Counter exceeded: Inverterlimit already at 1800 Watt
```

Hier vom Freitag morgen. 

Am Donnerstag Abend wurde das Limit auf 1800 Watt gesetzt, hier die letzten Werte:

```
2024-01-19 16:50:38,211 INFO     Ahoy: Inverter "HM-1500" reachable: True
2024-01-19 16:50:38,240 INFO     Ahoy: Inverter "HM-300" reachable: True
2024-01-19 16:50:38,264 INFO     powermeter EMLOG: 118  Watt
2024-01-19 16:50:38,265 INFO     Inverterlimit already at 1800 Watt
2024-01-19 16:50:50,397 INFO     Ahoy: Inverter "HM-1500" reachable: True
2024-01-19 16:50:50,425 INFO     Ahoy: Inverter "HM-300" reachable: True
2024-01-19 16:50:50,450 INFO     powermeter EMLOG: 119  Watt
2024-01-19 16:50:50,451 INFO     Inverterlimit already at 1800 Watt
2024-01-19 16:51:02,583 INFO     Ahoy: Inverter "HM-1500" reachable: False
2024-01-19 16:51:02,605 INFO     Ahoy: Inverter "HM-300" reachable: False
2024-01-19 16:51:15,692 INFO     Ahoy: Inverter "HM-1500" reachable: False
2024-01-19 16:51:15,718 INFO     Ahoy: Inverter "HM-300" reachable: False
2024-01-19 16:51:28,798 INFO     Ahoy: Inverter "HM-1500" reachable: False
```

Da ja die Inverter fest limitiert sind hatte ich auch morgens falsche Werte.

reserve85, 2024-01-22:
https://github.com/reserve85/HoymilesZeroExport/tree/dev_LimitRetry

Ollipop030, 2024-01-22:
1.67 läuft. Ich beobachte das mal ein paar Tage. Und wie immer: Danke für die schnelle Reaktion.

Ollipop030, 2024-01-23:
Gestern hatte ich 1-2 Timeouts:
```
2024-01-22 14:33:33,367 INFO     Ahoy: Inverter "HM-1500": setting new limit from 465 Watt to 460 Watt
2024-01-22 14:33:34,650 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
2024-01-22 14:33:34,651 INFO     Ahoy: Inverter "HM-300": setting new limit from 93 Watt to 92 Watt
2024-01-22 14:33:42,124 INFO     Ahoy: Inverter "HM-300": Limit timeout!
2024-01-22 14:33:54,150 INFO     intermediate meter EMLOG: 158  Watt
2024-01-22 14:33:54,151 INFO     Cut limit to 338 Watt, limit was higher than 10 percent of live-production
2024-01-22 14:33:54,210 INFO     Ahoy: Inverter "HM-1500" reachable: True
2024-01-22 14:33:54,236 INFO     Ahoy: Inverter "HM-300" reachable: True
2024-01-22 14:33:54,257 INFO     powermeter EMLOG: 159  Watt
2024-01-22 14:33:54,258 INFO     Inverterlimit 552 Watt was previously not accepted by the inverter, trying again...
2024-01-22 14:33:54,258 INFO     setting new limit to 552 Watt
2024-01-22 14:33:54,259 INFO     Ahoy: Inverter "HM-1500": setting new limit from 460 Watt to 460 Watt
2024-01-22 14:33:54,822 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
2024-01-22 14:33:54,823 INFO     Ahoy: Inverter "HM-300": setting new limit from 92 Watt to 92 Watt
2024-01-22 14:33:55,436 INFO     Ahoy: Inverter "HM-300": Limit acknowledged
```

Vielleicht sollte das Limit nur pro Inverter nochmal gesendet werden, der 1500 hat das Limit ja akzeptiert, nur der 300 nicht.

reserve85, 2024-01-23:
Ist angepasst, probierst du es nochmal bitte.

reserve85, 2024-01-23:
jein, der wiederholt nicht weil stattdessen ein neues Limit gesetzt wird.
DIe 97W -> 97W kann vorkommen, er geht immer vom "gesamtlimit" aus, also 489W vs 487W. 
Durch Rundung wird beim 300W Inverter dann das gleiche Limit gesetzt... ich schau es mir mal an

Ollipop030, 2024-01-23:
> jein, der wiederholt nicht weil stattdessen ein neues Limit gesetzt wird. DIe 97W -> 97W kann vorkommen, er geht immer vom "gesamtlimit" aus, also 489W vs 487W. Durch Rundung wird beim 300W Inverter dann das gleiche Limit gesetzt... 

Das wird dann wohl schon immer so gewesen sein. Ist mir nie aufgefallen. Übrigens: Die Wiederholung kam vorhin öfter mal vor, hat aber nichts mit dem Timeout zu tun, war wohl rein zufällig.

reserve85, 2024-01-23:
ich glaub ich habs, bitte nochmal die *.py ersetzen...

Ollipop030, 2024-01-23:
Läuft jetzt, ich berichte weiter...

Ollipop030, 2024-01-23:
Zwischendurch wird auf dem 300er kein Limit gesetzt, wenn es zu klein ist. Er taucht dann aber auch nicht im Log auf, bis das Limit sich signifikant ändert.

```
Jan 23 15:29:15 Pi-Hole python3[6058]: 2024-01-23 15:29:15 INFO     setting new limit to 645 Watt
Jan 23 15:29:15 Pi-Hole python3[6058]: 2024-01-23 15:29:15 INFO     Ahoy: Inverter "HM-1500": setting new limit from 555 Watt to 537 Watt
Jan 23 15:29:16 Pi-Hole python3[6058]: 2024-01-23 15:29:16 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
Jan 23 15:29:16 Pi-Hole python3[6058]: 2024-01-23 15:29:16 INFO     Ahoy: Inverter "HM-300": setting new limit from 111 Watt to 107 Watt
Jan 23 15:29:17 Pi-Hole python3[6058]: 2024-01-23 15:29:17 INFO     Ahoy: Inverter "HM-300": Limit acknowledged
Jan 23 15:29:30 Pi-Hole python3[6058]: 2024-01-23 15:29:30 INFO     intermediate meter EMLOG: 88  Watt
Jan 23 15:29:30 Pi-Hole python3[6058]: 2024-01-23 15:29:30 INFO     Cut limit to 268 Watt, limit was higher than 10 percent of live-production
Jan 23 15:29:30 Pi-Hole python3[6058]: 2024-01-23 15:29:30 INFO     Ahoy: Inverter "HM-1500" reachable: True
Jan 23 15:29:30 Pi-Hole python3[6058]: 2024-01-23 15:29:30 INFO     Ahoy: Inverter "HM-300" reachable: True
Jan 23 15:29:30 Pi-Hole python3[6058]: 2024-01-23 15:29:30 INFO     powermeter EMLOG: 322  Watt
Jan 23 15:29:30 Pi-Hole python3[6058]: 2024-01-23 15:29:30 INFO     Inverterlimit was already accepted at 645 Watt
Jan 23 15:29:42 Pi-Hole python3[6058]: 2024-01-23 15:29:42 INFO     intermediate meter EMLOG: 88  Watt
Jan 23 15:29:42 Pi-Hole python3[6058]: 2024-01-23 15:29:42 INFO     Cut limit to 268 Watt, limit was higher than 10 percent of live-production
Jan 23 15:29:42 Pi-Hole python3[6058]: 2024-01-23 15:29:42 INFO     Ahoy: Inverter "HM-1500" reachable: True
Jan 23 15:29:42 Pi-Hole python3[6058]: 2024-01-23 15:29:42 INFO     Ahoy: Inverter "HM-300" reachable: True
Jan 23 15:29:42 Pi-Hole python3[6058]: 2024-01-23 15:29:42 INFO     powermeter EMLOG: 320  Watt
Jan 23 15:29:42 Pi-Hole python3[6058]: 2024-01-23 15:29:42 INFO     setting new limit to 643 Watt
Jan 23 15:29:42 Pi-Hole python3[6058]: 2024-01-23 15:29:42 INFO     Ahoy: Inverter "HM-1500": setting new limit from 537 Watt to 535 Watt
Jan 23 15:29:42 Pi-Hole python3[6058]: 2024-01-23 15:29:42 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
Jan 23 15:29:54 Pi-Hole python3[6058]: 2024-01-23 15:29:54 INFO     intermediate meter EMLOG: 89  Watt
Jan 23 15:29:54 Pi-Hole python3[6058]: 2024-01-23 15:29:54 INFO     Cut limit to 269 Watt, limit was higher than 10 percent of live-production
Jan 23 15:29:54 Pi-Hole python3[6058]: 2024-01-23 15:29:54 INFO     Ahoy: Inverter "HM-1500" reachable: True
Jan 23 15:29:54 Pi-Hole python3[6058]: 2024-01-23 15:29:54 INFO     Ahoy: Inverter "HM-300" reachable: True
Jan 23 15:29:55 Pi-Hole python3[6058]: 2024-01-23 15:29:55 INFO     powermeter EMLOG: 320  Watt
Jan 23 15:29:55 Pi-Hole python3[6058]: 2024-01-23 15:29:55 INFO     setting new limit to 644 Watt
Jan 23 15:29:55 Pi-Hole python3[6058]: 2024-01-23 15:29:55 INFO     Ahoy: Inverter "HM-1500": setting new limit from 535 Watt to 536 Watt
Jan 23 15:29:56 Pi-Hole python3[6058]: 2024-01-23 15:29:56 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
Jan 23 15:30:08 Pi-Hole python3[6058]: 2024-01-23 15:30:08 INFO     intermediate meter EMLOG: 88  Watt
Jan 23 15:30:08 Pi-Hole python3[6058]: 2024-01-23 15:30:08 INFO     Cut limit to 268 Watt, limit was higher than 10 percent of live-production
Jan 23 15:30:08 Pi-Hole python3[6058]: 2024-01-23 15:30:08 INFO     Ahoy: Inverter "HM-1500" reachable: True
Jan 23 15:30:08 Pi-Hole python3[6058]: 2024-01-23 15:30:08 INFO     Ahoy: Inverter "HM-300" reachable: True
Jan 23 15:30:08 Pi-Hole python3[6058]: 2024-01-23 15:30:08 INFO     powermeter EMLOG: 370  Watt
Jan 23 15:30:08 Pi-Hole python3[6058]: 2024-01-23 15:30:08 INFO     setting new limit to 693 Watt
Jan 23 15:30:08 Pi-Hole python3[6058]: 2024-01-23 15:30:08 INFO     Ahoy: Inverter "HM-1500": setting new limit from 536 Watt to 577 Watt
Jan 23 15:30:09 Pi-Hole python3[6058]: 2024-01-23 15:30:09 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
Jan 23 15:30:09 Pi-Hole python3[6058]: 2024-01-23 15:30:09 INFO     Ahoy: Inverter "HM-300": setting new limit from 107 Watt to 115 Watt
Jan 23 15:30:10 Pi-Hole python3[6058]: 2024-01-23 15:30:10 INFO     Ahoy: Inverter "HM-300": Limit acknowledged
```

reserve85, 2024-01-23:
ja, so sollte es sein.

reserve85, 2024-01-25:
bei mir siehts in den Logs gut aus, konntest du noch was beobachten?

Ollipop030, 2024-01-25:
```
2024-01-25 13:43:00,964 INFO     Ahoy: Inverter "HM-300": setting new limit from 83 Watt to 82 Watt
2024-01-25 13:43:08,438 INFO     Ahoy: Inverter "HM-300": Limit timeout!
2024-01-25 13:43:08,464 INFO     Ahoy: Inverter "HM-1500" reachable: True
2024-01-25 13:43:08,488 INFO     Ahoy: Inverter "HM-300" reachable: True
2024-01-25 13:43:08,512 INFO     powermeter EMLOG: -77  Watt
2024-01-25 13:43:09,537 INFO     powermeter EMLOG: -77  Watt
2024-01-25 13:43:10,564 INFO     powermeter EMLOG: -77  Watt
2024-01-25 13:43:11,593 INFO     powermeter EMLOG: -77  Watt
2024-01-25 13:43:12,619 INFO     powermeter EMLOG: -77  Watt
2024-01-25 13:43:13,647 INFO     powermeter EMLOG: -73  Watt
2024-01-25 13:43:14,673 INFO     powermeter EMLOG: -73  Watt
2024-01-25 13:43:15,699 INFO     powermeter EMLOG: -73  Watt
2024-01-25 13:43:16,726 INFO     powermeter EMLOG: -73  Watt
2024-01-25 13:43:17,753 INFO     powermeter EMLOG: -73  Watt
2024-01-25 13:43:18,779 INFO     powermeter EMLOG: -73  Watt
2024-01-25 13:43:19,810 INFO     powermeter EMLOG: -73  Watt
2024-01-25 13:43:20,835 INFO     powermeter EMLOG: -73  Watt
2024-01-25 13:43:21,862 INFO     intermediate meter EMLOG: 313  Watt
2024-01-25 13:43:21,864 INFO     Inverterlimit 493 Watt was previously not accepted by at least one inverter, trying again...
2024-01-25 13:43:21,865 INFO     setting new limit to 493 Watt
2024-01-25 13:43:21,865 INFO     Ahoy: Inverter "HM-300": setting new limit from 82 Watt to 82 Watt
2024-01-25 13:43:23,187 INFO     Ahoy: Inverter "HM-300": Limit acknowledged
2024-01-25 13:43:23,214 INFO     Ahoy: Inverter "HM-1500" reachable: True
2024-01-25 13:43:23,243 INFO     Ahoy: Inverter "HM-300" reachable: True
```

Auch hier alles ok.

closed 2024-02-06 by reserve85