reserve85/HoymilesZeroExport #127: Powerlimit wegen Akku betrieb auf x setzen

tbretz65, 2024-01-24:
Hallo,
ist es möglich das Powerlimit z.B. auf xxxWatt zwischen Sonnenuntergang und Sonnenaufgang zu setzen.
Somit könnte man die Entladung des Akkus (B2500) begrenzen.
Oder kann man das anders regulieren ?

Danke

reserve85, 2024-01-25:
Hi, das geht so "out of the box" nicht.

Aber es gibt einen Workaround. 
Du bräuchtest zwei configs: eine für Tagsüber und eine für Nachts (z.b. config_day_override.ini und config_night_override.ini). 

Dann müsstest du mit einem cron-job jeweils zum sonnenaufgang / sonnenuntergang die HoymilesZeroExport_Config_Override.ini überschreiben und das Script neu starten. 

Zeitlich klappt das z.B. so: https://raspberrypi.stackexchange.com/questions/90099/execute-a-script-at-sunrise-sunset

tbretz65, 2024-01-25:
Ok.
Danke für die Info.
Hab ich mit cron und zusätzlicher config am laufen.
:-)

closed 2024-02-06 by reserve85