reserve85/HoymilesZeroExport #128: Log nachträglich aufrufen

echo1483, 2024-01-25:
Hallo, wenn ich den Befehl „sudo python3 HoymilesZeroExport.py“ ausführe werden nach kurzer Zeit die Daten angezeigt. 
Kann man nachträglich einen Befehl absenden um sich die Daten anzeigen zu lassen. 
Danke echo

irrwitzer42, 2024-01-25:
Hi @Echo1483 ,

> Hallo, wenn ich den Befehl „sudo python3 HoymilesZeroExport.py“ ausführe werden nach kurzer Zeit die Daten angezeigt.
> Kann man nachträglich einen Befehl absenden um sich die Daten anzeigen zu lassen.
> Danke echo

wenn Du der Anleitung entsprechend das Programm als Service/Daemon laufen lässt, kriegst Du “normale” Logfiles im Syslog (*nix Systeme).

Wenn Du es nur auf der Konsole händisch startest, müsstest Du den Output in eine Datei umleiten, um sie nachher ansehen zu können. Unter *nix Systemen geht das mit dem > Zeichen:

sudo python3 HoymilesZeroExport.py > ZeroExport.log

Dann könntest Du Dir das Logfile mit 

less ZeroExport.log

ansehen.

VG,

   Irrwitzer

reserve85, 2024-01-25:
kannst auch in der config "ENABLE_LOG_TO_FILE = true" setzen, dann wird der Output auch geloggt.

closed 2024-02-06 by reserve85