reserve85/HoymilesZeroExport #130: Start HoymilesZeroExport

echo1483, 2024-01-27:
Noch eine Anfägerfrage. 
Muss man eingentlich HoymilesZeroExport jedesmal manuel starten damit das Programm läuft.
Ich habe das Gefühl HoymilesZeroExport arbeitet nur wenn ich den Befehl sudo python3 HoymilesZeroExport.py ausführe.
Das wäre schon blöd jedesmal HoymilesZeroExport von Hand zu starten. Wenn das so wäre gibt es eine Möglichkeit den Start zu automatisieren. 
LG und Danke Paul

irrwitzer42, 2024-01-27:
Hi @echo1483,hast Du das Programm bisher nur heruntergeladen, oder es auch gemäß dem Abschnitt “Linux Installation” in der README.md installiert?Das install.sh Skript erstellt eigentlich ein Service-File und aktiviert dieses, sodass das Programm beim Reboot als Daemon gestartet wird.Das kannst Du zB mitsystemctl —list-units | grep Hoymilesüberprüfen.sonst einfach mal das Install Skript ausführen:sudo bash install.shVG

echo1483, 2024-01-27:
Danke für die schnelle Antwort. Ich bin genau nach Anleitung vorgegangen und habe den Raspi neu gestartet. Momentan läuft alles wie es soll. Mir ist nur aufgefallen das am nächsten Tag die Steuerung nicht funktioniert, Wenn ich den Befehl sudo python3 HoymilesZeroExport.py ausführe ist alles wieder OK.

reserve85, 2024-01-28:
Dann stimmt was nicht, das script sollte automatisch starten. 
Hast du vielleicht zum schluss versehentlich sudo ./uninstall_service.sh ausgeführt?

echo1483, 2024-01-29:
@reserve85
Alles OK, Danke für deine Arbeit hatte wohl einen kleinen Fehler gemacht.
LG Paul

closed 2024-02-06 by reserve85