reserve85/HoymilesZeroExport #131: Hoymiles Zero Export with Shelly em zero export Control

lektronik, 2024-02-01:
Hello, i have an issue running the script. Zero export script doesn't work... I tried to change the POWERMETER_TARGET_POINT to 50 from -75 as default, but nothing happened. I have an export of -100watt and more (is written on the script and also on shelly em interface)... Maybe this is not the minimum wattage from the grid  Any help here..?

[COMMON]

Number of Inverters
INVERTER_COUNT = 1

max difference in percent between SetpointLimit change to approximate the power to new setpoint
SLOW_APPROX_LIMIT_IN_PERCENT = 20

if slow approximation: additional limit based on the limit-difference to "smoot the curve": newLimitSetpoint = newLimitSetpoint + (LimitDifference * SLOW_APPROX_FACTOR_IN_PERCENT / 100)
SLOW_APPROX_FACTOR_IN_PERCENT = 20

interval time for setting limit to Hoymiles
LOOP_INTERVAL_IN_SECONDS = 20

delay time after sending limit to Hoymiles Inverter
SET_LIMIT_DELAY_IN_SECONDS = 5

Timeout time to wait for Acknowledge after sending limit to Hoymiles Inverter
SET_LIMIT_TIMEOUT_SECONDS = 10

delay time after sending limit to Hoymiles Inverter when using more than one Inverter
SET_LIMIT_DELAY_IN_SECONDS_MULTIPLE_INVERTER = 2

polling interval for powermeter (must be < LOOP_INTERVAL_IN_SECONDS)
POLL_INTERVAL_IN_SECONDS = 1

if your powermeter exceeds POWERMETER_MAX_POINT: immediatelly set the limit to predefined percent of HOY_MAX_WATT (if you have more than one inverter it´s the sum of all HOY_MAX_WATT)
value = 0 disables the feature. Values are possible from [0 to 100]
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 100

max difference between Limit and real output power in % of HOY_MAX_WATT (100 = disabled)
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 100

enable logging to file
ENABLE_LOG_TO_FILE = false

how many logfiles you wish to keep
LOG_BACKUP_COUNT = 30

defines how often the Inverter Power Status will be set, set it to "-1" for disabled (infinite repeat)
SET_POWERSTATUS_CNT = 10

log the inverter temperature
LOG_TEMPERATURE = false

delay time after turning the inverter off or on
SET_POWER_STATUS_DELAY_IN_SECONDS = 10

[CONTROL]

--- global defines for control behaviour ---
POWERMETER_TARGET_POINT is the target power for powermeter in watts
POWERMETER_TARGET_POINT = 50

POWERMETER_TOLERANCE is the tolerance (pos and neg) around the target point. in this range no adjustment will be set
POWERMETER_TOLERANCE = 25

POWERMETER_MAX_POINT is the maximum power of your powermeter for the normal "regulation loop".
if your powermeter jumps over this point, the limit will be increased instantly. it is like a "super high priority limit change".
if you defined ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT > 0, then the limit will jump to the defined percent when reaching this point.
POWERMETER_MAX_POINT = 0

[image]

reserve85, 2024-02-02:
hi,

the problem is that you can´t establish a connection to your DTU and your powermeter.
Can you ping the two ip-adresses?

lektronik, 2024-02-02:
[image]

reserve85, 2024-02-02:
your POWERMETER_TARGET_POINT must be lower than POWERMETER_MAX_POINT.

please attach your whole config, then i´ll have a look at it.

lektronik, 2024-02-02:
SET_POWERSTATUS_CNT = -1
# log the inverter temperature
LOG_TEMPERATURE = false
# delay time after turning the inverter off or on
SET_POWER_STATUS_DELAY_IN_SECONDS = 10

[CONTROL]
# --- global defines for control behaviour ---
# POWERMETER_TARGET_POINT is the target power for powermeter in watts
POWERMETER_TARGET_POINT = -75
# POWERMETER_TOLERANCE is the tolerance (pos and neg) around the target point. in this range no adjustment will be set
POWERMETER_TOLERANCE = 25
# POWERMETER_MAX_POINT is the maximum power of your powermeter for the normal "regulation loop".
# if your powermeter jumps over this point, the limit will be increased instantly. it is like a "super high priority limit change".
# if you defined ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT > 0, then the limit will jump to the defined percent when reaching this point.
POWERMETER_MAX_POINT = 0

# List of INVERTERS, based on COMMON/COUNT
[INVERTER_1]
# power rating of your inverter
HOY_MAX_WATT = 1200
# minimum limit in percent, e.g. 5%
HOY_MIN_WATT_IN_PERCENT = 5
# factor to multiply before set Limit. Some Inverters have some offsets, with that factor you can compensate it. Default = 1
HOY_COMPENSATE_WATT_FACTOR = 1
# battery powered?
HOY_BATTERY_MODE = false
# voltage to turn off the inverter
HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V = 47
# voltage to limit the inverter to a maximum of HOY_BATTERY_REDUCE_WATT
HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V = 48
# voltage to limit the inverter to a maximum of HOY_BATTERY_NORMAL_WATT
HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 48.5
# maximum limit in watts when battery is high (above HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V)
HOY_BATTERY_NORMAL_WATT = 1500
# maximum limit in watts when battery is low (below HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V)
HOY_BATTERY_REDUCE_WATT = 300
# reenable inverter at this voltage
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 51
# define which panels you want to ignore for the panel voltage. E.g. if you have input 1 + 2 battery powered and input 3 + 4 for solar panels you can disable panel 3 and 4.
# if you want to ignore Panel 3 and Panel 4 type: "HOY_BATTERY_IGNORE_PANELS = 3,4". leave it empty for no ignore.
HOY_BATTERY_IGNORE_PANELS =
# define prioriry of the inverters (possible values: 1 (high) ... 5 (low); default = 1). same priorities are also possible. Not supported in mixed modes (a mix of solar and battery powered inverters)
# example 1 (default):
# inverter 1 = 1000W, priority = 1 // inverter 2 = 500W, priority = 1:
# set limit of 1100W -> inverter 1 is set to 733W and inverter 2 is set to 367W
# set limit of 300W -> inverter 1 is set to 200W and inverter 2 is set to 100W
# example 2:
# inverter 1 = 1000W, priority = 1 // inverter 2 = 500W, priority = 2:
# set limit of 1100W -> inverter 1 is set to 1000W and inverter 2 is set to 100W
# set limit of 300W -> inverter 1 is set to 300W and inverter 2 is powered off
HOY_BATTERY_PRIORITY = 1
# Number of measured values for the moving average of the min panel voltage
HOY_BATTERY_AVERAGE_CNT = 1

lektronik, 2024-02-02:
[SELECT_DTU]
 USE_OPENDTU = True

[SELECT_POWERMETER]
USE_SHELLY_EM = True

[OPEN_DTU]
# --- defines for OPEN-DTU ---
OPENDTU_IP = 192.168.4.1
OPENDTU_USER = OpenDTU-5359328
OPENDTU_PASS = 123456781

[SHELLY]
# --- defines for Shelly ---
SHELLY_IP = 192.168.33.1
# if you restricted the web login enter your username and password, for Shelly "Generation 2" devices the username is always "admin" (e.g. Shelly pro 3EM)
SHELLY_USER =shellyem-34945477FDEE 2
SHELLY_PASS =123456789

[COMMON]
# Number of Inverters
INVERTER_COUNT = 1
# max difference in percent between SetpointLimit change to approximate the power to new setpoint
SLOW_APPROX_LIMIT_IN_PERCENT = 20
# if slow approximation: additional limit based on the limit-difference to "smoot the curve": newLimitSetpoint = newLimitSetpoint + (LimitDifference * SLOW_APPROX_FACTOR_IN_PERCENT / 100)
SLOW_APPROX_FACTOR_IN_PERCENT = 20
# interval time for setting limit to Hoymiles
LOOP_INTERVAL_IN_SECONDS = 20
# delay time after sending limit to Hoymiles Inverter
SET_LIMIT_DELAY_IN_SECONDS = 5
# Timeout time to wait for Acknowledge after sending limit to Hoymiles Inverter
SET_LIMIT_TIMEOUT_SECONDS = 10
# delay time after sending limit to Hoymiles Inverter when using more than one Inverter
SET_LIMIT_DELAY_IN_SECONDS_MULTIPLE_INVERTER = 2
# polling interval for powermeter (must be < LOOP_INTERVAL_IN_SECONDS)
POLL_INTERVAL_IN_SECONDS = 1
# if your powermeter exceeds POWERMETER_MAX_POINT: immediatelly set the limit to predefined percent of HOY_MAX_WATT (if you have more than one inverter it´s the sum of all HOY_MAX_WATT)
# value = 0 disables the feature. Values are possible from [0 to 100]
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 100
# max difference between Limit and real output power in % of HOY_MAX_WATT (100 = disabled)
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 15
# enable logging to file
ENABLE_LOG_TO_FILE = false
# how many logfiles you wish to keep
LOG_BACKUP_COUNT = 30
# defines how often the Inverter Power Status will be set, set it to "-1" for disabled (infinite repeat)
SET_POWERSTATUS_CNT = -1
# log the inverter temperature
LOG_TEMPERATURE = false
# delay time after turning the inverter off or on
SET_POWER_STATUS_DELAY_IN_SECONDS = 10

[CONTROL]
# --- global defines for control behaviour ---
# POWERMETER_TARGET_POINT is the target power for powermeter in watts
POWERMETER_TARGET_POINT = 50
# POWERMETER_TOLERANCE is the tolerance (pos and neg) around the target point. in this range no adjustment will be set
POWERMETER_TOLERANCE = 25
# POWERMETER_MAX_POINT is the maximum power of your powermeter for the normal "regulation loop".
# if your powermeter jumps over this point, the limit will be increased instantly. it is like a "super high priority limit change".
# if you defined ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT > 0, then the limit will jump to the defined percent when reaching this point.
POWERMETER_MAX_POINT = 0

# List of INVERTERS, based on COMMON/COUNT
[INVERTER_1]
# power rating of your inverter
HOY_MAX_WATT = 1200
# minimum limit in percent, e.g. 5%
HOY_MIN_WATT_IN_PERCENT = 5
# factor to multiply before set Limit. Some Inverters have some offsets, with that factor you can compensate it. Default = 1
HOY_COMPENSATE_WATT_FACTOR = 1
# battery powered?
HOY_BATTERY_MODE = false
# voltage to turn off the inverter
HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V = 47
# voltage to limit the inverter to a maximum of HOY_BATTERY_REDUCE_WATT
HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V = 48
# voltage to limit the inverter to a maximum of HOY_BATTERY_NORMAL_WATT
HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 48.5
# maximum limit in watts when battery is high (above HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V)
HOY_BATTERY_NORMAL_WATT = 1500
# maximum limit in watts when battery is low (below HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V)
HOY_BATTERY_REDUCE_WATT = 300
# reenable inverter at this voltage
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 51
# define which panels you want to ignore for the panel voltage. E.g. if you have input 1 + 2 battery powered and input 3 + 4 for solar panels you can disable panel 3 and 4.
# if you want to ignore Panel 3 and Panel 4 type: "HOY_BATTERY_IGNORE_PANELS = 3,4". leave it empty for no ignore.
HOY_BATTERY_IGNORE_PANELS =
# define prioriry of the inverters (possible values: 1 (high) ... 5 (low); default = 1). same priorities are also possible. Not supported in mixed modes (a mix of solar and battery powered inverters)
# example 1 (default):
# inverter 1 = 1000W, priority = 1 // inverter 2 = 500W, priority = 1:
# set limit of 1100W -> inverter 1 is set to 733W and inverter 2 is set to 367W
# set limit of 300W -> inverter 1 is set to 200W and inverter 2 is set to 100W
# example 2:
# inverter 1 = 1000W, priority = 1 // inverter 2 = 500W, priority = 2:
# set limit of 1100W -> inverter 1 is set to 1000W and inverter 2 is set to 100W
# set limit of 300W -> inverter 1 is set to 300W and inverter 2 is powered off
HOY_BATTERY_PRIORITY = 1
# Number of measured values for the moving average of the min panel voltage
HOY_BATTERY_AVERAGE_CNT = 1

reserve85, 2024-02-02:
```
[CONTROL]
# --- global defines for control behaviour ---
# POWERMETER_TARGET_POINT is the target power for powermeter in watts
POWERMETER_TARGET_POINT = 50
# POWERMETER_TOLERANCE is the tolerance (pos and neg) around the target point. in this range no adjustment will be set
POWERMETER_TOLERANCE = 25
# POWERMETER_MAX_POINT is the maximum power of your powermeter for the normal "regulation loop".
# if your powermeter jumps over this point, the limit will be increased instantly. it is like a "super high priority limit change".
# if you defined ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT > 0, then the limit will jump to the defined percent when reaching this point.
POWERMETER_MAX_POINT = 0
```

Your POWERMETER_TARGET_POINT must be lower than POWERMETER_MAX_POINT. Please adjust this.
Is this your override config? is your "normal" config unchanged?

lektronik, 2024-02-02:
I had it -75 as default but the outcome in shelly inteface and the script was running negative numbers... How i set up this in order not to return current?

reserve85, 2024-02-02:
ah ok, then i would suggest something like that:

```
[CONTROL]
# --- global defines for control behaviour ---
# POWERMETER_TARGET_POINT is the target power for powermeter in watts
POWERMETER_TARGET_POINT = 30
# POWERMETER_TOLERANCE is the tolerance (pos and neg) around the target point. in this range no adjustment will be set
POWERMETER_TOLERANCE = 30
# POWERMETER_MAX_POINT is the maximum power of your powermeter for the normal "regulation loop".
# if your powermeter jumps over this point, the limit will be increased instantly. it is like a "super high priority limit change".
# if you defined ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT > 0, then the limit will jump to the defined percent when reaching this point.
POWERMETER_MAX_POINT = 400
```

lektronik, 2024-02-02:
>  max difference between Limit and real output power in % of HOY_MAX_WATT (100 = disabled)
 MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 15
 
 
 Shall i change this back to 100?

reserve85, 2024-02-02:
yes, you don´t need it.

lektronik, 2024-02-02:
I have some negative numbers again....

[image]

reserve85, 2024-02-02:
that is totally normal, when you turn off your light or your TV you need time to measure the powermeter and adjust the solar generation.

lektronik, 2024-02-02:
Thanks mate! Really a good job!

lektronik, 2024-02-06:
Hello again, after using it in a sunny day now it goes again shelly meter shows -400 w... Is there a way to configure it better? 
Does it has to do with POWERMETER_MAX_POINT = 400?

reserve85, 2024-02-07:
nah, you can adjust
LOOP_INTERVAL_IN_SECONDS = 20
to a lower value, then the adjustment is made by a shorter intervall. I have it at 10s, you can also try to go lower.

lektronik, 2024-02-07:
Is it possible to connect Shelly via bluetooth in order not to use a second wifi adapter? I use raspb 3pi model b. I couldn't find a way to use only the on board wi-fi to communicate with opendtu and shelly so i added an extra adapter. i use rasp lite version(i found power issues) and i am planning to use also a second clap transformer(intermediate) for faster reading. i am kind new to coding, i am going to learn more for the mqtt and the docker and i am gonna check the speeds.

lektronik, 2024-02-08:
Sorry let me understand something,first of all i have the shelly clap on the grid, if i use POWERMETER_TARGET_POINT 400 and then i exceed 400 watts the with ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 100, it will stop zero export right? So maybe i set POWERMETER_TARGET_POINT to 20000?

lektronik, 2024-02-08:
Or maybe as the clap holds (50A*230) 11.500 lets say 10000.

reserve85, 2024-02-08:
no, POWERMETER_TARGET_POINT is the TARGET, so if you set this to 100 then the inverter is controlled to reach this value on your shelly.

if you set a POWERMETER_MAX_POINT to e.g. 400, then the limit will be immediatelly set to 100% (so "no limit") if your shelly reaches 400W. --> you get the full power of your inverter.

lektronik, 2024-02-08:
So sorry i meant POWERMETER_MAX_POINT instead of POWERMETER_TARGET_POINT. 
I would like to prevent going backwards. in second place, i would like to adjust POWERMETER_TARGET_POINT to 0. I will first check how it goes to 30 then to 20 and 10.

reserve85, 2024-02-08:
if you set ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 0 then the jump will be disabled.
i would set POWERMETER_MAX_POINT to something around +100 of your target point.

lektronik, 2024-02-08:
So i close the jump and have it running 0-100.

lektronik, 2024-02-09:
[image]
[image]

lektronik, 2024-02-09:
Nothing happens...

reserve85, 2024-02-09:
what limit in your openDTU webGUI?

lektronik, 2024-02-09:
[image]

reserve85, 2024-02-09:
nah there is a problem setting the limit over the script? can you please restart the script and check if the limit is set correctly to the log output?

lektronik, 2024-02-09:
It was maybe persistent at starters

lektronik, 2024-02-09:
[image]

Something's wrong with the panels. I unistalled the service, i restarted dtu but now i only take these ones...

lektronik, 2024-02-09:
[image]|

There are two options: peristent and non persistant. I set it at 5% but if i use more power in house, it won't set up more... I used both of them but nothing different happens.

reserve85, 2024-02-09:
Set it to 100% persistent and let the script run. It will adjust always non-persistant.
You´ll never have to set it manually.

lektronik, 2024-02-14:
I was using for OPENDTU_USER the ssid name, i changed it to the correct and everything's working right... Thanks!

closed 2024-02-14 by lektronik

reserve85, 2024-02-14:
Ahhh perfect, now it makes sense. FYI, I implemented an error output when the password or login is false, so with the newest version (actually in dev state) it should be clear. Thank you for your reply!