reserve85/HoymilesZeroExport #132: Limit setzen mit Ahoy Version 0.8.70

Ollipop030, 2024-02-02:
Mit Ahoy ab Version 0.8.70 bekomme ich immer Timeouts beim Limit setzen. Scheinbar muss man jetzt eingeloggt sein:

https://github.com/lumapu/ahoy/issues/1199#issuecomment-1922377276

Bekommt man das ins Script eingebaut? Möchte Ahoy ungern ohne Passwort im Netzwerk laufen lassen.

reserve85, 2024-02-02:
Muss ich mir mal anschauen wie und ob ich das Passwort mitschicken kann.

lumapu, 2024-02-02:
es könnte auch noch ein anderer Fehler sein, es gibt noch einen ähnlichen issue heute.
API soll ohne Passwort gehen

Ollipop030, 2024-02-04:
0.8.72 geht auch nicht, keine Ahnung was nicht stimmt.

reserve85, 2024-02-05:
Sorry war das Wochenende unterwegs, geht es denn wenn du das Passwort deaktivierst?

Ollipop030, 2024-02-05:
Keine Ahnung, nicht mehr probiert. Bin wieder auf die 0.8.69 gewechselt, musste den ESP komplett neu flashen, downgrade über Weboberfläche ging nicht.

reserve85, 2024-02-06:
Also mit gesetztem Passwort funktioniert es bei mir auch nicht, egal welche pages ich als protected setze. 
in der Live Ansicht sieht das bei mir dann auch so aus:
[image]

Ollipop030, 2024-02-06:
Anscheinend gibt es in der 0.8.74 einen Anzeigefehler. https://github.com/lumapu/ahoy/issues/1406

Aber lt. @lumapu sollte ja möglich sein, über die API ein Limit zu setzen, auch wenn ein Passwort vergeben wurde.

Ollipop030, 2024-02-08:
0.8.76 geht immer noch nicht, es klappt, wenn man in Ahoy das Passwort entfernt. Woran liegts? Sollte man drüben mal ein Ticket dazu aufmachen? http://AhoyIP/api/ ist ja ganz normal erreichbar, ob mit oder ohne Passwort.

reserve85, 2024-02-08:
Ja, ich habe da meines Wissens keine Möglichkeit was zu ändern. Ist mMn ein DTU-Thema.

lumapu, 2024-02-08:
ganz sauber ist es nicht, aber es würde helfen, wenn man das Passwort mitschickt. Ich denke gerade auch darüber nach, evtl. per API-Token ... Aktuell ist es tatsächlich so implementiert, dass es mit Passwort nicht funktoniert, da die ClientIP ausgewertet ist - und diese ist im Normalfall nicht eingeloggt. Ich versuche in Kürze abhilfe zu schaffen.

lumapu, 2024-02-10:
siehe in lumapu/ahoy#1415 für Neugikeiten

Spoiler: es geht wieder, aber erfordert eine kleine Anpassung des ZeroExport-Skripts @reserve85

reserve85, 2024-02-11:
Bestens, werde mich die Tage ransetzen, Sohnemann hat Magen Darm mit nach Hause gebracht, ist heute schwierig…

reserve85, 2024-02-11:
Sollte wieder mit der aktuellen DEV laufen.

MetaChuh, 2024-02-11:
> Sollte wieder mit der aktuellen DEV laufen.

@reserve85 
wie machst du's eigentlich mit rückwärtskompatibilität später ?
wirst du wie in v1.73 MinVersion = '0.8.78' zulassen und ältere droppen ?

ps:
sehe du verwendest nur limit_nonpersistent_absolute.
den wert limit_nonpersistent_relative, den ich verwende, ist ab jetzt leider auch val 1000 für 100% statt wie früher 100.

hab' s mal wie du gemacht:
deine variante mit `AHOY_FACTOR = GetAhoyLimitFactor()` mit ahoy version check ist für mich (in php) bisher die beste und einfachste idee 👍

pps "ausgedachter" token:
ja, das was du mit random token gesehen hast, ist bei mir mit der vorherigen 0.8.77 version auch passiert.
einmal `cmd: auth` gesendet, und danach konnte ich sogar den token leer lassen.

reserve85, 2024-02-11:
Hi, ich konnte es heute aus zeitgründen nicht mit einer alten Version testen.
Falls es nicht klappt würde ich die äteren Versionen bei der Änderung einfach droppen. Den LimitFactor würde ich dann auch entfernen beim nächsten update.

reserve85, 2024-02-12:
Also mit der alten Version geht die Implementierung trotzdem, daher hab ich die Min-Version wieder auf den ursprünglichen Wert gesetzt.

MetaChuh, 2024-02-12:
@reserve85 

thx 👍🙏

btw: ich hab' grad' bei mir statt deiner response textabfrage, doch zusätzlich einen versionsabfrage gemacht, wegen den unterschiedlichen response strings (de/en) die sich dauernd ändern.
sonst triggerte es bei mir keinen auth versuch.

beispiel responses con 0.8.78/0.8.79:
```
{"success":false,"error":"nicht angemeldet, Kommando nicht möglich!"}
{"success":false,"error":"Authentifizierungsfehler"}
```

dein code snippet in v1.7.3:
```python
if response_dict["success"] == False and response_dict["error"] == "not logged in, command not possible!":
        DTU.Authenticate()
        SetLimitAhoy(pInverterId, pLimit)
        return
```

mein php code snippet aktuell:
(quick and dirty, ich requeste momentan jedesmal einen neuen token)

```php
// ahoy dtu mit auth >= v0.8.78
	if ($dtu_version >= "0.8.78" and $dtu_password)
	{
		$auth_json=`curl --max-time $timeout --header "Content-Type:application/json" --request POST --data '{"cmd": "auth", "val": "$dtu_password"}' "http://$dtu_ip/api/ctrl"`;
				
		$token=json_decode($auth_json, true)["token"];
		
		$ergebnis=`curl --max-time $timeout --header "Content-Type:application/json" --request POST --data '{"id": $inverter_key, "cmd": "$command", "val": "$value", "token": "$token"}' "http://$dtu_ip/api/ctrl"`;
		
		file_put_contents("$log_path/set_dtu.log", date("Y.m.d H:i:s") . " L" . __LINE__ . " dtu_version auth >= 0.8.78 - auth_json: $auth_json - token: $token - ergebnis: $ergebnis\n", FILE_APPEND);
	}
	else
	{
		// ahoy dtu ohne auth < v0.8.70
		$ergebnis=`curl --max-time $timeout --header "Content-Type:application/json" --request POST --data '{"id": $inverter_key, "cmd": "$command", "val": "$value"}' "http://$dtu_ip/api/ctrl"`;
		
		file_put_contents("$log_path/set_dtu.log", date("Y.m.d H:i:s") . " L" . __LINE__ . " dtu_version no auth < 0.8.70 - ergebnis: $ergebnis\n", FILE_APPEND);
	}
```

MetaChuh, 2024-02-12:
@reserve85 

ps: hast du mit 0.8.78/79 beim regeln die gleichen performance (speed) probleme wie ich ?

ursache: wenn man jetzt `limit_nonpersistent_relative` oder `limit_nonpersistent_absolute` schickt, liefert die ahoy dtu für ein paar sekunden den wert `65535` also `n/a`zurück.
dadurch verzögert sich der nächste hoch/runterregelungsschritt.
(dürfte sich beim ungelösten ahoy q&d hack fix eingeschlichen haben, wo die limits nach dtu reboot alle -1 angezeigt haben)

sind bei mir ca 5-10 sekunden langsamere intervalle als mit opendtu und ahoy v0.6.9 möglich, weil ich derzeit vor der nächsten drosselungen auf ein gültiges power limit der dtu warte.

reserve85, 2024-02-12:
Oh no, es gibt deutsche responsestrings? @lumapu könnte man das vereinheitlichen? Oder einen Fehlercode definieren?

MetaChuh, 2024-02-12:
@reserve85 
> ich warte nicht auf das Limit von Ahoy, sobald das ack kommt ist das für mich gebongt.

hab' ich jetzt gerade versucht, aber bevor die ahoy dtu intern was anderes als 65535 n/a als wert hat, ignoriert sie bisher jegliches neue limit, dass man ihr schickt.

reserve85, 2024-02-12:
ne, das kann ich nicht bestätigen. Bei mir geht das, habs gerade probiert.  5x hintereinander Limit setzen und er hat zuletzt immer das zuletzt gesetzte Limit eingestellt.

Ollipop030, 2024-02-12:
Hier auch alles super responsive, Ahoy 0.8.79 und Script 1.74 (endlich mit Password!!!).

reserve85, 2024-02-12:
ok, 
das kommt wenn das Passwort falsch ist:
#ifdef LANG_DE
    #define AUTH_ERROR "Authentifizierungsfehler"
#else /*LANG_EN*/
    #define AUTH_ERROR "authentication error"

und das wenn es protected ist:
#ifdef LANG_DE
    #define IS_PROTECTED "nicht angemeldet, Kommando nicht möglich!"
#else /*LANG_EN*/
    #define IS_PROTECTED "not logged in, command not possible!"
#endif

demnach müsste man bei der authentification auf AUTH_ERROR prüfen und bei den Kommandos auf IS_PROTECTED.
@lumapu ich würde es super finden, wenn du zwei definierte Fehlercodes dafür mitsenden könntest. Sonst muss ich immer die Übersetzungen mitprüfen.
z.b. 
```
        bool setCtrl(JsonObject jsonIn, JsonObject jsonOut, const char *clientIP) {
          ...
                    jsonOut[F("error")] = F(AUTH_ERROR);
                    jsonOut[F("error_code")] = F(AUTH_ERROR_CODE);
          ...
```

MetaChuh, 2024-02-12:
> ich würde es super finden, wenn du zwei definierte Fehlercodes dafür mitsenden könntest.

http status code 401 Unauthorized zum beispiel.

> 5x hintereinander Limit setzen und er hat zuletzt immer das zuletzt gesetzte Limit eingestellt.

bei mir auch, nur langsamer als die 1-2 sekunden intervalle die ich bei der batterieeinspeisung für netz flat line gewohnt bin.

lumapu, 2024-02-12:
gibt wieder ein Update unter lumapu/ahoy#1415

bin gespannt, wie das Feedback diesmal ausfällt 😊

MetaChuh, 2024-02-12:
danke dir @lumapu.

> bin gespannt, wie das Feedback diesmal ausfällt 😊

bitte sei stressfrei gespannt 🙏

keine der hier, oder bei dir in #1415 gelisteten dinge sind in einer ahoy release.
0.8.36 ist die letzte public release, und zb. auth oder den ahoy_factor *10 (der name ahoy_factor gefällt mir irgendwie 😉),
gibts auch erst seit der non public dev version 0.8.39.

alles nur prophylaxe.

thx 🙏

closed 2024-02-16 by reserve85