reserve85/HoymilesZeroExport #134: Powermeter zeigt nur noch 0 Watt an bei Einspeisung.

Ollipop030, 2024-02-07:
```
2024-02-07 11:49:16,963 INFO     Not enough energy producing: limit already at maximum
2024-02-07 11:49:16,964 INFO     Inverterlimit was already accepted at 1800 Watt
2024-02-07 11:49:17,038 INFO     Ahoy: Inverter "HM-1500" reachable: True
2024-02-07 11:49:17,064 INFO     Ahoy: Inverter "HM-300" reachable: True
2024-02-07 11:49:17,085 INFO     powermeter Emlog: 0 Watt
2024-02-07 11:49:18,109 INFO     powermeter Emlog: 0 Watt
2024-02-07 11:49:19,133 INFO     powermeter Emlog: 0 Watt
2024-02-07 11:49:20,157 INFO     powermeter Emlog: 0 Watt
```

```
2024-02-07 13:42:42,990 INFO     Inverterlimit was already accepted at 1800 Watt
2024-02-07 13:42:43,045 INFO     Ahoy: Inverter "HM-1500" reachable: True
2024-02-07 13:42:43,070 INFO     Ahoy: Inverter "HM-300" reachable: True
2024-02-07 13:42:43,092 INFO     powermeter Emlog: 0 Watt
2024-02-07 13:42:44,116 INFO     powermeter Emlog: 0 Watt
2024-02-07 13:42:45,141 INFO     powermeter Emlog: 0 Watt
2024-02-07 13:42:46,166 INFO     powermeter Emlog: 0 Watt
2024-02-07 13:42:47,191 INFO     powermeter Emlog: 0 Watt
2024-02-07 13:42:48,216 INFO     powermeter Emlog: 0 Watt
2024-02-07 13:42:49,241 INFO     powermeter Emlog: 0 Watt
2024-02-07 13:42:50,266 INFO     powermeter Emlog: 0 Watt
2024-02-07 13:42:51,291 INFO     powermeter Emlog: 0 Watt
2024-02-07 13:42:52,316 INFO     powermeter Emlog: 0 Watt
```

Das zieht sich hier durch die ganze Mittagszeit, Limit sollte eigentlich -60 sein. Wäre schön wenn es so genau wäre. Irgendwo muss aber ein Fehler im Script sein, Version 1.71. Bei Unterproduktion steht übrigens der richte Wert vom Powermeter, also daran liegt es nicht.

Ollipop030, 2024-02-07:
```
Feb 07 13:56:28 Pi-Hole python3[159723]: 2024-02-07 13:56:28 INFO     setting new limit to 368 Watt
Feb 07 13:56:28 Pi-Hole python3[159723]: 2024-02-07 13:56:28 INFO     Ahoy: Inverter "HM-1500": setting new limit from 413 Watt to 306 Watt
Feb 07 13:56:29 Pi-Hole python3[159723]: 2024-02-07 13:56:29 INFO     Ahoy: Inverter "HM-1500": Limit acknowledged
Feb 07 13:56:29 Pi-Hole python3[159723]: 2024-02-07 13:56:29 INFO     Ahoy: Inverter "HM-300": setting new limit from 82 Watt to 61 Watt
Feb 07 13:56:30 Pi-Hole python3[159723]: 2024-02-07 13:56:30 INFO     Ahoy: Inverter "HM-300": Limit acknowledged
Feb 07 13:56:30 Pi-Hole python3[159723]: 2024-02-07 13:56:30 INFO     Ahoy: Inverter "HM-1500" reachable: True
Feb 07 13:56:30 Pi-Hole python3[159723]: 2024-02-07 13:56:30 INFO     Ahoy: Inverter "HM-300" reachable: True
Feb 07 13:56:30 Pi-Hole python3[159723]: 2024-02-07 13:56:30 INFO     powermeter EMLOG: -183  Watt
Feb 07 13:56:31 Pi-Hole python3[159723]: 2024-02-07 13:56:31 INFO     powermeter EMLOG: -183  Watt
Feb 07 13:56:32 Pi-Hole python3[159723]: 2024-02-07 13:56:32 INFO     powermeter EMLOG: -183  Watt
Feb 07 13:56:33 Pi-Hole python3[159723]: 2024-02-07 13:56:33 INFO     powermeter EMLOG: -183  Watt
Feb 07 13:56:34 Pi-Hole python3[159723]: 2024-02-07 13:56:34 INFO     powermeter EMLOG: -183  Watt
Feb 07 13:56:35 Pi-Hole python3[159723]: 2024-02-07 13:56:35 INFO     powermeter EMLOG: -183  Watt
Feb 07 13:56:36 Pi-Hole python3[159723]: 2024-02-07 13:56:36 INFO     powermeter EMLOG: -183  Watt
Feb 07 13:56:37 Pi-Hole python3[159723]: 2024-02-07 13:56:37 INFO     powermeter EMLOG: -183  Watt
Feb 07 13:56:38 Pi-Hole python3[159723]: 2024-02-07 13:56:38 INFO     powermeter EMLOG: -183  Watt
Feb 07 13:56:39 Pi-Hole python3[159723]: 2024-02-07 13:56:39 INFO     powermeter EMLOG: -183  Watt
Feb 07 13:56:40 Pi-Hole python3[159723]: 2024-02-07 13:56:40 INFO     powermeter EMLOG: -183  Watt
Feb 07 13:56:42 Pi-Hole python3[159723]: 2024-02-07 13:56:42 INFO     powermeter EMLOG: -73  Watt
Feb 07 13:56:43 Pi-Hole python3[159723]: 2024-02-07 13:56:43 INFO     powermeter EMLOG: -73  Watt
```

Das ist die Version vorher, 1.67. Damit läuft es.

reserve85, 2024-02-07:
ja ist beim refactor untergegangen (2.7.0 wird nicht abgezogen) - ich schau morgen / übermorgen mal drüber.

reserve85, 2024-02-08:
https://github.com/reserve85/HoymilesZeroExport/tree/emlog_fix

Ollipop030, 2024-02-08:
Besten Dank. Ich berichte, aber ich bezweifle das es die nächsten Tage zu einer Überproduktion kommt. Wetter ist schon wieder mies.

Ollipop030, 2024-02-08:
Hier kam grad für 2 Minuten Sonne, es wird wieder richtig berechnet.

closed 2024-02-08 by Ollipop030