reserve85/HoymilesZeroExport #136: Wo kann ich das richtig einstellen damit nicht so viel eingespeist wird?

crisi-solar, 2024-02-10:
Hallo!
Ich habe gestern HoymilesZeroExport komplett neu aufgesetzt und heute gestartet, jedoch speist er meiner Meinung nach viel zu viel ein. 
An welcher Stelle muss ich das richtig einstellen damit annähernd 0-Einspeisung rauskommt?
So schaut es im Editor aus:
Feb 10 08:46:58 raspberrypi python3[454]: 2024-02-10 08:46:58 INFO     OpenDTU: Inverter "HMS-1600-4T": setting new limit from 202 Watt to 1500 Watt
Feb 10 08:47:00 raspberrypi python3[454]: 2024-02-10 08:47:00 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit acknowledged
Feb 10 08:47:00 raspberrypi python3[454]: 2024-02-10 08:47:00 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Feb 10 08:47:01 raspberrypi python3[454]: 2024-02-10 08:47:01 INFO     powermeter Shelly3EMPro: -202 Watt
Feb 10 08:47:02 raspberrypi python3[454]: 2024-02-10 08:47:02 INFO     powermeter Shelly3EMPro: -199 Watt
Feb 10 08:47:03 raspberrypi python3[454]: 2024-02-10 08:47:03 INFO     powermeter Shelly3EMPro: -198 Watt
Feb 10 08:47:04 raspberrypi python3[454]: 2024-02-10 08:47:04 INFO     powermeter Shelly3EMPro: -188 Watt
Feb 10 08:47:05 raspberrypi python3[454]: 2024-02-10 08:47:05 INFO     powermeter Shelly3EMPro: -195 Watt
Feb 10 08:47:06 raspberrypi python3[454]: 2024-02-10 08:47:06 INFO     powermeter Shelly3EMPro: -196 Watt
Feb 10 08:47:07 raspberrypi python3[454]: 2024-02-10 08:47:07 INFO     powermeter Shelly3EMPro: -196 Watt
Feb 10 08:47:08 raspberrypi python3[454]: 2024-02-10 08:47:08 INFO     powermeter Shelly3EMPro: -191 Watt
Feb 10 08:47:09 raspberrypi python3[454]: 2024-02-10 08:47:09 INFO     powermeter Shelly3EMPro: -194 Watt
Feb 10 08:47:10 raspberrypi python3[454]: 2024-02-10 08:47:10 INFO     powermeter Shelly3EMPro: -193 Watt
Feb 10 08:47:11 raspberrypi python3[454]: 2024-02-10 08:47:11 INFO     powermeter Shelly3EMPro: -193 Watt
Feb 10 08:47:13 raspberrypi python3[454]: 2024-02-10 08:47:13 INFO     powermeter Shelly3EMPro: -190 Watt
Feb 10 08:47:14 raspberrypi python3[454]: 2024-02-10 08:47:14 INFO     powermeter Shelly3EMPro: -194 Watt
Feb 10 08:47:15 raspberrypi python3[454]: 2024-02-10 08:47:15 INFO     powermeter Shelly3EMPro: -194 Watt
Feb 10 08:47:16 raspberrypi python3[454]: 2024-02-10 08:47:16 INFO     powermeter Shelly3EMPro: -197 Watt
Feb 10 08:47:17 raspberrypi python3[454]: 2024-02-10 08:47:17 INFO     powermeter Shelly3EMPro: -201 Watt
Feb 10 08:47:18 raspberrypi python3[454]: 2024-02-10 08:47:18 INFO     powermeter Shelly3EMPro: -205 Watt
Feb 10 08:47:19 raspberrypi python3[454]: 2024-02-10 08:47:19 INFO     powermeter Shelly3EMPro: -207 Watt
Feb 10 08:47:20 raspberrypi python3[454]: 2024-02-10 08:47:20 INFO     powermeter Shelly3EMPro: -213 Watt
Feb 10 08:47:21 raspberrypi python3[454]: 2024-02-10 08:47:21 INFO     powermeter Shelly3EMPro: -217 Watt
Feb 10 08:47:22 raspberrypi python3[454]: 2024-02-10 08:47:22 INFO     intermediate meter OpenDTU: 353 Watt
Feb 10 08:47:22 raspberrypi python3[454]: 2024-02-10 08:47:22 INFO     overproducing: reduce limit based on actual power
Feb 10 08:47:22 raspberrypi python3[454]: 2024-02-10 08:47:22 INFO     setting new limit to 239 Watt
Feb 10 08:47:22 raspberrypi python3[454]: 2024-02-10 08:47:22 INFO     OpenDTU: Inverter "HMS-1600-4T": setting new limit from 1500 Watt to 239 Watt
Feb 10 08:47:25 raspberrypi python3[454]: 2024-02-10 08:47:25 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit acknowledged
Feb 10 08:47:25 raspberrypi python3[454]: 2024-02-10 08:47:25 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Feb 10 08:47:25 raspberrypi python3[454]: 2024-02-10 08:47:25 INFO     powermeter Shelly3EMPro: -105 Watt
Feb 10 08:47:26 raspberrypi python3[454]: 2024-02-10 08:47:26 INFO     powermeter Shelly3EMPro: -94 Watt
Feb 10 08:47:27 raspberrypi python3[454]: 2024-02-10 08:47:27 INFO     powermeter Shelly3EMPro: -91 Watt
Feb 10 08:47:28 raspberrypi python3[454]: 2024-02-10 08:47:28 INFO     powermeter Shelly3EMPro: -98 Watt
Feb 10 08:47:30 raspberrypi python3[454]: 2024-02-10 08:47:30 INFO     powermeter Shelly3EMPro: -81 Watt
Feb 10 08:47:31 raspberrypi python3[454]: 2024-02-10 08:47:31 INFO     powermeter Shelly3EMPro: -96 Watt
Feb 10 08:47:32 raspberrypi python3[454]: 2024-02-10 08:47:32 INFO     powermeter Shelly3EMPro: -101 Watt
Feb 10 08:47:33 raspberrypi python3[454]: 2024-02-10 08:47:33 INFO     powermeter Shelly3EMPro: -96 Watt
Feb 10 08:47:34 raspberrypi python3[454]: 2024-02-10 08:47:34 INFO     powermeter Shelly3EMPro: -109 Watt
Feb 10 08:47:35 raspberrypi python3[454]: 2024-02-10 08:47:35 INFO     powermeter Shelly3EMPro: -110 Watt
Feb 10 08:47:36 raspberrypi python3[454]: 2024-02-10 08:47:36 INFO     powermeter Shelly3EMPro: -111 Watt
Feb 10 08:47:37 raspberrypi python3[454]: 2024-02-10 08:47:37 INFO     powermeter Shelly3EMPro: -113 Watt
Feb 10 08:47:38 raspberrypi python3[454]: 2024-02-10 08:47:38 INFO     powermeter Shelly3EMPro: -108 Watt
Feb 10 08:47:39 raspberrypi python3[454]: 2024-02-10 08:47:39 INFO     powermeter Shelly3EMPro: -109 Watt
Feb 10 08:47:40 raspberrypi python3[454]: 2024-02-10 08:47:40 INFO     powermeter Shelly3EMPro: -112 Watt
Feb 10 08:47:42 raspberrypi python3[454]: 2024-02-10 08:47:42 INFO     powermeter Shelly3EMPro: -112 Watt
Feb 10 08:47:43 raspberrypi python3[454]: 2024-02-10 08:47:43 INFO     powermeter Shelly3EMPro: -110 Watt
Feb 10 08:47:44 raspberrypi python3[454]: 2024-02-10 08:47:44 INFO     powermeter Shelly3EMPro: -107 Watt
Feb 10 08:47:45 raspberrypi python3[454]: 2024-02-10 08:47:45 INFO     powermeter Shelly3EMPro: -95 Watt
Feb 10 08:47:46 raspberrypi python3[454]: 2024-02-10 08:47:46 INFO     powermeter Shelly3EMPro: -108 Watt
Feb 10 08:47:47 raspberrypi python3[454]: 2024-02-10 08:47:47 INFO     overproducing: reduce limit based on previous limit setpoint by approximation
Feb 10 08:47:47 raspberrypi python3[454]: 2024-02-10 08:47:47 INFO     setting new limit to 213 Watt
Feb 10 08:47:47 raspberrypi python3[454]: 2024-02-10 08:47:47 INFO     OpenDTU: Inverter "HMS-1600-4T": setting new limit from 239 Watt to 213 Watt
Feb 10 08:47:50 raspberrypi python3[454]: 2024-02-10 08:47:50 INFO     OpenDTU: Inverter "HMS-1600-4T": Limit acknowledged
Feb 10 08:47:50 raspberrypi python3[454]: 2024-02-10 08:47:50 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Feb 10 08:47:50 raspberrypi python3[454]: 2024-02-10 08:47:50 INFO     powermeter Shelly3EMPro: -54 Watt
Feb 10 08:47:51 raspberrypi python3[454]: 2024-02-10 08:47:51 INFO     powermeter Shelly3EMPro: -56 Watt
Feb 10 08:47:52 raspberrypi python3[454]: 2024-02-10 08:47:52 INFO     powermeter Shelly3EMPro: -58 Watt
Feb 10 08:47:53 raspberrypi python3[454]: 2024-02-10 08:47:53 INFO     powermeter Shelly3EMPro: -65 Watt
Feb 10 08:47:54 raspberrypi python3[454]: 2024-02-10 08:47:54 INFO     powermeter Shelly3EMPro: -65 Watt
Feb 10 08:47:55 raspberrypi python3[454]: 2024-02-10 08:47:55 INFO     powermeter Shelly3EMPro: -65 Watt
Feb 10 08:47:56 raspberrypi python3[454]: 2024-02-10 08:47:56 INFO     powermeter Shelly3EMPro: -65 Watt
Feb 10 08:47:57 raspberrypi python3[454]: 2024-02-10 08:47:57 INFO     powermeter Shelly3EMPro: -66 Watt
Feb 10 08:47:58 raspberrypi python3[454]: 2024-02-10 08:47:58 INFO     powermeter Shelly3EMPro: -65 Watt
Feb 10 08:47:59 raspberrypi python3[454]: 2024-02-10 08:47:59 INFO     powermeter Shelly3EMPro: -68 Watt
Feb 10 08:48:01 raspberrypi python3[454]: 2024-02-10 08:48:01 INFO     powermeter Shelly3EMPro: -65 Watt
Feb 10 08:48:02 raspberrypi python3[454]: 2024-02-10 08:48:02 INFO     powermeter Shelly3EMPro: -64 Watt
[... 101 lines cut ...]
Feb 10 08:49:43 raspberrypi python3[454]: 2024-02-10 08:49:43 INFO     powermeter Shelly3EMPro: -60 Watt
Feb 10 08:49:44 raspberrypi python3[454]: 2024-02-10 08:49:44 INFO     Inverterlimit was already accepted at 205 Watt
Feb 10 08:49:44 raspberrypi python3[454]: 2024-02-10 08:49:44 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True
Feb 10 08:49:44 raspberrypi python3[454]: 2024-02-10 08:49:44 INFO     powermeter Shelly3EMPro: -60 Watt
Feb 10 08:49:45 raspberrypi python3[454]: 2024-02-10 08:49:45 INFO     powermeter Shelly3EMPro: -60 Watt
Feb 10 08:49:47 raspberrypi python3[454]: 2024-02-10 08:49:47 INFO     powermeter Shelly3EMPro: -61 Watt
Feb 10 08:49:48 raspberrypi python3[454]: 2024-02-10 08:49:48 INFO     powermeter Shelly3EMPro: -62 Watt
Feb 10 08:49:49 raspberrypi python3[454]: 2024-02-10 08:49:49 INFO     powermeter Shelly3EMPro: -58 Watt
[... cut ...]
Feb 10 08:50:03 raspberrypi python3[454]: 2024-02-10 08:50:03 INFO     powermeter Shelly3EMPro: -59 Watt
Feb 10 08:50:04 raspberrypi python3[454]: 2024-02-10 08:50:04 INFO     powermeter Shelly3EMPro: -59 Watt
Feb 10 08:50:05 raspberrypi python3[454]: 2024-02-10 08:50:05 INFO     powermeter Shelly3EMPro: -58 Watt
Feb 10 08:50:06 raspberrypi python3[454]: 2024-02-10 08:50:06 INFO     Inverterlimit was already accepted at 205 Watt
Feb 10 08:50:06 raspberrypi python3[454]: 2024-02-10 08:50:06 INFO     OpenDTU: Inverter "HMS-1600-4T" reachable: True

reserve85, 2024-02-10:
LOOP_INTERVAL_IN_SECONDS = 10
POWERMETER_TARGET_POINT = 0
POWERMETER_TOLERANCE = 25
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 0 (oder auch 100, hängt von deiner Meinung ab ob du lieber einspeist und weniger einkaufen musst)
POWERMETER_MAX_POINT = 100

crisi-solar, 2024-02-10:
Danke, schaut schon mal besser aus!
Jedoch steht in der openDTU beim Limit 256W ein anderer Wert als im Editor 270W:
[image]

crisi-solar, 2024-02-10:
Ok, danke!

closed 2024-02-21 by reserve85

crisi-solar, 2024-04-11:
Hallo,
wo kann ich bitte einstellen dass beim einschalten eines z.B. 2kw Verbrauchers schneller rauf und wieder runtergeregelt wird?
Bei 2 WR dauert es manchmal bis zu 20 Sekunden bis von 1,8kW auf 0 runtergeregelt wird und über den Tag verteilt wird daher dementsprechend auch eingespeist.
Vielen Dank!

reserve85, 2024-04-11:
postest du mal deine config bitte, dann kann ich mir es mal anschauen.

crisi-solar, 2024-04-11:
# HoymilesZeroExport - https://github.com/reserve85/HoymilesZeroExport
# Copyright (C) 2023, Tobias Kraft

# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.

# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.

# You should have received a copy of the GNU General Public License
# along with this program.  If not, see <http://www.gnu.org/licenses/>.

# ---------------------------------------------------------------------
# --- DO NOT DELETE ANY ENTRIES HERE - else your script won't start ---
# ---------------------------------------------------------------------

[VERSION]
VERSION = 1.89

[SELECT_DTU]
# --- define your DTU (only one) ---
USE_AHOY = false
USE_OPENDTU = true

[SELECT_POWERMETER]
# --- define your Powermeter (only one) ---
USE_TASMOTA = false
USE_SHELLY_EM = false
USE_SHELLY_3EM = false
USE_SHELLY_3EM_PRO = true
USE_SHRDZM = false
USE_EMLOG = false
USE_IOBROKER = false
USE_HOMEASSISTANT = false
USE_VZLOGGER = false
USE_SCRIPT = false

[AHOY_DTU]
# --- defines for AHOY-DTU ---
# in settings/inverter set interval to 6 seconds!
AHOY_IP = xxx.xxx.xxx.xxx
AHOY_PASS =

[OPEN_DTU]
# --- defines for OPEN-DTU ---
OPENDTU_IP = 192.168.68.54
OPENDTU_USER = admin
OPENDTU_PASS = openDTU42

[TASMOTA]
# --- defines for Tasmota Smartmeter Modul---
TASMOTA_IP = xxx.xxx.xxx.xxx
# the following three constants describes how to navigate through the Tasmota-JSON. 
# if you do not know the format of your Tasmota reader, open a browser and put in the following string replacing xxx with the IP address of your Tasmota device: http://xxx.xxx.xxx.xxx/cm?cmnd=status%2010
# e.g. JSON_Result = {"StatusSNS":{"Time":"2023-02-28T12:49:49","SML":{"total_kwh":15011.575,"curr_w":-71}}}
TASMOTA_JSON_STATUS = StatusSNS
# Prefix for Web UI and MQTT JSON payload
TASMOTA_JSON_PAYLOAD_MQTT_PREFIX = SML
# Power-MQTT label (the current power in Watt, positive (import more energy than export) or negative (export more energy than import)).
TASMOTA_JSON_POWER_MQTT_LABEL = curr_w
# if your powermeter does NOT output the current power: you need to calculate it -> Power(W) = OBIS(1.7.0) - OBIS(2.7.0)
TASMOTA_JSON_POWER_CALCULATE = FALSE
# Power-MQTT Input label (positive active instantaneous power, e.g. OBIS Code 1.7.0)
TASMOTA_JSON_POWER_INPUT_MQTT_LABEL =
# Power-MQTT output label (negative active instantaneous power, e.g. OBIS Code 2.7.0)
TASMOTA_JSON_POWER_OUTPUT_MQTT_LABEL =
[... 852 lines cut ...]
# Number of measured values for the moving average of the min panel voltage
HOY_BATTERY_AVERAGE_CNT = 1

# grid power
#    ...
#     |
#     | -------> if you defined ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT > 0, then the limit will jump to the defined percent of HOY_MAX_WATT when reaching this point. else: increasing limit <-------
#     |
#   [0W]      [POWERMETER_MAX_POINT]
#     |
#     | -------> increasing limit <-------
#     |
#  [-50W]     [POWERMETER_TARGET_POINT + POWERMETER_TOLERANCE]
#     |
#     | -------> no limit change between -100W ... -50W <-------
#     |
#  [-75W]     [POWERMETER_TARGET_POINT]
#     |
#     | -------> no limit change between -100W ... -50W <-------
#     |
#  [-100W]    [POWERMETER_TARGET_POINT - POWERMETER_TOLERANCE]
#     |
#     | -------> decreasing limit <-------
#     |
#    ...

crisi-solar, 2024-04-11:
Danke für die Hilfe!!!!!

reserve85, 2024-04-11:
Sieht auf den ersten Blick gut aus, kannst du mal ein logausschnitt zeigen bzw. anhängen mit dem Problem?

crisi-solar, 2024-04-11:
Z.B. gestern hat er in der Zeit 12 Uhr bis 13.30 Uhr jede Menge eingespeist weil vermutlich irgendwo hängengeblieben.
[image]
[image]

crisi-solar, 2024-04-11:
Z.B. hier sieht an an den Spitzen wenn sich ein großer Verbraucher einschaltet dass es dann einige Zeit nachregelt und nicht gleich auf 0 regelt.
[image]

crisi-solar, 2024-04-11:
Sorry, oben die falsche Grafik gepostet, die ist ja vom 9.4.

Hier die von heute:
[image]

reserve85, 2024-04-12:
da brauche ich ein Log, z.b. von diesem ausschnitt hier:
[image]
ich bin mir fast sicher, dass da was hängt...
du kannst ja mal ENABLE_LOG_TO_FILE = true setzen, dann kommst du immer wieder dran.

crisi-solar, 2024-04-12:
Heute früh zeigt mir die openDTU die beiden WR nicht an, der Logausschnitt von der Nacht schaut so aus:
Apr 11 20:49:01 raspberrypi python3[507]: 2024-04-11 20:49:01 INFO     OpenDTU: Inverter "HMS20004T": Limit timeout!
Apr 11 20:49:16 raspberrypi python3[507]: 2024-04-11 20:49:16 WARNING  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broke       n by 'ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0xb5dc0130>, 'Connection to 192.168.68.54 timed out. (connect timeout=10)')': /api/livedata/status?in       v=116491626836
Apr 11 20:49:27 raspberrypi python3[507]: 2024-04-11 20:49:27 WARNING  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broke       n by 'RemoteDisconnected('Remote end closed connection without response')': /api/livedata/status?inv=116491626836
Apr 11 20:49:34 raspberrypi python3[507]: 2024-04-11 20:49:34 WARNING  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broke       n by 'RemoteDisconnected('Remote end closed connection without response')': /api/livedata/status?inv=116491626836
Apr 11 20:49:44 raspberrypi python3[507]: 2024-04-11 20:49:44 ERROR    Exception at GetHoymilesAvailable, Inverter 0 (HMS16004T) not reachable
Apr 11 20:49:44 raspberrypi python3[507]: 2024-04-11 20:49:44 ERROR    HTTPConnectionPool(host='192.168.68.54', port=80): Max retries exceeded with url: /api/livedata/statu       s?inv=116491626836 (Caused by ReadTimeoutError("HTTPConnectionPool(host='192.168.68.54', port=80): Read timed out. (read timeout=10)"))
Apr 11 20:49:48 raspberrypi python3[507]: 2024-04-11 20:49:48 WARNING  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broke       n by 'RemoteDisconnected('Remote end closed connection without response')': /api/livedata/status?inv=116491439579
Apr 11 20:49:58 raspberrypi python3[507]: 2024-04-11 20:49:58 WARNING  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broke       n by 'ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0xb5dc0778>, 'Connection to 192.168.68.54 timed out. (connect timeout=10)')': /api/livedata/status?in       v=116491439579
Apr 11 20:50:08 raspberrypi python3[507]: 2024-04-11 20:50:08 WARNING  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broke       n by 'ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0xb5dc0d00>, 'Connection to 192.168.68.54 timed out. (connect timeout=10)')': /api/livedata/status?in       v=116491439579
Apr 11 20:50:18 raspberrypi python3[507]: 2024-04-11 20:50:18 ERROR    Exception at GetHoymilesAvailable, Inverter 1 (HMS20004T) not reachable
Apr 11 20:50:18 raspberrypi python3[507]: 2024-04-11 20:50:18 ERROR    HTTPConnectionPool(host='192.168.68.54', port=80): Max retries exceeded with url: /api/livedata/statu       s?inv=116491439579 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0xb5dc0c88>, 'Connection to 192.168.68.54 timed out. (connect timeout=10)'))
Apr 11 20:50:34 raspberrypi python3[507]: 2024-04-11 20:50:34 WARNING  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broke       n by 'ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0xb5dc03e8>, 'Connection to 192.168.68.54 timed out. (connect timeout=10)')': /api/livedata/status?in       v=116491626836
Apr 11 20:50:47 raspberrypi python3[507]: 2024-04-11 20:50:47 WARNING  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broke       n by 'ReadTimeoutError("HTTPConnectionPool(host='192.168.68.54', port=80): Read timed out. (read timeout=10)")': /api/livedata/status?inv=116491626836
Apr 11 20:50:59 raspberrypi python3[507]: 2024-04-11 20:50:59 WARNING  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broke       n by 'ReadTimeoutError("HTTPConnectionPool(host='192.168.68.54', port=80): Read timed out. (read timeout=10)")': /api/livedata/status?inv=116491626836
Apr 11 20:51:09 raspberrypi python3[507]: 2024-04-11 20:51:09 ERROR    Exception at GetHoymilesAvailable, Inverter 0 (HMS16004T) not reachable
Apr 11 20:51:09 raspberrypi python3[507]: 2024-04-11 20:51:09 ERROR    HTTPConnectionPool(host='192.168.68.54', port=80): Max retries exceeded with url: /api/livedata/statu       s?inv=116491626836 (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0xb5ddfe08>, 'Connection to 192.168.68.54 timed out. (connect timeout=10)'))
Apr 11 20:51:19 raspberrypi python3[507]: 2024-04-11 20:51:19 WARNING  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broke       n by 'ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0xb5dab580>, 'Connection to 192.168.68.54 timed out. (connect timeout=10)')': /api/livedata/status?in       v=116491439579
Apr 11 20:51:29 raspberrypi python3[507]: 2024-04-11 20:51:29 WARNING  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broke       n by 'ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0xb5dab430>, 'Connection to 192.168.68.54 timed out. (connect timeout=10)')': /api/livedata/status?in       v=116491439579
[... cut ...] Error(<urllib3.connection.HTTPConnection object at 0xb5dab7c0>, 'Connection to 192.168.68.54 timed out. (connect timeout=10)'))
Apr 11 21:25:06 raspberrypi python3[507]: 2024-04-11 21:25:06 WARNING  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broke       n by 'ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0xb5dabfd0>, 'Connection to 192.168.68.54 timed out. (connect timeout=10)')': /api/livedata/status?in       v=116491439579

crisi-solar, 2024-04-12:
Habe den ENABLE_LOG_TO_FILE = true gesetzt, script neu gestartet und siehe da - jetzt laufen beide WR in der openDTU wieder - d.h. es kommen wieder laufende Daten rein.
Hat es was damit zu tun dass ich das script neu gestartet habe?

reserve85, 2024-04-12:
ne, deine DTU ist einfach vom Netzwerk nicht erreichbar. Das erklärt quasi alle Probleme.

crisi-solar, 2024-04-12:
ich bekomme heute eine neue openDTU von einem Händler, die erste war eine gekaufte aus kleinanzeigen.de

Hier mal ein aktueller Logauszug wo ersichtlich ist wie lange die Regelung braucht um von z.B. -60 auf 0 zu regeln:

Apr 12 06:56:54 raspberrypi python3[3894]: 2024-04-12 06:56:54 INFO     powermeter Shelly3EMPro: -8 Watt
Apr 12 06:56:55 raspberrypi python3[3894]: 2024-04-12 06:56:55 INFO     powermeter Shelly3EMPro: -9 Watt
Apr 12 06:56:56 raspberrypi python3[3894]: 2024-04-12 06:56:56 INFO     powermeter Shelly3EMPro: -12 Watt
Apr 12 06:56:57 raspberrypi python3[3894]: 2024-04-12 06:56:57 INFO     powermeter Shelly3EMPro: -4 Watt
Apr 12 06:56:59 raspberrypi python3[3894]: 2024-04-12 06:56:59 INFO     powermeter Shelly3EMPro: -2 Watt
Apr 12 06:57:00 raspberrypi python3[3894]: 2024-04-12 06:57:00 INFO     Inverterlimit was already accepted at 148 Watt
Apr 12 06:57:00 raspberrypi python3[3894]: 2024-04-12 06:57:00 INFO     OpenDTU: Inverter "HMS16004T" reachable: True
Apr 12 06:57:00 raspberrypi python3[3894]: 2024-04-12 06:57:00 INFO     OpenDTU: Inverter "HMS20004T" reachable: True
Apr 12 06:57:00 raspberrypi python3[3894]: 2024-04-12 06:57:00 INFO     powermeter Shelly3EMPro: 0 Watt
Apr 12 06:57:01 raspberrypi python3[3894]: 2024-04-12 06:57:01 INFO     powermeter Shelly3EMPro: 2 Watt
Apr 12 06:57:02 raspberrypi python3[3894]: 2024-04-12 06:57:02 INFO     powermeter Shelly3EMPro: -4 Watt
Apr 12 06:57:03 raspberrypi python3[3894]: 2024-04-12 06:57:03 INFO     powermeter Shelly3EMPro: -1 Watt
Apr 12 06:57:04 raspberrypi python3[3894]: 2024-04-12 06:57:04 INFO     powermeter Shelly3EMPro: 2 Watt
Apr 12 06:57:05 raspberrypi python3[3894]: 2024-04-12 06:57:05 INFO     Inverterlimit was already accepted at 148 Watt
Apr 12 06:57:05 raspberrypi python3[3894]: 2024-04-12 06:57:05 INFO     OpenDTU: Inverter "HMS16004T" reachable: True
Apr 12 06:57:06 raspberrypi python3[3894]: 2024-04-12 06:57:06 INFO     OpenDTU: Inverter "HMS20004T" reachable: True
Apr 12 06:57:06 raspberrypi python3[3894]: 2024-04-12 06:57:06 INFO     powermeter Shelly3EMPro: 6 Watt
Apr 12 06:57:07 raspberrypi python3[3894]: 2024-04-12 06:57:07 INFO     powermeter Shelly3EMPro: 2 Watt
Apr 12 06:57:08 raspberrypi python3[3894]: 2024-04-12 06:57:08 INFO     powermeter Shelly3EMPro: 0 Watt
Apr 12 06:57:09 raspberrypi python3[3894]: 2024-04-12 06:57:09 INFO     powermeter Shelly3EMPro: 35 Watt
Apr 12 06:57:10 raspberrypi python3[3894]: 2024-04-12 06:57:10 INFO     powermeter Shelly3EMPro: 3 Watt
Apr 12 06:57:11 raspberrypi python3[3894]: 2024-04-12 06:57:11 INFO     Inverterlimit was already accepted at 148 Watt
Apr 12 06:57:11 raspberrypi python3[3894]: 2024-04-12 06:57:11 INFO     OpenDTU: Inverter "HMS16004T" reachable: True
Apr 12 06:57:11 raspberrypi python3[3894]: 2024-04-12 06:57:11 INFO     OpenDTU: Inverter "HMS20004T" reachable: True
Apr 12 06:57:11 raspberrypi python3[3894]: 2024-04-12 06:57:11 INFO     powermeter Shelly3EMPro: -3 Watt
Apr 12 06:57:12 raspberrypi python3[3894]: 2024-04-12 06:57:12 INFO     powermeter Shelly3EMPro: 10 Watt
Apr 12 06:57:13 raspberrypi python3[3894]: 2024-04-12 06:57:13 INFO     powermeter Shelly3EMPro: 9 Watt
Apr 12 06:57:15 raspberrypi python3[3894]: 2024-04-12 06:57:15 INFO     powermeter Shelly3EMPro: 7 Watt
Apr 12 06:57:16 raspberrypi python3[3894]: 2024-04-12 06:57:16 INFO     powermeter Shelly3EMPro: 5 Watt
Apr 12 06:57:17 raspberrypi python3[3894]: 2024-04-12 06:57:17 INFO     Inverterlimit was already accepted at 148 Watt
Apr 12 06:57:17 raspberrypi python3[3894]: 2024-04-12 06:57:17 INFO     OpenDTU: Inverter "HMS16004T" reachable: True
Apr 12 06:57:17 raspberrypi python3[3894]: 2024-04-12 06:57:17 INFO     OpenDTU: Inverter "HMS20004T" reachable: True
Apr 12 06:57:17 raspberrypi python3[3894]: 2024-04-12 06:57:17 INFO     powermeter Shelly3EMPro: -2 Watt
Apr 12 06:57:18 raspberrypi python3[3894]: 2024-04-12 06:57:18 INFO     powermeter Shelly3EMPro: 4 Watt
Apr 12 06:57:19 raspberrypi python3[3894]: 2024-04-12 06:57:19 INFO     powermeter Shelly3EMPro: 20 Watt
Apr 12 06:57:20 raspberrypi python3[3894]: 2024-04-12 06:57:20 INFO     powermeter Shelly3EMPro: -2 Watt
Apr 12 06:57:21 raspberrypi python3[3894]: 2024-04-12 06:57:21 INFO     powermeter Shelly3EMPro: 0 Watt
Apr 12 06:57:22 raspberrypi python3[3894]: 2024-04-12 06:57:22 INFO     Inverterlimit was already accepted at 148 Watt
Apr 12 06:57:22 raspberrypi python3[3894]: 2024-04-12 06:57:22 INFO     OpenDTU: Inverter "HMS16004T" reachable: True
Apr 12 06:57:23 raspberrypi python3[3894]: 2024-04-12 06:57:23 INFO     OpenDTU: Inverter "HMS20004T" reachable: True
Apr 12 06:57:23 raspberrypi python3[3894]: 2024-04-12 06:57:23 INFO     powermeter Shelly3EMPro: 2 Watt
Apr 12 06:57:24 raspberrypi python3[3894]: 2024-04-12 06:57:24 INFO     powermeter Shelly3EMPro: 0 Watt
Apr 12 06:57:25 raspberrypi python3[3894]: 2024-04-12 06:57:25 INFO     powermeter Shelly3EMPro: 5 Watt
Apr 12 06:57:26 raspberrypi python3[3894]: 2024-04-12 06:57:26 INFO     powermeter Shelly3EMPro: 3 Watt
Apr 12 06:57:27 raspberrypi python3[3894]: 2024-04-12 06:57:27 INFO     powermeter Shelly3EMPro: 4 Watt
Apr 12 06:57:28 raspberrypi python3[3894]: 2024-04-12 06:57:28 INFO     Inverterlimit was already accepted at 148 Watt
Apr 12 06:57:28 raspberrypi python3[3894]: 2024-04-12 06:57:28 INFO     OpenDTU: Inverter "HMS16004T" reachable: True
Apr 12 06:57:29 raspberrypi python3[3894]: 2024-04-12 06:57:29 INFO     OpenDTU: Inverter "HMS20004T" reachable: True
[... cut ...]
Apr 12 07:02:29 raspberrypi python3[3894]: 2024-04-12 07:02:29 INFO     OpenDTU: Inverter "HMS20004T": setting new limit from 83 Watt to 75 Watt
Apr 12 07:02:32 raspberrypi python3[3894]: 2024-04-12 07:02:32 INFO     OpenDTU: Inverter "HMS20004T": Limit acknowledged
Apr 12 07:02:32 raspberrypi python3[3894]: 2024-04-12 07:02:32 INFO     OpenDTU: Inverter "HMS16004T" reachable: True
Apr 12 07:02:32 raspberrypi python3[3894]: 2024-04-12 07:02:32 INFO     OpenDTU: Inverter "HMS20004T" reachable: True
Apr 12 07:02:32 raspberrypi python3[3894]: 2024-04-12 07:02:32 INFO     powermeter Shelly3EMPro: 20 Watt
^C