reserve85/HoymilesZeroExport #137: python versin

fatzo, 2024-02-10:
Ich bekomme auf einem frisch instaliertem Raspi5 diese fehlermeldung

Feb 10 18:50:37 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Feb 10 18:50:37 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Feb 10 18:50:37 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 4.
Feb 10 18:50:37 raspberrypi5 systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Feb 10 18:50:38 raspberrypi5 systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Feb 10 18:50:38 raspberrypi5 python3[26099]: Traceback (most recent call last):
F*eb 10 18:50:38 raspberrypi5 python3[26099]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 30, in <module>
Feb 10 18:50:38 raspberrypi5 python3[26099]:     from packaging import version
Feb 10 18:50:38 raspberrypi5 python3[26099]: ModuleNotFoundError: No module named 'packaging'
Feb 10 18:50:38 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILU**RE
Feb 10 18:50:38 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Feb 10 18:50:38 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 5.
Feb 10 18:50:38 raspberrypi5 systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Feb 10 18:50:38 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly.
Feb 10 18:50:38 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Feb 10 18:50:38 raspberrypi5 systemd[1]: Failed to start HoymilesZeroExport.service - HoymilesZeroExport Service.

ich habe schon rausgefunden das es an der Python Versin liegen muss
sudo pip3 install argparse
error: externally-managed-environment

This environment is externally managed
 nun komme ich leider nicht weiter
hilfe ist willkommen

gruß fatzo

reserve85, 2024-02-11:
Schau mal hier, https://stackoverflow.com/questions/75608323/how-do-i-solve-error-externally-managed-environment-every-time-i-use-pip-3

da sind mehrere Lösungswege beschrieben.

lullatscho, 2024-02-13:
> Ich bekomme auf einem frisch instaliertem Raspi5 diese fehlermeldung
> 
> Feb 10 18:50:37 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE Feb 10 18:50:37 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'. Feb 10 18:50:37 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 4. Feb 10 18:50:37 raspberrypi5 systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service. Feb 10 18:50:38 raspberrypi5 systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service. Feb 10 18:50:38 raspberrypi5 python3[26099]: Traceback (most recent call last): F*eb 10 18:50:38 raspberrypi5 python3[26099]: File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 30, in Feb 10 18:50:38 raspberrypi5 python3[26099]: from packaging import version Feb 10 18:50:38 raspberrypi5 python3[26099]: ModuleNotFoundError: No module named 'packaging' Feb 10 18:50:38 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILU**RE Feb 10 18:50:38 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'. Feb 10 18:50:38 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 5. Feb 10 18:50:38 raspberrypi5 systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service. Feb 10 18:50:38 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly. Feb 10 18:50:38 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'. Feb 10 18:50:38 raspberrypi5 systemd[1]: Failed to start HoymilesZeroExport.service - HoymilesZeroExport Service.
> 
> ich habe schon rausgefunden das es an der Python Versin liegen muss sudo pip3 install argparse error: externally-managed-environment
> 
> This environment is externally managed nun komme ich leider nicht weiter hilfe ist willkommen
> 
> gruß fatzo

Hatte das gleiche Problem, als ich es auf dem Pi5 zum laufen bringen wollte. Debian Bookworm hat defaultmäßig auf dem Pi5 Python3.11 installiert. Man muss sich quasi Python3.9 aus den Sourcen kompilieren und dann installieren.

Ich versuch das mal hier zu dokumentieren zeitnah, da ich auch ne Weile gebraucht habe und mehrere Lösungswege eher ins Nirwana geführt haben.

Jetzt läuft es aber.

lullatscho, 2024-02-14:
> Das wäre super. Hab auch schon viele Lösungswege ausprobiert, die nicht funktioniert haben. Dann warte ich auf deinen post Gruß fatzo Gesendet von Outlook für Android<https://aka.ms/AAb9ysg>
> […](#)

Habe es oben bereits angepasst. Probier es mal aus und geb mal Rückmeldung, ob es bei dir ebenso anschließend funktioniert.

reserve85, 2024-02-14:
Wenn das funktioniert dann würde ich das anheften oder in eine FAQ speichern…

fatzo, 2024-02-14:
> @fatzo Habe es oben bereits angepasst. Probier es mal aus und geb mal Rückmeldung, ob es bei dir ebenso anschließend funktioniert.

Bin gerade dabei

reserve85, 2024-02-14:
Übeogens hab ich durch Zufall gesehen, dass bei ahoy auch ein pull request zu dem Thema gibt. https://github.com/lumapu/ahoy/pull/1392
vielleicht könnt ihr ja mal überprüfen wie es dort gemacht wird (im diff der readme)

lullatscho, 2024-02-14:
@reserve85 hab mir das mal angeschaut. Dort wird das über das Virtual Environment (venv) gelöst. Das hatte ich auch irgendwo gelesen mit dem venv, aber hab es nicht so richtig durchblickt auf die schnelle.

reserve85, 2024-02-14:
Ja ich auch nicht und ich habe Gottseidank das Problem nicht 😁

fatzo, 2024-02-14:
Step 9:

xxx@raspberrypi:/usr/bin $ python3.9 --version
python3.9: error while loading shared libraries: libpython3.9.so.1.0: cannot open shared object file: No such file or directory
xxx@raspberrypi:/usr/bin $ sudo ldconfig /usr/local/lib/python3.9
xxx@raspberrypi:/usr/bin $ python3.9 --version
Python 3.9.18

womit muss ich das xxx ersetzen oder den pfad?
bei mir liegt das Pyhton in:
/home/pi/Python-3.9.18

Fehlermeldung

pi@raspberrypi5:~/Python-3.9.18 $ xxx@raspberrypi:/usr/bin $ python3.9 --version
python3.9: error while loading shared libraries: libpython3.9.so.1.0: cannot open shared object file: No such file or directory
xxx@raspberrypi:/usr/bin $ sudo ldconfig /usr/local/lib/python3.9
xxx@raspberrypi:/usr/bin $ python3.9 --version
Python 3.9.18
bash: xxx@raspberrypi:/usr/bin: Datei oder Verzeichnis nicht gefunden
bash: python3.9:: Kommando nicht gefunden.
bash: xxx@raspberrypi:/usr/bin: Datei oder Verzeichnis nicht gefunden
bash: xxx@raspberrypi:/usr/bin: Datei oder Verzeichnis nicht gefunden
bash: Python: Kommando nicht gefunden.

reserve85, 2024-02-16:
Bist du weiter gekommen?

lullatscho, 2024-02-16:
@fatzo wir können sonst gern mal zusammen drauf schauen

fatzo, 2024-02-16:
> @fatzo
> 
> Wir können sonst gern mal zusammen drauf schauen.
> 
> Schau bitte oben nochmal in meinen Kommentar. Ich habe die Formatierung und die Eingabe nochmal etwas angepasst, damit es übersichtlicher und eindeutiger wird.
> 
> Der Ordner den du anmerkst ist der entpackte Ordner in deinem Home-Verzeichnis. In den solltest du in **Step 5** wechseln und die folgenden Schritte 6, 7 und 8 ausführen. Bei **Step 8** hab ich noch hinzugefügt wie du checkst, dass dieser Ordner dann existiert.

Vielen dank jetzt hat es funktioniert.
Ich bedanke mich bei euch/dir 🥇 

Ich habe nochmals neu instaliert den Raspberry und bin dann nach deinem erweitertem post vorgegangen.
Danach muste ich nur noch die Dtu auf das neueste Release v24.2.12 updaten.
Nun ein Start ohne Fehler, allerdings ist es schon dunkel und morgen kann ich dann sagen ob es 100% läuft

gruß vom Unwissenden :-)

closed 2024-02-17 by fatzo