reserve85/HoymilesZeroExport #138: Steuerung Ladegerät

HA-IDIPS, 2024-02-12:
Das Script ist definitiv herausragend für Nulleinspeisung. Von mir herzlichen Dank für die tolle Umsetzung.

Nun frage ich mich, ob es auch eine Möglichkeit gäbe, bei PV-Überschuss (kein Akku-Bezug und Powermeter liefert negative Werte) ein Ladegerät anzusteuern bzw. in die Regelung einzubinden, welches sich per URL-Aufruf dynamisch steuern lässt. Beispiel:

Powermeter: -280 Watt
Akku-Bezug: +/- 0
-> http://ladegeraet.local/A_DC=5&V_DC=54 (Ladegerät mit 270 W ansteuern)

reserve85, 2024-02-12:
Hi,

das ist leider kein Anwendungsfall für das Script, du müsstest das über eine andere zusätzliche Steuerung lösen. 
Im prinzip reicht es dir ja den Zähler auszulesen ob er bei roundabout -280-250W steht und dann das Ladegerät anzuschalten. Sobald der Zähler wieder bei +200 oder so steht -> ausschalten.

HA-IDIPS, 2024-02-13:
Ich hatte gehofft, dass man bei Überschuss - statt auf Null runter zu regeln - einfach den URL-Aufruf mit einbaut. Wäre das denn nicht einfach möglich oder bei welchem Skriptteil könnte ich ggf. mein Glück versuchen? Leider kann ich nur PHP einigermaßen, kein Python, da müsste ich mich rein arbeiten.

Ollipop030, 2024-02-13:
Ich hab das auch schon versucht, das geht aber so nicht. Dafür arbeitet das Script einfach zu genau.

Problem: Ab einer Leistung von -280 Watt wird ein Ladegerät eingeschaltet (mit 270 Watt), dann speist du noch 10 Watt ein. Wenn jetzt im Haus ein Verbraucher noch ein paar Watt zieht, und du bei 200Watt Bezug bist, schaltet das Ladegerät schon wieder ab. Das Gleiche passiert an einem wolkigen Tag, wenn immer kurz die Sonne raus kommt, dann wieder Schatten ist. Dafür müsste man dann Überbrückungszeiten einbauen, das nur alle 10 Minuten geschaltet werden darf. Das habe ich alles durch, habs aufgegeben, auch um den Akku nicht zu schrotten bei 10-20x ein/aus pro Stunde. Dafür gibt es intelligentere Systeme als nur diese kleine (aber hervorragende) Script.

HA-IDIPS, 2024-02-13:
@Ollipop030 Frage zum Verständnis für mich: "auch um den Akku nicht zu schrotten bei 10-20x ein/aus pro Stunde"

Inwiefern wäre das für den Akku schlecht, wenn sich die Ladeleistung durch ein Ladegerät dynamisch vielfach pro Stunde ändert? Wäre das nicht das gleiche Prinzip wie bei einem "normalen" MPPT-Laderegler?

Ollipop030, 2024-02-13:
Ein Laderegler regelt aber auch so wie es soll, zB wird zum Ende hin die Ladeleistung immer weiter reduziert. Bei meinem fängt das ab ca 80% Kapazität an. Das Script müsste also auch mit dem Laderegler kommunizieren um den Ladestand abzufragen, damit der Akku ordnungsgemäß geladen wird.

closed 2024-02-21 by reserve85