reserve85/HoymilesZeroExport #140: Change in OpenDTU API Endpoint since 2024-01-30

schulti, 2024-02-17:
Thank you for the great work! The script failed after my update to the latest OpenDTU version. I found that there was a "Breaking Change" in API Endpoint which broke this script:

FromOpenDTU README:
```
Generated using: `git log --date=short --pretty=format:"* %h%x09%ad%x09%s" | grep BREAKING`

```code
* 1b637f08      2024-01-30      BREAKING CHANGE: Web API Endpoint /api/livedata/status and /api/prometheus/metrics
* e1564780      2024-01-30      BREAKING CHANGE: Web API Endpoint /api/livedata/status and /api/prometheus/metrics
* f0b5542c      2024-01-30      BREAKING CHANGE: Web API Endpoint /api/livedata/status and /api/prometheus/metrics
* c27ecc36      2024-01-29      BREAKING CHANGE: Web API Endpoint /api/livedata/status
```

The error looks like this:
```
INFO     OpenDTU: Inverter "yet unknown": Limit timeout!
ERROR    Exception at GetHoymilesActualPower
ERROR    'AC'
ERROR    try reading actual power from DTU:
ERROR    Exception at GetHoymilesActualPower
```

The JSON now received via /api/livedata/status/inverters does not contain the "AC" or "DC" part in the "inverters" Array. It's removed.

This breaks the functionality of this script when using OpenDTU version 24.2.12 or later. **The latest working version is OpenDTU 24.1.26**.

reserve85, 2024-02-17:
Hi Stefan,
You need to update to the latest version of the script. I already updated it.

schulti, 2024-02-17:
Wow! Thank you, it works after updating to the latest version of your script + config file! Amazing!

closed 2024-02-17 by schulti