reserve85/HoymilesZeroExport #142: SerialNumber when more than one inverter

schulti, 2024-02-20:
Hi,

I use your script to handle two Hoymiles Inverter with one OpenDTU. The inverters have different watt numbers (let's say 800W and 400W. I can configure it in the sections [INVERTER_1] for 800W and [INVERTER_2] for 400W.

Usually it works, but now I realized that sometimes the inverters get a different order (Inverter 800W is treated as 400W and the inverter 800W is treated as 400W). This is detected in the morning when the inverters start up and then stays for the day. Next day it could be fine again.

As you identify the serial number during Init, would it be specify to define the serial number in the config file so that this is checked during Init and kept in all situations?
2024-02-20 084:09:16,056 INFO     OpenDTU: Inverter "hm800 / serial number "123456789 / temperature 24.5 degC
2024-02-20 08:09:16,153 INFO     OpenDTU: Inverter "hms2000" / serial number "134567890 / temperature 23.3 degC

Thank you!

reserve85, 2024-02-20:
puh, then it seems that the order of the inverter changes in the API... 
I didn´t intend it that way. So i need to cross-check the serial number in the config and in the api to get the correct InverterID?
Can you check the Inverter order in OpenDTU? Did it changed?

schulti, 2024-02-20:
It's not critical but would help if the inverters have different wattage.

Yes, it worked changing the order in OpenDTU, but some days later it did not work. I believe it depends on what inverter sees the sun first in the morning and this is then the number 1. If the other inverter starts 10 minutes later, it's number 2. This could change on foggy days etc.

reserve85, 2024-02-20:
Hi, i took a look into the code - the whole openDTU communication is based on the specific serial numbers.
So every API-Request is called by the serial number and every API command too. No way to mix stuff here.
The only mix-up can happen here:
```
    def GetInfo(self, pInverterId: int):
        ParsedData = self.GetJson('/api/livedata/status')
        SERIAL_NUMBER[pInverterId] = str(ParsedData['inverters'][pInverterId]['serial'])
```
Get Info is called every time a inverter gets available == true.

--> For my understanding it would be better to only read the serial number one time at startup and don´t override... 
Also i would add the serial number field in the config as an optional parameter.

reserve85, 2024-02-20:
i uploaded it to the dev-path, it would be great if you could test this. just replace the .ini and .py and define your serial_number in the override.ini file

schulti, 2024-02-20:
Thank you, amazing! I will do at sunrise ;-)

schulti, 2024-02-21:
Good morning! It seems to work very well! I don't see any change in log or so, but it picks the right inverter even when I change the order in OpenDTU or try many restarts. Thank you very much!

closed 2024-02-21 by reserve85

schulti, 2024-02-21:
It works perfectly. I can now only regulate one inverter and due to the serial number in the config file it's always selecting the correct one.