reserve85/HoymilesZeroExport #144: Update/Neuinstallation auf 1.77 läuft nicht

jimknopf63, 2024-02-21:
Ich möchte von Version 1.64 updates bzw neu installieren auf die aktuelle Version. Grund weil Ahoy mit 0.8.84 mir falsche Leistungswerte ausgibt (über 100%). Ich habe das Script auf Proxmox in einem CT laufen. Nach dem Update/Neuinstallation (neue CT, vorheriges Script gestoppt) startet der Service nicht und gibt immer folgende Fehlermeldung raus. Finde aber jetzt den Fehler nicht.... Habe es mit kopierter Override.ini probiert aber auch mit ergänzter Config.ini. Das Ergebnis ändert sich nicht.

[image]

schulti, 2024-02-21:
Das schaut für mich so aus, als würde das Script im Config File nach `IOBROKER_POWER_CALCULATE` suchen und nicht finden. Hast du ioBroker im Einsatz?
Ist das Script in aktueller Version und hast du das Konfig-File von der aktuellen Version mit deinen Einstellungen angepasst?

jimknopf63, 2024-02-21:
Habe hier die aktuelle HoymilesZeroExport_Config.ini und meine HoymilesZeroExport_Config_Override.ini mal angehängt.
In der Config.ini fehlen die Inverter 3-xx sind aber bei mir vorhanden und unverändert.

[HMSE.zip](https://github.com/reserve85/HoymilesZeroExport/files/14359281/HMSE.zip)

jimknopf63, 2024-02-21:
> Das schaut für mich so aus, als würde das Script im Config File nach `IOBROKER_POWER_CALCULATE` suchen und nicht finden. Hast du ioBroker im Einsatz? Ist das Script in aktueller Version und hast du das Konfig-File von der aktuellen Version mit deinen Einstellungen angepasst?

Ja, ich habe den IoBroker im Einsatz und er läuft auch unter der richtigen IP und Port für die API

reserve85, 2024-02-21:
ah ja, ich sehe es schon. Ist eine Anpassung im Script... Das ist wohl beim refactoring untergegangen.
probiere bitte diese *.py nochmal aus: https://github.com/reserve85/HoymilesZeroExport/tree/dev

jimknopf63, 2024-02-21:
> ah ja, ich sehe es schon. Ist eine Anpassung im Script... Das ist wohl beim refactoring untergegangen. probiere bitte diese *.py nochmal aus: https://github.com/reserve85/HoymilesZeroExport/tree/dev

Danke, mit dieser Version funktioniert es wieder..... 
[image]

reserve85, 2024-02-21:
also sorry dafür, ich werde es nachher ins main übernehmen.

Gruß Tobi

jimknopf63, 2024-02-21:
> also sorry dafür, ich werde es nachher ins main übernehmen.
> 
> Gruß Tobi

Kein Problem, man kann ja nicht alle Möglichkeiten testen.... dafür sind zu viele vorhanden. Aber hier wurde mir ja jetzt wiederholt schnell geholfen. Vielen Dank für die Top Arbeit die hier geleistet wird. Gruß Stephan

closed 2024-02-21 by reserve85