reserve85/HoymilesZeroExport #148: OpenDTU v24.2.16 und HoymilesZeroExport wollen nicht miteinander

robson1311, 2024-02-24:
Hallo in die Runde!
Zur Info, zumindest war oder ist es bei mir so, dass die neueste Version (24.2.16) micht mit diesem Script kommuniziert.
Hatte einiges durchprobiert und nur ein Downgrade auf meine vorige Version (24.1.4) hat das ganze gefixt.

Wie gesagt, bei einem Update auf die neueste OpenDTU könnte es Probleme geben!
LG Robert

reserve85, 2024-02-24:
Korrekt, ihr müsst auf die neuste Script Version updaten dann funktioniert es auch mit der neuen OpenDTU Version.

robson1311, 2024-02-24:
Ok, habe jetzt die py und die ini auf die neueste geupdatet, leider funktionierts aber so nicht.
Fehlermeldung:
Feb 24 14:47:41 Nulleinspeisung python3[20900]: 2024-02-24 14:47:41 INFO     OpenDTU: Inverter "HM1200" reachable: True
Feb 24 14:47:41 Nulleinspeisung python3[20900]: 2024-02-24 14:47:41 ERROR    Exception at GetHoymilesPanelMinVoltage, Inverter 0 not reachable

Werde wohl eine neue Komplettinstallation brauchen. Bin einstweilen wieder zurück auf die alte openDTU.

reserve85, 2024-02-24:
Hast du das Script neu gestartet? Welche Version wird da angezeigt?

Esojma-Silverbullet, 2024-02-24:
Das hat bei mir auch nicht funktioniert, neue saubere Installation hat es gebracht.
Dauert doch max. 10 Minuten.

robson1311, 2024-02-25:
> Hast du das Script neu gestartet? Welche Version wird da angezeigt?

Version der ini 1.78, der py 1.79

robson1311, 2024-02-25:
> Das hat bei mir auch nicht funktioniert, neue saubere Installation hat es gebracht. Dauert doch max. 10 Minuten.

da muss ich aber den ordner "hoymileszereoexport" löschen, mal sehen wie das geht.

closed 2024-02-25 by reserve85