reserve85/HoymilesZeroExport #153: Docker Image Version

smaicloud, 2024-02-26:
Kannst du dem Docker Image auch die Versionsnummer geben anstatt main?

reserve85, 2024-02-27:
@tomquist ich hab da keine Ahnung von, kannst du das mal anschauen?

reserve85, 2024-02-28:
ja wenn du mir sagst wie ich das mache, dann wäre das für mich kein Problem.
Ich bin hier quasi nur Hobby-User

tomquist, 2024-02-29:
@reserve85 Ich habe einen PR erstellt. Sobald dieser gemerged ist, wird ein neuer Docker build angestoßen, sobald ein neuer Git tag erstellt (und gepushed) wird. Das Docker image wird dann immer sowohl mit dem Namen des Git-tags getagged, als auch mit `latest`.

reserve85, 2024-03-01:
schaut so als als würde das klappen

[image]

PR #155 (2024-03-01): Update workflow to run on tags
With this, docker images get built and tagged whenever a git tag is pushed.

Fixes #153
closed 2024-03-01 by reserve85