reserve85/HoymilesZeroExport #154: Adding HMS-xxxW support

julianxv, 2024-02-26:
Hallo,

wäre es möglich Support für die HMS-XXXW Geräte, also mit integriertem WIFI einzufügen? Es gibt schon eine Python Library zur Kommunikation damit. Das dazu passende Home Assistant Plugin läuft zuverlässig.

https://github.com/suaveolent/hoymiles-wifi

tomquist, 2024-02-27:
Oh wow, 2 requests pro Minute ist ganz schön einschränkend. Kann mir nicht vorstellen, dass man damit eine sinnvolle Nulleinspeisung hinbekommt, von dem her weiß ich nicht ob sich der Aufwand überhaupt lohnt.

mi-hol, 2024-04-18:
@reserve85 Damit die gleichartige Anfrage nicht immer wieder kommt, würde ich statt "[closed this as [completed]"vorschlagen mit Label "not planned" und Status "open" zu kennzeichnen

mi-hol, 2024-04-18:
Ab wieviel Requests/Min wird es denn sinnvoll?

> inwiefern? Bisher geht nur HM und HMS? Oder hab ich einen Denkfehler?

Also die meisten Anwender würden mMn annehmen, dass ein HMS-800W in der jetztigen Definition der "supported inverter" reinfällt. Nur ein technisch sehr versierter Anwender kann das unterscheiden.

reserve85, 2024-04-18:
ok, die readme hab ich da erweitert.

mit zwei Limitänderungen ist das viel zu träge, da musst du ja im worst case 29s warten bis du ein gesetztes Limit wieder erhöhen kannst wenn z.B. dein Ofen oder die Freitteuse anspringt. 
Aktuell gibt es da gar keine Einschränkung seitens der DTU, es kann sofort wieder das Limit neu gesetzt werden sobald die DTU "limit acknowledged" zurückmeldet.

reserve85, 2024-04-18:
Also 5s sind kein Problem.

mi-hol, 2024-04-18:
> mit zwei Limitänderungen ist das viel zu träge, da musst du ja im worst case 29s warten bis du ein gesetztes Limit wieder erhöhen kannst wenn z.B. dein Ofen oder die Fritteuse anspringt.

ist die typische Installation denn in der Lage einen 2kW Verbrauchssprung mit einem Balkonkraftwerk wegzupuffern?
Die ca 15 Installationen, die ich kenne, haben zu 90% weniger als **0,8kWp** Maximalleistung, so dass es praktisch bei einem 
 solchen Verbrauchssprung immer auf Netzbezug raus läuft.

reserve85, 2024-04-18:
Klar, aber du hast zumindest dann "sofort" 100% Limit um möglichst wenig Strom einzukaufen und nicht noch weitere, im schnitt 15s, Wartezeit bis dein Inverter reagiert...

Leuchte777, 2024-05-09:
Leistungssprung von 2KW sind kein Problem, wenn Du entsprechende Wechselrichter hast und einen Akku. Dann beziehst Du keinen Strom vom Versorger. Wenn der Ofen dann wieder ausgeht, wird der Akku wieder aufgeladen.

Ich würde dies aber nicht empfehlen, denn Backofen/Kochplatte/Bügeleisen haben den Nachteil, daß diese nur immer kurz an sind. Du speißt die ganze verbrauchte Leistung aus Deinem Akku ein, soweit OK. Aber kurz darauf sinkt der Verbrauch und Du speißt bis zum nächsten Intervall in das öffentliche Netz. Ergebnis: Du verschenkst Energie aus dem Akku

vandinter, 2024-05-28:
ich habe hier ( https://github.com/ohAnd/dtuGateway?tab=readme-ov-file#-experiences-with-the-hoymiles-HMS-800W-2T ) gelesen dass man quasi beliebig oft das limit setzen kann, „lediglich“ Werte LESEN geht nur alle 31 Sekunden.

reserve85, 2026-11-11:
Wie oben besprochen ist die Unterstützung der HMS-xxxW-Geräte mit integriertem WLAN nicht geplant. Ich schließe das Issue.

closed 2026-11-11 by reserve85