reserve85/HoymilesZeroExport #159: User/Password for Tasmota

jhedtmann, 2024-03-06:
Hi, my Tasmota (Hichi) power meter demands user/password when querying the API. I have augmented the config and the constructor accordingly. If that is a common issue this should maybe go into the main dev branch...

reserve85, 2024-03-06:
Hi,

currently user and password is not supported for tasmota webAPI, but it shouldn´t be a problem to implement it.
`http://<ip>/cm?user=<username>&password=<password>&cmnd=status%2010`
instead of 
`http://<ip>/cm?cmnd=status%2010`

i´ll add it soon and let you know.

jhedtmann, 2024-03-06:
I can create a PR, if that is of any help. Its really a quite trivial
modification...

Thanks for your efforts!

Best, Jörg

reserve85, 2024-03-06:
That would be great if you have already implemented it.

closed 2024-04-23 by reserve85