reserve85/HoymilesZeroExport #160: Regelung bei Lastspitzen

HA-IDIPS, 2024-03-06:
Skript für die Akkuregelung über Hoymiles-Wechselrichter: Wie sollte man die Config am besten für Klassiker wie Backofen, Kochherd oder Bügeleisen gestalten, die zwischendurch immer wieder kurzzeitig hohe Leistung für einige Sekunden anfordern, um die Temperatur zu halten, und dann wieder stark abfallen. Meist ist es so, dass das Skript die hohe Stromanforderung erkennt und dann gegenregelt (habe ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 100). Bis das allerdings durch den Wechselrichter umgesetzt ist (ca. 10-20 Sekunden), hat sich die Stromanforderung schon wieder erledigt. Dann läuft der WR auf 100%, aber leider ins Leere.

Gibt es Tipps, wie man da am besten die Konfiguration gestalten könnte?

Zweite Frage: Wenn der Akku immer wieder kurzzeitig hohe Lastspitzen gegenregeln muss (quasi von 0 auf 100) - ist das für LifePo4 Akkus und/oder Hoymiles WR schädlich?

reserve85, 2024-03-06:
Ja bei Akku würde ich den Jump definitiv deaktivieren. Ansonsten kannst du da relativ wenig machen, es wird eben immer in einem festen Intervall das Limit an den aktuellen Verbrauch angepasst. Ein Mittelwert oder ähnliches sehe ich da nicht zielführend… aber 10-20s sind doch viel zu viel? Mit ahoy kannst du das Intervall auf 5s runtersetzen. Das geht ratz fatz.

HA-IDIPS, 2024-03-06:
Ich setze OpenDTU für die Ansteuerung ein. Mit AhoyDTU wäre es demnach schneller?

Lässt sich noch etwas zu den kurzzeitigen Lastspitzen sagen, schädlich für LiFePo4 oder Hoymiles WR?

reserve85, 2024-03-07:
also bei mir geht das definitiv schneller als 10s, ich würde eher sagen sobald der Limit Befehl gesendet wurde vielleicht 1-2s...
Dem WR dürfte es nichts ausmachen, aber ich bin da nicht der Hersteller.
Akku habe ich keine, dazu kann ich nichts sagen.

closed 2024-04-15 by reserve85