reserve85/HoymilesZeroExport #163: Balkony Solarpower with Batterysolution

Burton1224, 2024-03-09:
Hello first a big thank you. 
Second is there a posibility to add script active time so the script starts itself and stops itself during the selected time frame?

Why Im asking:
I have 2 Solarmodules and a Battery the battery is one of the new Solutions which came up last year for balcony solar power grids. The story is my Hoymiles HM 600 has still control over the Solarmodules if I make this script the HM600 goes down to as an example 60W but the Battery is not charging because it still controlls the panels. And it starts charging above 150w but the problem is i almost never use 150W in my flat except PC, Cooking and so on. 
Also if the Hm600 selects 180W I would send 150W in the flat and judt 30W into the battery instead of the potential of what ever the sun is currently capable of. 
Thats why i would like to run the script just after sunset til sunrise.

reserve85, 2024-03-09:
Hi,

please have a look at this sites:
https://raspberrypi.stackexchange.com/questions/90099/execute-a-script-at-sunrise-sunset
https://github.com/mfreeborn/heliocron

With this you can set up a cronjob and enable / disable the ZeroExportService. I think that´s what you want.

Burton1224, 2024-03-09:
Danke werde mal rein schauen. Wird dabei der Wert vor Sonnenaufgang wieder zurück gesetze auf max oder etwas was man einstellt?

reserve85, 2024-03-11:
na da wird dann einfach das script neu gestartet, so wie du es in der config definiert hast.

Burton1224, 2024-03-11:
Ok. Ich habe zuhause die Anker Solix E1600 plus einen Hoymiles HM600 plus Module. 
Das Problem ist der Hoymiles muss zurück auf 600Watt gehen wenn er sozusagen die Nacht lang Strom liefert und dann das Script nicht mehr korrigiert aber der Hoymiles nie aufhört zu arbeiten fährt der ja einfach fort mit der letzten Einstellung die er bekommen hat. Und dann hat man eben das Problem, dass beim Anker Solix der MPPT des HM600 die Panele regelt und somit fährt er dann einfach so weiter mit z.B. 60W bis er den Strom verliert und der Speicher sich dadurch zurück setzt. Die E1600 wird also nie mehr geladen weil die erst ab 150W + geladen wird und da der Wechselrichter nur 60W "fordert" von den Modulen wird das nie überschritten. 
Habe jetzt eine eigene Lösung:
Euer super tolles Script startet mit Start des Raspberry Pi und läuft dann in der Nacht bis die Batterie alle ist. Sobald kein Strom mehr kommt von der Batterie schaltet sich das Pi aus der Wechselrichter geht schlafen und resetted sich so. Spätestens aber 8Uhr morgens geht das Pi sowiso schlafen. So bekomme ich die Nacht durch den Strom und den Tag durch einfach bis 150Watt. So ist es nicht 100 Prozent Nulleinspeisung aber fast. Bis dann Anker Solix die Probleme auf ihrer Seite mal lösen und dann sollte es 100 Prozent Nulleinspeisung sein.

Burton1224, 2024-03-11:
Ja wenns neu Startet aber dann arbeitet es ja wieder und wenn der Verbrauch z.B. 60 Watt ist regelt es bei mir auf 70Watt in die Wohnung. Das Problem wenn drausen Sonne für 400 Watt scheint und der Wechselrichter nur 70Watt anfordert sind die restlichen 330Watt futsch weil die nicht in die Batterie gehen sondern einfach nicht produziert werden. Das ist weil der MPPT des WR die Module regelt und nicht der dieser Batterie. Ist der Nachteil einer Fertiglösung vs selber ne Batterie bauen und dann in dein Skript integrieren.

closed 2024-04-15 by reserve85

Burton1224, 2024-04-15:
Dud you add a active time for the script now?

reserve85, 2024-04-15:
Ne, das ist so speziell - würde ich ungern implementieren. 
An deiner Stelle würde ich das wie oben geschrieben über ein cronjob machen und darin auch beim "ausschalten" des Scripts den CURL-Befehl "maxLimit" an deinen Inverter schicken.