reserve85/HoymilesZeroExport #171: No module named 'config_provider'

tester277, 2024-03-24:
Hallo und Danke für diese tolle Script,
Ich kann seit der Version 1.87 leider HoymilesZeroExport nicht ausführen.
Es erscheint die Fehlermeldung: No module named 'config_provider'
Entweder fehlt da eine Python Komponente oder meine Konfiguration (override) ist unvollständig.
 Mar 24 17:41:06 raspberrypi systemd[1]: Started HoymilesZeroExport Service.
Mar 24 17:41:07 raspberrypi python3[248469]: Traceback (most recent call last):
Mar 24 17:41:07 raspberrypi python3[248469]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 36, in <module>
Mar 24 17:41:07 raspberrypi python3[248469]:     from config_provider import ConfigFileConfigProvider, MqttConfigProvider, ConfigProviderChain
Mar 24 17:41:07 raspberrypi python3[248469]: ModuleNotFoundError: No module named 'config_provider'
Was könnte die Ursache sein?
Vielen Dank

tomquist, 2024-03-24:
Dir fehlt die Datei [config_provider.py](https://github.com/reserve85/HoymilesZeroExport/blob/main/config_provider.py)

tester277, 2024-03-24:
Danke - das war die Lösung. Die Datei hatte gefehlt: https://github.com/reserve85/HoymilesZeroExport/blob/main/config_provider.py

closed 2024-03-24 by tester277