reserve85/HoymilesZeroExport #173: Fehlermeldung

raffy1234, 2024-03-27:
Sehr ärgerlich, es lief schon mal aber jetzt nach Umzug auf openDTU komme ich nicht wirklich weiter.

Mar 27 00:05:21 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:24 raspberrypi python3[819]: Traceback (most recent call last):
Mar 27 00:05:24 raspberrypi python3[819]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
Mar 27 00:05:24 raspberrypi python3[819]:     from packaging import version
Mar 27 00:05:24 raspberrypi python3[819]: ModuleNotFoundError: No module named 'packaging'
Mar 27 00:05:24 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 00:05:24 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 00:05:24 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 1.
Mar 27 00:05:24 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:24 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:25 raspberrypi python3[834]: Traceback (most recent call last):
Mar 27 00:05:25 raspberrypi python3[834]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
Mar 27 00:05:25 raspberrypi python3[834]:     from packaging import version
Mar 27 00:05:25 raspberrypi python3[834]: ModuleNotFoundError: No module named 'packaging'
Mar 27 00:05:25 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 00:05:25 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 00:05:25 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 2.
Mar 27 00:05:25 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:25 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:26 raspberrypi python3[835]: Traceback (most recent call last):
Mar 27 00:05:26 raspberrypi python3[835]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
Mar 27 00:05:26 raspberrypi python3[835]:     from packaging import version
Mar 27 00:05:26 raspberrypi python3[835]: ModuleNotFoundError: No module named 'packaging'
Mar 27 00:05:26 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 00:05:26 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 00:05:26 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 3.
Mar 27 00:05:26 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:26 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:26 raspberrypi python3[837]: Traceback (most recent call last):
Mar 27 00:05:26 raspberrypi python3[837]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
Mar 27 00:05:26 raspberrypi python3[837]:     from packaging import version
Mar 27 00:05:26 raspberrypi python3[837]: ModuleNotFoundError: No module named 'packaging'
Mar 27 00:05:27 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 00:05:27 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 00:05:27 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 4.
Mar 27 00:05:27 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:27 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:27 raspberrypi python3[839]: Traceback (most recent call last):
Mar 27 00:05:27 raspberrypi python3[839]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
Mar 27 00:05:27 raspberrypi python3[839]:     from packaging import version
Mar 27 00:05:27 raspberrypi python3[839]: ModuleNotFoundError: No module named 'packaging'
Mar 27 00:05:27 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 00:05:27 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 00:05:28 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 5.
Mar 27 00:05:28 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 00:05:28 raspberrypi systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly.
Mar 27 00:05:28 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 00:05:28 raspberrypi systemd[1]: Failed to start HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 20:35:43 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 20:35:44 raspberrypi python3[3052]: Traceback (most recent call last):
Mar 27 20:35:44 raspberrypi python3[3052]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
Mar 27 20:35:44 raspberrypi python3[3052]:     from packaging import version
Mar 27 20:35:44 raspberrypi python3[3052]: ModuleNotFoundError: No module named 'packaging'
Mar 27 20:35:44 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 20:35:44 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 20:35:44 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 1.
Mar 27 20:35:44 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 20:35:44 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 20:35:44 raspberrypi python3[3055]: Traceback (most recent call last):
Mar 27 20:35:44 raspberrypi python3[3055]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
Mar 27 20:35:44 raspberrypi python3[3055]:     from packaging import version
Mar 27 20:35:44 raspberrypi python3[3055]: ModuleNotFoundError: No module named 'packaging'
Mar 27 20:35:44 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 20:35:44 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 20:35:45 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 2.
Mar 27 20:35:45 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 20:35:45 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 20:35:45 raspberrypi python3[3056]: Traceback (most recent call last):
[... 101 lines cut ...]
Mar 27 22:01:54 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 3.
Mar 27 22:01:54 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 22:01:54 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 22:01:54 raspberrypi python3[3212]: Traceback (most recent call last):
Mar 27 22:01:54 raspberrypi python3[3212]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
Mar 27 22:01:54 raspberrypi python3[3212]:     from packaging import version
Mar 27 22:01:54 raspberrypi python3[3212]: ModuleNotFoundError: No module named 'packaging'
Mar 27 22:01:54 raspberrypi systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Mar 27 22:01:54 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 22:01:55 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 4.
Mar 27 22:01:55 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 22:01:55 raspberrypi systemd[1]: Started HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 22:01:55 raspberrypi python3[3213]: Traceback (most recent call last):
Mar 27 22:01:55 raspberrypi python3[3213]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 33, in <module>
[... cut ...]
Mar 27 22:01:55 raspberrypi systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 5.
Mar 27 22:01:55 raspberrypi systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Mar 27 22:01:55 raspberrypi systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly.
Mar 27 22:01:55 raspberrypi systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Mar 27 22:01:55 raspberrypi systemd[1]: Failed to start HoymilesZeroExport.service - HoymilesZeroExport Service.

wo habe ich einen Fehler gemacht?

tomquist, 2024-03-27:
> No module named 'packaging'

Das deutet darauf hin, dass dir das Modul `packaging` fehlt. Versuche mal das:
```
pip3 install -r requirements.txt
```

raffy1234, 2024-03-28:
> > No module named 'packaging'
> 
> Das deutet darauf hin, dass dir das Modul `packaging` fehlt. Versuche mal das:
> 
> ```
> pip3 install -r requirements.txt
> ```

Hat leider nicht funktioniert. Ich werde wohl Python komplett neu installieren müssen?!

reserve85, 2024-03-28:
Muss gehen, hast du alle Dateien neu geladen?
 Ansonsten: 
pip3 install packaging

raffy1234, 2024-03-28:
> Muss gehen, hast du alle Dateien neu geladen? Ansonsten: pip3 install packaging

Danke für die Info. Ich werde es probieren wenn ich wieder zu Hause bin.

raffy1234, 2024-04-01:
Nach der Anleitung von lullatscho
https://github.com/reserve85/HoymilesZeroExport/issues/137 
hat es nun geklappt.
Vielen Dank für die tolle Hilfe hier.

lullatscho, 2024-04-01:
Super dann hat die Anleitung ja schon jemanden geholfen

closed 2024-04-15 by reserve85