reserve85/HoymilesZeroExport #174: Script regelt im Batteriemodus nicht hoch sobald die Spannung ansteigt

Janl1, 2024-04-01:
Hallo,

ich nutze das Script im Batteriemodus mit einem 24V LiFePo4 Akku und habe seit dem Upgrade auf die neueren Versionen das Problem, dass das Limit nicht von Reduce auf Normal angehoben wird, wenn die Spannung über HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V steigt.

Sobald ich das Script bei hoher Spannung neustarte, läuft die Steuerung wunderbar.

```
[COMMON]
INVERTER_COUNT = 1
SLOW_APPROX_LIMIT_IN_PERCENT = 20
SLOW_APPROX_FACTOR_IN_PERCENT = 20
LOOP_INTERVAL_IN_SECONDS = 5
SET_LIMIT_TIMEOUT_SECONDS = 5
POLL_INTERVAL_IN_SECONDS = 1
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 0
MAX_DIFFERENCE_BETWEEN_LIMIT_AND_OUTPUTPOWER = 100
ENABLE_LOG_TO_FILE = false
LOG_BACKUP_COUNT = 30
SET_LIMIT_RETRY = -1
LOG_TEMPERATURE = false
SET_POWER_STATUS_DELAY_IN_SECONDS = 1

[CONTROL]
POWERMETER_TARGET_POINT = -2
POWERMETER_TOLERANCE = 10
POWERMETER_MAX_POINT = 100

[INVERTER_1]
HOY_INVERTER_WATT = 800
HOY_MAX_WATT = 392
HOY_MIN_WATT_IN_PERCENT = 8
HOY_COMPENSATE_WATT_FACTOR = 1
HOY_BATTERY_MODE = true
HOY_BATTERY_THRESHOLD_OFF_LIMIT_IN_V = 25
HOY_BATTERY_THRESHOLD_REDUCE_LIMIT_IN_V = 25.5
HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V = 25.7
HOY_BATTERY_NORMAL_WATT = 390
HOY_BATTERY_REDUCE_WATT = 150
HOY_BATTERY_THRESHOLD_ON_LIMIT_IN_V = 25.5
```

HOY_MAX_WATT = 392 ist absichtlich so gewählt, da offenbar bei 24V Akkus der Hoymiles ab einer gewissen Leistung auf 100 % und sich nicht mehr gescheit drosseln lässt.

```
python3[1438346]: 2024-04-01 17:33:30 INFO     Average min-panel voltage, inverter "Balkon": 26.080000304000002 Volt
python3[1438346]: 2024-04-01 17:33:30 INFO     Retry Counter exceeded: Inverter PowerStatus already ON
python3[1438346]: 2024-04-01 17:33:30 INFO     powermeter Tasmota: 211 Watt
python3[1438346]: 2024-04-01 17:33:30 INFO     Inverterlimit was already accepted at 150 Watt
python3[1438346]: 2024-04-01 17:33:35 INFO     OpenDTU: Inverter "Balkon" reachable: True
python3[1438346]: 2024-04-01 17:33:35 INFO     Average min-panel voltage, inverter "Balkon": 26.120000456 Volt
python3[1438346]: 2024-04-01 17:33:35 INFO     Retry Counter exceeded: Inverter PowerStatus already ON
python3[1438346]: 2024-04-01 17:33:35 INFO     powermeter Tasmota: 211 Watt
python3[1438346]: 2024-04-01 17:33:35 INFO     Inverterlimit was already accepted at 150 Watt
python3[1438346]: 2024-04-01 17:33:40 INFO     OpenDTU: Inverter "Balkon" reachable: True
python3[1438346]: 2024-04-01 17:33:40 INFO     Average min-panel voltage, inverter "Balkon": 26.140000532000002 Volt
python3[1438346]: 2024-04-01 17:33:40 INFO     Retry Counter exceeded: Inverter PowerStatus already ON
python3[1438346]: 2024-04-01 17:33:40 INFO     powermeter Tasmota: 824 Watt
python3[1438346]: 2024-04-01 17:33:40 INFO     Inverterlimit was already accepted at 150 Watt
python3[1438346]: 2024-04-01 17:33:46 INFO     OpenDTU: Inverter "Balkon" reachable: True
python3[1438346]: 2024-04-01 17:33:46 INFO     Average min-panel voltage, inverter "Balkon": 26.160000608 Volt
python3[1438346]: 2024-04-01 17:33:46 INFO     Retry Counter exceeded: Inverter PowerStatus already ON
python3[1438346]: 2024-04-01 17:33:46 INFO     powermeter Tasmota: 422 Watt
python3[1438346]: 2024-04-01 17:33:46 INFO     Inverterlimit was already accepted at 150 Watt
python3[1438346]: 2024-04-01 17:33:51 INFO     OpenDTU: Inverter "Balkon" reachable: True
python3[1438346]: 2024-04-01 17:33:51 INFO     Average min-panel voltage, inverter "Balkon": 26.180000684000003 Volt
python3[1438346]: 2024-04-01 17:33:51 INFO     Retry Counter exceeded: Inverter PowerStatus already ON
python3[1438346]: 2024-04-01 17:33:51 INFO     powermeter Tasmota: 440 Watt
python3[1438346]: 2024-04-01 17:33:51 INFO     Inverterlimit was already accepted at 150 Watt
python3[1438346]: 2024-04-01 17:33:56 INFO     OpenDTU: Inverter "Balkon" reachable: True
python3[1438346]: 2024-04-01 17:33:56 INFO     Average min-panel voltage, inverter "Balkon": 26.20000076 Volt
python3[1438346]: 2024-04-01 17:33:56 INFO     Retry Counter exceeded: Inverter PowerStatus already ON
python3[1438346]: 2024-04-01 17:33:56 INFO     powermeter Tasmota: 430 Watt
python3[1438346]: 2024-04-01 17:33:56 INFO     Inverterlimit was already accepted at 150 Watt
python3[1438346]: 2024-04-01 17:34:01 INFO     OpenDTU: Inverter "Balkon" reachable: True
python3[1438346]: 2024-04-01 17:34:01 INFO     Average min-panel voltage, inverter "Balkon": 26.20000076 Volt
python3[1438346]: 2024-04-01 17:34:01 INFO     Retry Counter exceeded: Inverter PowerStatus already ON
python3[1438346]: 2024-04-01 17:34:01 INFO     powermeter Tasmota: 431 Watt
python3[1438346]: 2024-04-01 17:34:01 INFO     Inverterlimit was already accepted at 150 Watt
python3[1438346]: 2024-04-01 17:34:06 INFO     OpenDTU: Inverter "Balkon" reachable: True
python3[1438346]: 2024-04-01 17:34:06 INFO     Average min-panel voltage, inverter "Balkon": 26.20000076 Volt
python3[1438346]: 2024-04-01 17:34:06 INFO     Retry Counter exceeded: Inverter PowerStatus already ON
python3[1438346]: 2024-04-01 17:34:06 INFO     powermeter Tasmota: 431 Watt
python3[1438346]: 2024-04-01 17:34:06 INFO     Inverterlimit was already accepted at 150 Watt
python3[1438346]: 2024-04-01 17:34:11 INFO     OpenDTU: Inverter "Balkon" reachable: True
python3[1438346]: 2024-04-01 17:34:11 INFO     Average min-panel voltage, inverter "Balkon": 26.20000076 Volt
python3[1438346]: 2024-04-01 17:34:11 INFO     Retry Counter exceeded: Inverter PowerStatus already ON
python3[1438346]: 2024-04-01 17:34:11 INFO     powermeter Tasmota: 423 Watt
python3[1438346]: 2024-04-01 17:34:11 INFO     Inverterlimit was already accepted at 150 Watt
python3[1438346]: 2024-04-01 17:34:17 INFO     OpenDTU: Inverter "Balkon" reachable: True
python3[1438346]: 2024-04-01 17:34:17 INFO     Average min-panel voltage, inverter "Balkon": 26.20000076 Volt
python3[1438346]: 2024-04-01 17:34:17 INFO     Retry Counter exceeded: Inverter PowerStatus already ON
python3[1438346]: 2024-04-01 17:34:17 INFO     powermeter Tasmota: 424 Watt
python3[1438346]: 2024-04-01 17:34:17 INFO     Inverterlimit was already accepted at 150 Watt
python3[1438346]: 2024-04-01 17:34:22 INFO     OpenDTU: Inverter "Balkon" reachable: True
python3[1438346]: 2024-04-01 17:34:22 INFO     Average min-panel voltage, inverter "Balkon": 26.20000076 Volt
python3[1438346]: 2024-04-01 17:34:22 INFO     Retry Counter exceeded: Inverter PowerStatus already ON
python3[1438346]: 2024-04-01 17:34:22 INFO     powermeter Tasmota: 415 Watt
python3[1438346]: 2024-04-01 17:34:22 INFO     Inverterlimit was already accepted at 150 Watt
python3[1438346]: 2024-04-01 17:34:27 INFO     OpenDTU: Inverter "Balkon" reachable: True
python3[1438346]: 2024-04-01 17:34:27 INFO     Average min-panel voltage, inverter "Balkon": 26.20000076 Volt
python3[1438346]: 2024-04-01 17:34:27 INFO     Retry Counter exceeded: Inverter PowerStatus already ON
python3[1438346]: 2024-04-01 17:34:27 INFO     powermeter Tasmota: 413 Watt
python3[1438346]: 2024-04-01 17:34:27 INFO     Inverterlimit was already accepted at 150 Watt
python3[1438346]: 2024-04-01 17:34:32 INFO     OpenDTU: Inverter "Balkon" reachable: True
```

reserve85, 2024-04-01:
Hi,

ja denke es liegt daran:

GetBatteryNormalWatt:
if normal_watt > HOY_MAX_WATT[pInverter]:
        normal_watt = HOY_MAX_WATT[pInverter]
        
Ich bin leider noch 2 Tage unterwegs, vielleicht kann es @tomquist mal anschauen falls du dazu kommst? Ich denke HOY_MAX_WATT steht noch auf 150W und wird daher nicht erhöht..,

reserve85, 2024-04-02:
wäre super wenn du diesen Branch testen könntest:

https://github.com/reserve85/HoymilesZeroExport/tree/HOY_BATTERY_THRESHOLD_NORMAL_LIMIT_IN_V

Janl1, 2024-04-02:
Besten Dank für die schnelle Rückmeldung.

Die nächsten drei Tage sieht es etwas mager aus mit Sonne, aber ich gebe Bescheid!

Janl1, 2024-04-02:
Die Sonne hat sich doch noch mal blicken lassen und es läuft wieder alles wie es soll. Noch einmal vielen Dank für den schnellen Fix,

closed 2024-04-02 by reserve85