reserve85/HoymilesZeroExport #178: Home Assistant API nicht erreichbar über HTTPS

in-famous-raccoon, 2024-04-13:
Hallo,

vielen Dank erstmal für das klasse Skript! Funktioniert sehr gut.

Ich hatte bei dem Einrichten aber leider start Schwierigkeiten weil ich Home-Assistant über https eingerichtet habe. Du hast in deinen Script aber http fest eingestellt: 
```
778    def GetJson(self, path):
779       url = f"http://{self.ip}:{self.port}{path}"
780       headers = {"Authorization": "Bearer " + self.access_token, "content-type": "application/json"}
781       return session.get(url, headers=headers, timeout=10).json()
```

Ich habe die url einfach auf https ( url = f"https://{self.ip}:{self.port}{path}" )gesetzt aber vielleicht wäre ein kleiner schalter bei der Home-Assistant API da sinnvoller für die Zukunft. Ich meine in etwa so:
```
# --- defines for HOME ASSISTANT ---
HA_IP = 10.1.0.5
HA_PORT = 14813
HA_HTTPS = true|false
```

Viele Grüße,
Basti

reserve85, 2024-04-13:
Nehm ich mit rein beim nächsten update.

in-famous-raccoon, 2024-04-22:
Vielen Dank für die schnelle Umsetzung! Habe soeben V1.91 getestet mit dem neuen Parameter "HA_HTTPS = true" und die Verbindung über HTTPS klappt jetzt sofort!

closed 2024-04-22 by in-famous-raccoon