reserve85/HoymilesZeroExport #179: Fehlermeldung nach update

fatzo, 2024-04-17:
nach dem update habe ich diese fehlermeldung

2024-04-17 13:49:26 INFO     Config file V 1.89
Apr 17 13:49:26 raspberrypi5 python3[161782]: Traceback (most recent call last):
Apr 17 13:49:26 raspberrypi5 python3[161782]:   File "/usr/local/lib/python3.9/configparser.py", line 789, in get
Apr 17 13:49:26 raspberrypi5 python3[161782]:     value = d[option]
Apr 17 13:49:26 raspberrypi5 python3[161782]:   File "/usr/local/lib/python3.9/collections/__init__.py", line 941, in __getitem__
Apr 17 13:49:26 raspberrypi5 python3[161782]:     return self.__missing__(key)            # support subclasses that define __missing__
Apr 17 13:49:26 raspberrypi5 python3[161782]:   File "/usr/local/lib/python3.9/collections/__init__.py", line 933, in __missing__
Apr 17 13:49:26 raspberrypi5 python3[161782]:     raise KeyError(key)
Apr 17 13:49:26 raspberrypi5 python3[161782]: KeyError: 'tasmota_json_power_mqtt_label'
Apr 17 13:49:26 raspberrypi5 python3[161782]: During handling of the above exception, another exception occurred:
Apr 17 13:49:26 raspberrypi5 python3[161782]: Traceback (most recent call last):
Apr 17 13:49:26 raspberrypi5 python3[161782]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 1295, in <module>
Apr 17 13:49:26 raspberrypi5 python3[161782]:     POWERMETER = CreatePowermeter()
Apr 17 13:49:26 raspberrypi5 python3[161782]:   File "/home/pi/HoymilesZeroExport/HoymilesZeroExport.py", line 1124, in CreatePowermeter
Apr 17 13:49:26 raspberrypi5 python3[161782]:     config.get('TASMOTA', 'TASMOTA_JSON_POWER_MQTT_LABEL'),
Apr 17 13:49:26 raspberrypi5 python3[161782]:   File "/usr/local/lib/python3.9/configparser.py", line 792, in get
Apr 17 13:49:26 raspberrypi5 python3[161782]:     raise NoOptionError(option, section)
Apr 17 13:49:26 raspberrypi5 python3[161782]: configparser.NoOptionError: No option 'tasmota_json_power_mqtt_label' in section: 'TASMOTA'
Apr 17 13:49:26 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Apr 17 13:49:26 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Apr 17 13:49:26 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 5.
Apr 17 13:49:26 raspberrypi5 systemd[1]: Stopped HoymilesZeroExport.service - HoymilesZeroExport Service.
Apr 17 13:49:26 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly.
Apr 17 13:49:26 raspberrypi5 systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Apr 17 13:49:26 raspberrypi5 systemd[1]: Failed to start HoymilesZeroExport.service - HoymilesZeroExport Service.

hab ich da etwas zerschossen?

gruß fatzo

tomquist, 2024-04-17:
Dir fehlt die Datei configparser.py

reserve85, 2024-04-18:
Oder er findet den key tasmota_json_power_mqtt_label nicht??

Zeig mal bitte deine config_override…

fatzo, 2024-04-19:
Wie immer vorbildlicher Support hier

closed 2024-04-19 by fatzo