reserve85/HoymilesZeroExport #181: Shelly em value problem 2 probes

pierre673, 2024-04-21:
Good morning
I installed your script with a shelly em 2 probes.
1 probe for consumption coming from the network and 1 probe for the production of solar panels.
the problem is that the script adds the 2 values.
example :
my network indicates -300w on probe 1 and 1500w production on probe 2. the value taken by the script is 1200w.
It should only take the value of probe 1.
Is there a solution ?
THANKS

reserve85, 2024-04-21:
Hi,

good idea, i´ll take a look into this topic in the next days.

reserve85, 2024-04-22:
i added it into the "dev" branch: https://github.com/reserve85/HoymilesZeroExport/tree/dev
would be nice if you can test it before i merge it into the main branch?

pierre673, 2024-04-22:
Great, it works.
Just wait for the sun.
Now I have to figure out how to install it on my docker NAS.
Because for the moment the script works on a Linux virtual machine.

reserve85, 2024-04-22:
perfect, i´ll copy it into the main branch!

closed 2024-04-23 by reserve85