reserve85/HoymilesZeroExport #182: DDSU666 Smart Meter Support

Wind06, 2024-04-22:
Hello,

My config is as follow:
Raspberry
USB to RS485 Converter
DDSU666
AhoyDTU

- The USB is connected to the DDSU666 using the rs485 port.
- I can get DDSU666  data using the following septum
      Modbus - method='RTU', port='/dev/ttyUSB0', timeout=1, baudrate=9600, stopbits=2, bytesize=8, parity='N'
- I do have a script that report such value
- The DDSU666 report negative power value in case of returning to the grid

I  am trying to integrate this smart meter, but to be honnest I don't know how to proceed.
Is this feasible ? can someone help ?

Many thanks

reserve85, 2024-04-22:
Hi,
is it possible for you to copy and modify your script like this one: https://github.com/reserve85/HoymilesZeroExport/blob/main/GetPowerFromVictronMultiplus.sh

This is also based on modbus and provides the total power of victron multiplus...

Wind06, 2024-04-22:
Thanks for your answer, let me give a try and I will let you know.

reserve85, 2024-04-23:
Hi,
So it is inverted? Then you´ll need to define that in the custom script (i.e. GetPowerFromDDSU666.sh).

Wind06, 2024-04-23:
Hi, well I do have some issues with mbpoll command, my script is using pymodbus much easier. Anyway give me sometimes and I will succeed

reserve85, 2024-04-24:
Alright, mbpoll is for command line and pymodbus for python. So these are two different things.
Unfortunately I do not have any Modbus devices, so i really can´t help you...

Wind06, 2024-04-24:
Hello,

Well I do have issues with mbpoll communication is not working within my environment. 
however I succeed with **modpoll** some more tests required. If I provide a shell with modpoll instead of mbpoll, is that ok ?

otherwise I have to understand way mbpoll is not working for me.

Wind06, 2024-04-26:
Hi, I finally created a python script using pymodbus for the communication. I had communication issues with mbpoll or modpoll. Hope that's not an issue. attached is the script which is working within my environment. 
I added .txt because I was not able to include .py file directly. I further information requiered let me know.
[GetPowerFromDDSU666.py.txt](https://github.com/reserve85/HoymilesZeroExport/files/15129585/GetPowerFromDDSU666.py.txt)

reserve85, 2024-04-26:
perfect, please try this: 
https://github.com/reserve85/HoymilesZeroExport/tree/dev_DDSU666
i had to make a few adjustments so that there were no compiling errors for me. Can you check it please?

Wind06, 2024-04-26:
Thanks sure I will test it and let you know, however today it’s raining so no overproduction. Tomorrow should be better, I will keep you posted during the weekend. Thanks

reserve85, 2024-04-26:
Ok, but it’s „only“ the Powermeter value, should be independent of the weather…

Wind06, 2024-04-26:
Hello, I do have an error at start attached the log file 

avril 26 17:13:52 rpiB3 python3[106265]: 2024-04-26 17:13:52 ERROR    Exception at GetPowermeterWatts
avril 26 17:13:52 rpiB3 python3[106265]: 2024-04-26 17:13:52 ERROR    cannot import name 'ModbusSerialClient' from 'pymodbus.client' (/usr/local/lib/python3.9/dist-packages/pymodbus/client/__init__.py)

[GetPowerFromDDSU666.log](https://github.com/reserve85/HoymilesZeroExport/files/15132441/GetPowerFromDDSU666.log)

Wind06, 2024-04-26:
> Hello, I do have an error at start attached the log file
>
> avril 26 17:13:52 rpiB3 python3[106265]: 2024-04-26 17:13:52 ERROR Exception at GetPowermeterWatts avril 26 17:13:52 rpiB3 python3[106265]: [...]

I found the issue in the GetPowerFromDDSU666.py you updated the first line 
from **pymodbus.client** import ModbusSerialClient as ModbusClient
maybe a typo
I change it back to  from **pymodbus.client.sync** import ModbusSerialClient as ModbusClient
not the module is starting correctly.

reserve85, 2024-04-26:
Ok, I think this depends on the version of python? I needed to change that because your import line raised an error at my system… I’ll take a look tomorrow

Wind06, 2024-04-26:
ok perfect, for info
python --version
Python 3.9.2

reserve85, 2024-04-26:
Which version of pymodbus do you have installed?

pip show pymodbus

Wind06, 2024-04-26:
Name: pymodbus
Version: 2.4.0

reserve85, 2024-04-27:
Ok, that’s really old, I think that is the compatibility issue to my code. 

Can you try installing a newer version and check my code again?

pip install pymodbus==3.6.8

Wind06, 2024-04-27:
Ok sure, I updated my system, not I do have issues with my script. let me figure this out and will comme back to you asap

reserve85, 2024-04-27:
What error do you get? Perhaps I can help you.

Wind06, 2024-04-27:
Hi, thanks for the offer the issue is on the modbus side because working with version 2.xx and not 3.xx
I need to check what has been changed when reading and decoding received values with modbus V3
I will have time to check this tomorrow.

The error I got  is the following:

Erreur lors de la lecture des registres: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)

Wind06, 2024-04-28:
Hi, sorry for my very late answer, I was quite busy today . 
Anyway, find a .sh script, I finally managed the mbpoll command.
I dropped pymodbus because of many changes in between versions.
[GetPowerFromDDSU666.sh.txt](https://github.com/reserve85/HoymilesZeroExport/files/15142939/GetPowerFromDDSU666.sh.txt)

reserve85, 2024-04-28:
Ok, so this new script is already working with the current main Version of hoymileszeroexport? Then I would add it to the repository?

Wind06, 2024-04-28:
sounds good I will test it and will let you know. thanks

Wind06, 2024-04-29:
I am quite sure that I am doing some stupide things but I get each time an error message 
[Errno 2] No such file or directory: 'GetPowerFromDDSU666.sh' the script is in the HoymilesZeroExport directory.
Thanks for and help.

the HoymilesZeroExport_Config_Override.ini
-------------------
[VERSION]
VERSION = 1.91

[SELECT_DTU]
USE_AHOY = true

[SELECT_POWERMETER]
USE_SCRIPT = true

[AHOY_DTU]
AHOY_IP = 192.168.1.42

[SCRIPT]
SCRIPT_IP = 77
SCRIPT_FILE = GetPowerFromDDSU666.sh

--------------------
the log is as below
-------------------

Log write to file: False
Apr 29 09:35:47 rpiB3 python3[21703]: 2024-04-29 09:35:47 INFO     Python Version: 3.11.2 (main, Mar 13 2023, 12:18:29) [GCC 12.2.0]
Apr 29 09:35:47 rpiB3 python3[21703]: 2024-04-29 09:35:47 INFO     Author: Tobias Kraft / Script Version: 1.91
Apr 29 09:35:47 rpiB3 python3[21703]: 2024-04-29 09:35:47 INFO     read config file: /home/erik/HoymilesZeroExport/HoymilesZeroExport_Config.ini
Apr 29 09:35:47 rpiB3 python3[21703]: 2024-04-29 09:35:47 INFO     read additional config file: /home/erik/HoymilesZeroExport/HoymilesZeroExport_Config_Override.ini
Apr 29 09:35:47 rpiB3 python3[21703]: 2024-04-29 09:35:47 INFO     Config file V 1.91
Apr 29 09:35:47 rpiB3 python3[21703]: 2024-04-29 09:35:47 INFO     ---Init---
Apr 29 09:35:47 rpiB3 python3[21703]: 2024-04-29 09:35:47 INFO     Ahoy: Current Version: 0.8.83
Apr 29 09:35:47 rpiB3 python3[21703]: 2024-04-29 09:35:47 INFO     Ahoy: Inverter "yet unknown" Available: True
Apr 29 09:35:48 rpiB3 python3[21703]: 2024-04-29 09:35:48 INFO     Ahoy: Inverter "HM800" / serial number "114190556486" / temperature 28.8 degC
Apr 29 09:35:48 rpiB3 python3[21703]: 2024-04-29 09:35:48 INFO     Ahoy: Inverter "HM800": Turn on
Apr 29 09:35:58 rpiB3 python3[21703]: 2024-04-29 09:35:58 INFO     setting new limit to 75 Watt
Apr 29 09:35:58 rpiB3 python3[21703]: 2024-04-29 09:35:58 INFO     Ahoy: Inverter "HM800": setting new limit from -1 Watt to 75 Watt
Apr 29 09:35:59 rpiB3 python3[21703]: 2024-04-29 09:35:59 INFO     Ahoy: Inverter "HM800": Limit acknowledged
Apr 29 09:35:59 rpiB3 python3[21703]: 2024-04-29 09:35:59 INFO     intermediate meter AhoyDTU: 72 Watt
Apr 29 09:35:59 rpiB3 python3[21703]: 2024-04-29 09:35:59 ERROR    Exception at GetPowermeterWatts
Apr 29 09:35:59 rpiB3 python3[21703]: 2024-04-29 09:35:59 ERROR    [Errno 2] No such file or directory: 'GetPowerFromDDSU666.sh'

reserve85, 2024-04-29:
Can you start the GetPowerFromDDSU666.sh script from the same directory?

Wind06, 2024-04-29:
Yes I can and get correct output

reserve85, 2024-04-29:
is your file-access missing?
can you try setting:
sudo chmod +x GetPowerFromDDSU666.sh

reserve85, 2024-04-29:
got it:

[SCRIPT]
SCRIPT_IP = 77
SCRIPT_FILE = /home/pi/HoymilesZeroExport/GetPowerFromDDSU666.sh

you need to provide the full path

Wind06, 2024-04-29:
you are right so stupide I am :-(, it's working I did not get the error message. 
I will continue testing tomorrow.

good night,
take care

Wind06, 2024-04-30:
Hello, so I finalised the .sh script 

- Reading register 8196: output format is in KW like "[8196]: 	-0.4397"
- Extract and Convert in W: -439
- Reverse the sign "**+ to -** " and "**- to +**" because of DDSU666 Hoymiles. Not really sure about other DDSU666     Version. 
 DDSU666 hoymiles is working like: to the Grid **+** from the Grid **-** 

Hope that the process is ok with HoymilesZeroExport, if I am wrong please let me know.
I am currently waiting for sunny days to fully test it :-)

reserve85, 2024-05-01:
When it’s working it would be nice from you to create a pull request to add the file to the repository.

Wind06, 2024-05-01:
Sure I will.

Wind06, 2024-05-01:
Script seems to be  ok,  I created a pull request on the DDSU666 development branch. 
in case off I attach the script.
[GetPowerFromDDSU666.sh.txt](https://github.com/reserve85/HoymilesZeroExport/files/15175951/GetPowerFromDDSU666.sh.txt)

closed 2024-05-04 by Wind06