reserve85/HoymilesZeroExport #184: http request and json answer string

kthemall, 2024-04-25:
hi,

my question is: 

is it possible to get the consumption directly from a http request for example:

http://x.x.x.x/rest which gives back this json string and my consumption saldo is called saldo:

{"1_8_0":27424805,"2_8_0":70703,"3_8_1":1429460,"4_8_1":3086146,"1_7_0":501,"2_7_0":507,"3_7_0":0,"4_7_0":256,"1_128_0":0,"saldo":-6}

maybe it is and i do not know where to config it.

thx tom

kthemall, 2024-04-29:
hi, i use a custom made reader [https://www.mitterbaur.at/amis-leser.html](url) which works very fine for Siemens TD-3511/TD-3512 Smartmeter.

i just got the OK from the Developer and we are welcomed to implement it here in Hoymileszeroexport. thx tom

reserve85, 2024-04-29:
https://github.com/reserve85/HoymilesZeroExport/tree/dev_AMIS_Reader should do the trick, can you test it?

kthemall, 2024-04-29:
just tried but did not work:

kthemall@iobroker:~/HoymilesZeroExport $ sudo ./update.sh dev_AMIS_Reader
Start update of HoymilesZeroExport
--2024-04-29 12:26:40--  https://github.com/reserve85/HoymilesZeroExport/archive/refs/heads/dev_AMIS_Reader.zip
Aufl▒sen des Hostnamens github.com (github.com)... 140.82.121.4
Verbindungsaufbau zu github.com (github.com)|140.82.121.4|:443 ... verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet ... 302 Found
Platz: https://codeload.github.com/reserve85/HoymilesZeroExport/zip/refs/heads/dev_AMIS_Reader [folgend]
--2024-04-29 12:26:40--  https://codeload.github.com/reserve85/HoymilesZeroExport/zip/refs/heads/dev_AMIS_Reader
Aufl▒sen des Hostnamens codeload.github.com (codeload.github.com)... 140.82.121.9
Verbindungsaufbau zu codeload.github.com (codeload.github.com)|140.82.121.9|:443 ... verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet ... 200 OK
L▒nge: nicht spezifiziert [application/zip]
Wird in ▒dev_AMIS_Reader.zip▒ gespeichert.

dev_AMIS_Reader.zip                        [ <=>                                                                      ] 196,70K  --.-KB/s    in 0,1s

2024-04-29 12:26:41 (1,80 MB/s) - ▒dev_AMIS_Reader.zip▒ gespeichert [201418]

Archive:  dev_AMIS_Reader.zip
59608c445330944df3f592ec4a790a117a07a8d2
   creating: HoymilesZeroExport-dev_AMIS_Reader/
   creating: HoymilesZeroExport-dev_AMIS_Reader/.github/
   creating: HoymilesZeroExport-dev_AMIS_Reader/.github/workflows/
  inflating: HoymilesZeroExport-dev_AMIS_Reader/.github/workflows/ci.yml
  inflating: HoymilesZeroExport-dev_AMIS_Reader/CHANGELOG.md
  inflating: HoymilesZeroExport-dev_AMIS_Reader/Dockerfile
  inflating: HoymilesZeroExport-dev_AMIS_Reader/GetPowerFromVictronMultiplus.sh
  inflating: HoymilesZeroExport-dev_AMIS_Reader/HoymilesZeroExport.py
  inflating: HoymilesZeroExport-dev_AMIS_Reader/HoymilesZeroExport_Config.ini
  inflating: HoymilesZeroExport-dev_AMIS_Reader/LICENSE
  inflating: HoymilesZeroExport-dev_AMIS_Reader/README.md
  inflating: HoymilesZeroExport-dev_AMIS_Reader/config_provider.py
  inflating: HoymilesZeroExport-dev_AMIS_Reader/install.sh
  inflating: HoymilesZeroExport-dev_AMIS_Reader/requirements.txt
   creating: HoymilesZeroExport-dev_AMIS_Reader/res/
  inflating: HoymilesZeroExport-dev_AMIS_Reader/res/HA_Comparison_1.png
  inflating: HoymilesZeroExport-dev_AMIS_Reader/res/HA_Comparison_2.png
  inflating: HoymilesZeroExport-dev_AMIS_Reader/restart.sh
  inflating: HoymilesZeroExport-dev_AMIS_Reader/uninstall_service.sh
  inflating: HoymilesZeroExport-dev_AMIS_Reader/update.sh
Installing packages
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: certifi==2023.11.17 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 1)) (2023.11.17)
Requirement already satisfied: charset-normalizer==3.3.2 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 2)) (3.3.2)
Requirement already satisfied: idna==3.4 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 3)) (3.4)
Requirement already satisfied: packaging==23.2 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 4)) (23.2)
Requirement already satisfied: requests==2.31.0 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 5)) (2.31.0)
Requirement already satisfied: urllib3==2.1.0 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 6)) (2.1.0)
Requirement already satisfied: paho-mqtt==2.0.0 in /usr/local/lib/python3.9/dist-packages (from -r requirements.txt (line 7)) (2.0.0)
Packages install completed
Uninstall HoymilesZeroExport.service
Removed /etc/systemd/system/multi-user.target.wants/HoymilesZeroExport.service.
Uninstallation of HoymilesZeroExport.service completed
[Unit]
Description=HoymilesZeroExport Service
After=multi-user.target
[Service]
Type=simple
Restart=always
ExecStart=/usr/bin/python3 /home/kthemall/HoymilesZeroExport/HoymilesZeroExport.py -c /home/kthemall/HoymilesZeroExport/HoymilesZeroExport_Config_Override.ini
[Install]
WantedBy=multi-user.target
Created symlink /etc/systemd/system/multi-user.target.wants/HoymilesZeroExport.service -> /etc/systemd/system/HoymilesZeroExport.service.
* HoymilesZeroExport.service - HoymilesZeroExport Service
     Loaded: loaded (/etc/systemd/system/HoymilesZeroExport.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2024-04-29 12:26:55 CEST; 38ms ago
   Main PID: 20498 (python3)
      Tasks: 1 (limit: 1595)
        CPU: 28ms
     CGroup: /system.slice/HoymilesZeroExport.service
             `-20498 /usr/bin/python3 /home/kthemall/HoymilesZeroExport/HoymilesZeroExport.py -c /home/kthemall/HoymilesZeroExport/HoymilesZeroExport_Co>

Apr 29 12:26:55 iobroker systemd[1]: Started HoymilesZeroExport Service.

installation i think was fine then i put that in my **override** file:

[SELECT_POWERMETER]
#USE_SHELLY_3EM_PRO = true
USE_AMIS_READER = true

[AMIS_READER]
AMIS_READER_IP = 192.168.188.84

it does not change the power any longer. hmmm

reserve85, 2024-04-29:
can you check the log?

sudo journalctl -u HoymilesZeroExport.service -n 2000 -e -f

kthemall, 2024-04-29:
Apr 29 12:46:05 iobroker python3[20861]: 2024-04-29 12:46:05 INFO     Config file V 1.92
Apr 29 12:46:05 iobroker python3[20861]: Traceback (most recent call last):
Apr 29 12:46:05 iobroker python3[20861]:   File "/home/kthemall/HoymilesZeroExport/HoymilesZeroExport.py", line 1336, in <module>
Apr 29 12:46:05 iobroker python3[20861]:     POWERMETER = CreatePowermeter()
Apr 29 12:46:05 iobroker python3[20861]:   File "/home/kthemall/HoymilesZeroExport/HoymilesZeroExport.py", line 1203, in CreatePowermeter
Apr 29 12:46:05 iobroker python3[20861]:     return VZLogger(
Apr 29 12:46:05 iobroker python3[20861]: TypeError: __init__() missing 2 required positional arguments: 'port' and 'uuid'
Apr 29 12:46:05 iobroker systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Apr 29 12:46:05 iobroker systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Apr 29 12:46:05 iobroker systemd[1]: HoymilesZeroExport.service: Consumed 1.350s CPU time.
Apr 29 12:46:05 iobroker systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 5.
Apr 29 12:46:05 iobroker systemd[1]: Stopped HoymilesZeroExport Service.
Apr 29 12:46:05 iobroker systemd[1]: HoymilesZeroExport.service: Consumed 1.350s CPU time.
Apr 29 12:46:05 iobroker systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly.
Apr 29 12:46:05 iobroker systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Apr 29 12:46:05 iobroker systemd[1]: Failed to start HoymilesZeroExport Service.

reserve85, 2024-04-29:
ah ja, fehler von mir, probiere jetzt nochmal bitte

kthemall, 2024-04-29:
muss ein intermediate auch eingetragen sein???

 2024-04-29 12:52:23 INFO     Config file V 1.92
Apr 29 12:52:23 iobroker python3[20974]: Traceback (most recent call last):
Apr 29 12:52:23 iobroker python3[20974]:   File "/usr/lib/python3.9/configparser.py", line 789, in get
Apr 29 12:52:23 iobroker python3[20974]:     value = d[option]
Apr 29 12:52:23 iobroker python3[20974]:   File "/usr/lib/python3.9/collections/__init__.py", line 941, in __getitem__
Apr 29 12:52:23 iobroker python3[20974]:     return self.__missing__(key)            # support subclasses that define __missing__
Apr 29 12:52:23 iobroker python3[20974]:   File "/usr/lib/python3.9/collections/__init__.py", line 933, in __missing__
Apr 29 12:52:23 iobroker python3[20974]:     raise KeyError(key)
Apr 29 12:52:23 iobroker python3[20974]: KeyError: 'use_amis_reader_intermediate'
Apr 29 12:52:23 iobroker python3[20974]: During handling of the above exception, another exception occurred:
Apr 29 12:52:23 iobroker python3[20974]: Traceback (most recent call last):
Apr 29 12:52:23 iobroker python3[20974]:   File "/home/kthemall/HoymilesZeroExport/HoymilesZeroExport.py", line 1337, in <module>
Apr 29 12:52:23 iobroker python3[20974]:     INTERMEDIATE_POWERMETER = CreateIntermediatePowermeter(DTU)
Apr 29 12:52:23 iobroker python3[20974]:   File "/home/kthemall/HoymilesZeroExport/HoymilesZeroExport.py", line 1281, in CreateIntermediatePowermeter
Apr 29 12:52:23 iobroker python3[20974]:     elif config.getboolean('SELECT_POWERMETER', 'USE_AMIS_READER_INTERMEDIATE'):
Apr 29 12:52:23 iobroker python3[20974]:   File "/usr/lib/python3.9/configparser.py", line 828, in getboolean
Apr 29 12:52:23 iobroker python3[20974]:     return self._get_conv(section, option, self._convert_to_boolean,
Apr 29 12:52:23 iobroker python3[20974]:   File "/usr/lib/python3.9/configparser.py", line 808, in _get_conv
Apr 29 12:52:23 iobroker python3[20974]:     return self._get(section, conv, option, raw=raw, vars=vars,
Apr 29 12:52:23 iobroker python3[20974]:   File "/usr/lib/python3.9/configparser.py", line 803, in _get
Apr 29 12:52:23 iobroker python3[20974]:     return conv(self.get(section, option, **kwargs))
Apr 29 12:52:23 iobroker python3[20974]:   File "/usr/lib/python3.9/configparser.py", line 792, in get
Apr 29 12:52:23 iobroker python3[20974]:     raise NoOptionError(option, section)
Apr 29 12:52:23 iobroker python3[20974]: configparser.NoOptionError: No option 'use_amis_reader_intermediate' in section: 'SELECT_POWERMETER'
Apr 29 12:52:23 iobroker systemd[1]: HoymilesZeroExport.service: Main process exited, code=exited, status=1/FAILURE
Apr 29 12:52:23 iobroker systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Apr 29 12:52:23 iobroker systemd[1]: HoymilesZeroExport.service: Consumed 1.204s CPU time.
Apr 29 12:52:24 iobroker systemd[1]: HoymilesZeroExport.service: Scheduled restart job, restart counter is at 5.
Apr 29 12:52:24 iobroker systemd[1]: Stopped HoymilesZeroExport Service.
Apr 29 12:52:24 iobroker systemd[1]: HoymilesZeroExport.service: Consumed 1.204s CPU time.
Apr 29 12:52:24 iobroker systemd[1]: HoymilesZeroExport.service: Start request repeated too quickly.
Apr 29 12:52:24 iobroker systemd[1]: HoymilesZeroExport.service: Failed with result 'exit-code'.
Apr 29 12:52:24 iobroker systemd[1]: Failed to start HoymilesZeroExport Service.

reserve85, 2024-04-29:
ne sorry, da ist noch ein bug, probiere nochmal in 1 min :-)

kthemall, 2024-04-29:
läuft ;-)

 Inverterlimit was already accepted at 462 Watt
Apr 29 13:00:09 iobroker python3[21065]: 2024-04-29 13:00:09 INFO     OpenDTU: Inverter "HM-800-Phase2" reachable: True
Apr 29 13:00:09 iobroker python3[21065]: 2024-04-29 13:00:09 INFO     powermeter AmisReader: -13 Watt
Apr 29 13:00:10 iobroker python3[21065]: 2024-04-29 13:00:10 INFO     powermeter AmisReader: -57 Watt
Apr 29 13:00:11 iobroker python3[21065]: 2024-04-29 13:00:11 INFO     powermeter AmisReader: -66 Watt
Apr 29 13:00:12 iobroker python3[21065]: 2024-04-29 13:00:12 INFO     overproducing: reduce limit based on previous limit setpoint by approximation
Apr 29 13:00:12 iobroker python3[21065]: 2024-04-29 13:00:12 INFO     setting new limit to 421 Watt
Apr 29 13:00:12 iobroker python3[21065]: 2024-04-29 13:00:12 INFO     OpenDTU: Inverter "HM-800-Phase2": setting new limit from 462 Watt to 421 Watt
Apr 29 13:00:15 iobroker python3[21065]: 2024-04-29 13:00:15 INFO     OpenDTU: Inverter "HM-800-Phase2": Limit acknowledged
Apr 29 13:00:15 iobroker python3[21065]: 2024-04-29 13:00:15 INFO     OpenDTU: Inverter "HM-800-Phase2" reachable: True
Apr 29 13:00:15 iobroker python3[21065]: 2024-04-29 13:00:15 INFO     powermeter AmisReader: -6 Watt
Apr 29 13:00:16 iobroker python3[21065]: 2024-04-29 13:00:16 INFO     powermeter AmisReader: 30 Watt
Apr 29 13:00:16 iobroker python3[21065]: 2024-04-29 13:00:16 INFO     setting new limit to 466 Watt
Apr 29 13:00:16 iobroker python3[21065]: 2024-04-29 13:00:16 INFO     OpenDTU: Inverter "HM-800-Phase2": setting new limit from 421 Watt to 466 Watt
Apr 29 13:00:19 iobroker python3[21065]: 2024-04-29 13:00:19 INFO     OpenDTU: Inverter "HM-800-Phase2": Limit acknowledged
Apr 29 13:00:21 iobroker python3[21065]: 2024-04-29 13:00:21 INFO     OpenDTU: Inverter "HM-800-Phase2" reachable: True
Apr 29 13:00:21 iobroker python3[21065]: 2024-04-29 13:00:21 INFO     powermeter AmisReader: -26 Watt
Apr 29 13:00:22 iobroker python3[21065]: 2024-04-29 13:00:22 INFO     powermeter AmisReader: -27 Watt
Apr 29 13:00:23 iobroker python3[21065]: 2024-04-29 13:00:23 INFO     powermeter AmisReader: -24 Watt
Apr 29 13:00:24 iobroker python3[21065]: 2024-04-29 13:00:24 INFO     Inverterlimit was already accepted at 466 Watt
Apr 29 13:00:25 iobroker python3[21065]: 2024-04-29 13:00:25 INFO     OpenDTU: Inverter "HM-800-Phase2" reachable: True
Apr 29 13:00:25 iobroker python3[21065]: 2024-04-29 13:00:25 INFO     powermeter AmisReader: -26 Watt
Apr 29 13:00:26 iobroker python3[21065]: 2024-04-29 13:00:26 INFO     powermeter AmisReader: -47 Watt

reserve85, 2024-04-29:
perfect, thanks for your feedback.

kthemall, 2024-04-29:
ich danke für die realisierung. ich gebe dass dann mal so an den entwickler der hard/software weiter. der hätte immer ganz gerne use-cases für seine tätigkeiten. danke nochmal und LG Tom

kthemall, 2024-04-29:
hi, kannst du bitte noch umbenennen auf: Amis IR Reader (NetzOÖ)
btw. hier liegt das amis-reader repository [https://github.com/mgerhard74/amis_smartmeter_reader](url)

danke und LG Tom

reserve85, 2024-04-29:
du meinst in der readme?

kthemall, 2024-04-29:
ja wenn das geht, er meinte das wäre passender. im code denke ich sollte man umlaute usw. vermeiden (außer in den kommentaren vielleicht ;-))