reserve85/HoymilesZeroExport #185: 3 WR Ausregelung pro Phase

smaicloud, 2024-04-27:
Beispiel:

3x HMS 1600 4T WR jeweils an L1, L2, L3.

Grundlast 400W an L1
Wasserboiler 2000W an L2
Spülmaschine 2000W an L3

Aktuell wir parallel ausgeregelt. Läuft der Boiler und die Spülmaschine nicht, sind die Lasten an L2 und L3 gering. Läuft der Boiler ist L2 stärker belastet, alle 3 WR gleichen das parallel aus.

Den WR an L1 habe ich bereits mit min_watt etwas angehoben, so das die 400W ausgeregelt werden (weil die relativ konstant sind).

Ist der Boiler an L2 in Betrieb, so wäre es schön wenn der 2. WR versucht die Leistung bereit zu stellen. Der WR hat 1600W, der Boiler 2000W. Der 2. WR regelt auf max, schafft er die Leistung nicht so regeln die anderen parallel nach.

Oder Klimaanlage an L2, Last 400W, WR1 versucht die 400W Grundlast abzugeben, WR2 versucht die 400W Klima zu versorgen und auf L3 werden nur 50W verbraucht, WR3 stellt nur 50W bereit.

Somit könnte man die Schieflasten besser ausgleichen.

Ich hoffe mein anliegen verständlich erklärt zu haben.

smaicloud, 2024-04-27:
Wenn man bei dem WR in der config den Power Sensor für die Phasen angeben würde, so wüsste der WR wie er die Leistung anpassen müsste. Schafft der eine das nicht, so regeln alle bis zu voller Leistung parallel nach.

smaicloud, 2024-04-29:
Ich sehe gerade, es gab den request schon. Du kannst diesen mergen mit dem #172 .

reserve85, 2024-04-29:
Yep hab ich auch gesehen, ich denke mal darüber nach. Demnach müsste ich an den Zählern alle drei Phasen über REST auslesen und dann priorisieren… bringt das Vorteile? Ist eigentlich nicht genau deswegen der Zähler saldierend?

smaicloud, 2024-04-30:
Vorteil wäre, man würde die Schieflast zu den Phasen ausgleichen. Diese ist in DE zwar 4,6 kW, aber es wäre somit balancierter. Ein Wasserboiler hat ja schon 2 kW. Ich habe 2 Boiler einphasig mit á 2 kW. Mit 3 WR macht das dann schon Sinn. Im #172 hat er den Sachverhalt, das sein Zähler wohl nicht saldierend ist. Somit macht der Ausgleich für ihn ebenfalls Sinn mit 3 WR.

Leuchte777, 2024-05-09:
Zu #172 möchte ich noch weiter ausführen.
Bei Verwendung von Shelly 3EM oder Shelly Pro 3EM ist bereits die Unterscheidung für die einzelnen Phasen möglich. 

Im Script müßte der Verbrauch der einzelnen Phasen nur an die entsprechenden WR weitergereicht werden.
Die Zuordnung der WR zu den Phasen soll aber in der Config geschehen um flexible zu sein. Denn es kann vorkommen, daß bei einigen Installationen die Phasen vertauscht sind. Natürlich soll auch die Config berücksichtigt werden bzgl. HOY_MAX_WATT.

Im Script ist für Shelly (Pro) 3EM lediglich geringfügige Anpassung für das Logging nötig. Lediglich die Ausgabe-Funktion sowie die Classen-Definition für die Shelly´s. Das habe ich geschafft, aber die Daten dann im ganzen Script richtig weiter zu nutzen, dafür reicht mein Python (noch) nicht. Anbei gerne meine Ergänzung (nur für das Logging).

Was die anderen Strommess-Sensoren liefern, ist mir nicht bekannt. Die habe ich nicht zum testen.

Würde mich freuen, wenn mein Änderungswunsch umgesetzt werden kann, die Einspeisung der einzelnen Wechselrichter (L1-L3) entsprechend dem Verbrauch auf den Phasen zu steuern.

(Habe es leider nicht geschafft, hier im github die Script´s sauber einzubinden. Daher als PDF.)
[Change_L1-L3.pdf](https://github.com/reserve85/HoymilesZeroExport/files/15260301/Change_L1-L3.pdf)

reserve85, 2026-11-11:
Eine Regelung pro Phase ist nicht geplant. Sie würde eine eigene Regelung je Wechselrichter mit Phasenwerten vom Zähler erfordern, also einen Umbau des Regelkerns, und bringt bei saldierenden Zählern keinen finanziellen Vorteil. Ich schließe das Issue.

closed 2026-11-11 by reserve85