reserve85/HoymilesZeroExport #187: MQTT Config section

smaicloud, 2024-04-29:
Ich wollte gerade die mqtt config probieren. Leider kann man kein user/password angeben, somit funktioniert es nicht. Oder übersehe ich etwas?

reserve85, 2024-04-29:
doch, müsste ich noch in die Config mit reinnehmen.
Aus dem Quellcode:
if config.has_section("MQTT_CONFIG"):
    broker = config.get("MQTT_CONFIG", "MQTT_BROKER")
    port = config.getint("MQTT_CONFIG", "MQTT_PORT", fallback=1883)
    client_id = config.get("MQTT_CONFIG", "MQTT_CLIENT_ID", fallback="HoymilesZeroExport")
    username = config.get("MQTT_CONFIG", "**MQTT_USERNAME**", fallback=None)
    password = config.get("MQTT_CONFIG", "**MQTT_PASSWORD**", fallback=None)
    set_topic = config.get("MQTT_CONFIG", "MQTT_SET_TOPIC", fallback="zeropower/set")
    reset_topic = config.get("MQTT_CONFIG", "MQTT_RESET_TOPIC", fallback="zeropower/reset")

tomquist, 2024-04-29:
Ist schon in der config mit drin:

https://github.com/reserve85/HoymilesZeroExport/blob/346b1af68786aefead23cf138dbf5ef291f986e5/HoymilesZeroExport_Config.ini#L227-L243

smaicloud, 2024-04-29:
Getestet und funktioniert.

closed 2024-04-29 by smaicloud