reserve85/HoymilesZeroExport #189: Leistung aufteilen auf 3 Inverter (einer pro phase)

kthemall, 2024-05-01:
hi, 
habe gerade testweise versucht die leistung meiner 3 hoymiles "gerecht" aufzuteilen. ich dachte eigentlich dass die angeforderte leistung gelesen wird und dann je nach anzahl an invertern dividiert wird damit jeder gleich viel einspeist. jetzt ist es aber bei mir so dass ich jetzt 3x soviel einspeise wie ich müsste. habe ich hier noch irgendwie ein verständnisproblem oder fehlt mir noch in der config ein setting?
btw. das setting HOY_MAX_WATT = 50 bei inverter 2 und 3 ignoriert er komplett. geht immer drüber. stimmt da vielleicht auch noch was nicht?

danke tom

ich habe in meiner override ini datei folgende einträge:

[COMMON]
INVERTER_COUNT = 3
LOOP_INTERVAL_IN_SECONDS = 5
SET_LIMIT_TIMEOUT_SECONDS = 3
ON_GRID_USAGE_JUMP_TO_LIMIT_PERCENT = 0
SET_INVERTER_TO_MIN_ON_POWERMETER_ERROR = true

[CONTROL]
POWERMETER_TARGET_POINT = -15
POWERMETER_TOLERANCE = 10
POWERMETER_MAX_POINT = 0

[INVERTER_1]
SERIAL_NUMBER = zzzzzzzzzzzz
HOY_INVERTER_WATT = 800
HOY_MAX_WATT = 800
HOY_MIN_WATT_IN_PERCENT = 5

[INVERTER_2]
SERIAL_NUMBER = yyyyyyyyyyyy
HOY_INVERTER_WATT = 800
HOY_MAX_WATT = 50
HOY_MIN_WATT_IN_PERCENT = 5

[INVERTER_3]
SERIAL_NUMBER = xxxxxxxxxxx
HOY_INVERTER_WATT = 800
HOY_MAX_WATT = 50
HOY_MIN_WATT_IN_PERCENT = 5

reserve85, 2024-05-01:
Kannst du mal ein log zeigen?

kthemall, 2024-05-01:
anbei das log:

Mai 01 11:12:43 iobroker python3[31257]: 2024-05-01 11:12:43 INFO     OpenDTU: Inverter "HM-800-Phase1" reachable: True
Mai 01 11:12:43 iobroker python3[31257]: 2024-05-01 11:12:43 INFO     OpenDTU: Inverter "HM-800-Phase2" reachable: True
Mai 01 11:12:43 iobroker python3[31257]: 2024-05-01 11:12:43 INFO     OpenDTU: Inverter "HM-800-Phase3" reachable: True
Mai 01 11:12:43 iobroker python3[31257]: 2024-05-01 11:12:43 INFO     powermeter AmisReader: -211 Watt
Mai 01 11:12:44 iobroker python3[31257]: 2024-05-01 11:12:44 INFO     powermeter AmisReader: -203 Watt
Mai 01 11:12:45 iobroker python3[31257]: 2024-05-01 11:12:45 INFO     powermeter AmisReader: -50 Watt
Mai 01 11:12:46 iobroker python3[31257]: 2024-05-01 11:12:46 INFO     powermeter AmisReader: -50 Watt
Mai 01 11:12:47 iobroker python3[31257]: 2024-05-01 11:12:47 INFO     powermeter AmisReader: -107 Watt
Mai 01 11:12:48 iobroker python3[31257]: 2024-05-01 11:12:48 INFO     overproducing: reduce limit based on previous limit setpoint by approximation
Mai 01 11:12:48 iobroker python3[31257]: 2024-05-01 11:12:48 INFO     setting new limit to 556 Watt
Mai 01 11:12:48 iobroker python3[31257]: 2024-05-01 11:12:48 INFO     OpenDTU: Inverter "HM-800-Phase1": setting new limit from 560 Watt to 494 Watt
Mai 01 11:12:50 iobroker python3[31257]: 2024-05-01 11:12:50 INFO     OpenDTU: Inverter "HM-800-Phase1": Limit acknowledged
Mai 01 11:12:50 iobroker python3[31257]: 2024-05-01 11:12:50 INFO     OpenDTU: Inverter "HM-800-Phase2": setting new limit from 35 Watt to 30 Watt
Mai 01 11:12:54 iobroker python3[31257]: 2024-05-01 11:12:54 INFO     OpenDTU: Inverter "HM-800-Phase2": Limit timeout!
Mai 01 11:12:54 iobroker python3[31257]: 2024-05-01 11:12:54 INFO     OpenDTU: Inverter "HM-800-Phase3": setting new limit from 35 Watt to 30 Watt
Mai 01 11:12:57 iobroker python3[31257]: 2024-05-01 11:12:57 INFO     OpenDTU: Inverter "HM-800-Phase3": Limit timeout!
Mai 01 11:12:57 iobroker python3[31257]: 2024-05-01 11:12:57 INFO     OpenDTU: Inverter "HM-800-Phase1" reachable: True
Mai 01 11:12:57 iobroker python3[31257]: 2024-05-01 11:12:57 INFO     OpenDTU: Inverter "HM-800-Phase2" reachable: True
Mai 01 11:12:57 iobroker python3[31257]: 2024-05-01 11:12:57 INFO     OpenDTU: Inverter "HM-800-Phase3" reachable: True
Mai 01 11:12:57 iobroker python3[31257]: 2024-05-01 11:12:57 INFO     powermeter AmisReader: 39 Watt
Mai 01 11:12:57 iobroker python3[31257]: 2024-05-01 11:12:57 INFO     setting new limit to 610 Watt
Mai 01 11:12:57 iobroker python3[31257]: 2024-05-01 11:12:57 INFO     OpenDTU: Inverter "HM-800-Phase1": setting new limit from 494 Watt to 542 Watt

er speist aber trotzdem mehr ein als ich ihm erlaube (eingestellt wäre 50 mal zum testen, die opendtu ignoriert das aber)

lg tom

reserve85, 2024-05-01:
Ja, alles richtig aber openDTU gibt timeout zurück. Stimmt die Seriennummer?

kthemall, 2024-05-01:
ja die stimmen, gerade noch gecheckt. iwie ignoriert er die HOY_MAX_WATT = 50 von inverter 2 und 3 (solange ich da noch nicht pv angemeldet habe (bereits in arbeit) möchte ich nur auf 1 phase einspeisen, sonst kommt die netzagentur und macht dudu). kann es sein dass er für die [INVERTER_x] die Parameter ignoriert? die dtu hat timeout geliefert weil ich inv2 und3 wieder abgeschaltet habe (will da keine troubles haben momentan). der inv1 hat brav geantwortet.

Mai 01 12:07:02 iobroker python3[31257]: 2024-05-01 12:07:02 INFO     OpenDTU: Inverter "HM-800-Phase1": setting new limit from 489 Watt to 473 Watt
Mai 01 12:07:04 iobroker python3[31257]: 2024-05-01 12:07:04 INFO     OpenDTU: Inverter "HM-800-Phase1": Limit acknowledged
Mai 01 12:07:04 iobroker python3[31257]: 2024-05-01 12:07:04 INFO     OpenDTU: Inverter "HM-800-Phase2": setting new limit from 30 Watt to 29 Watt
Mai 01 12:07:08 iobroker python3[31257]: 2024-05-01 12:07:08 INFO     OpenDTU: Inverter "HM-800-Phase2": Limit timeout!
Mai 01 12:07:08 iobroker python3[31257]: 2024-05-01 12:07:08 INFO     OpenDTU: Inverter "HM-800-Phase3": setting new limit from 30 Watt to 29 Watt
Mai 01 12:07:11 iobroker python3[31257]: 2024-05-01 12:07:11 INFO     OpenDTU: Inverter "HM-800-Phase3": Limit acknowledged
Mai 01 12:07:11 iobroker python3[31257]: 2024-05-01 12:07:11 INFO     OpenDTU: Inverter "HM-800-Phase1" reachable: True
Mai 01 12:07:11 iobroker python3[31257]: 2024-05-01 12:07:11 INFO     OpenDTU: Inverter "HM-800-Phase2" reachable: True
Mai 01 12:07:11 iobroker python3[31257]: 2024-05-01 12:07:11 INFO     OpenDTU: Inverter "HM-800-Phase3" reachable: True

[INVERTER_1]
SERIAL_NUMBER = xxxx
HOY_INVERTER_WATT = 800
HOY_MAX_WATT = 800
HOY_MIN_WATT_IN_PERCENT = 5

[INVERTER_2]
SERIAL_NUMBER = yyyy
HOY_INVERTER_WATT = 800
HOY_MAX_WATT = 50
HOY_MIN_WATT_IN_PERCENT = 5

[INVERTER_3]
SERIAL_NUMBER = zzzz
HOY_INVERTER_WATT = 800
HOY_MAX_WATT = 50
HOY_MIN_WATT_IN_PERCENT = 5

danke tom

reserve85, 2024-05-01:
ich verstehe es nicht, das passt doch erstmal? du hast max. 50W bei Inverter 2 und 3 eingestellt und das Script stellt für den jeweiligen Inverter 29W ein... Alles im Lot?

Allerdings verstehe ich nicht, wieso er das macht (auf 29W) - normal sind ja 5% von 800W -> 40W als Minimum... Da muss ich mal schauen, auf die schnelle konnte ich da keinen Fehler finden.

Hast du neu gestartet nachdem du die Config angepasst hast?

kthemall, 2024-05-01:
so hab mich noch ein wenig gespielt und habe einen fehler in der override datei gefunden: bei 

[SELECT_INTERMEDIATE_METER]

war noch ein # drinnen. so ein blödsinn, sorry. jetzt läufts. kann zu.

closed 2024-06-13 by kthemall